Koenkk/zigbee2mqtt #22391: 1.37.0-1 caused Aqara SJCGQ11LM to report {"water_leak": null}

mojaki8286, 2024-05-02:
### What happened?

After upgrading to 1.37.0-1 my Aqara SJCGQ11LM water leak sensors started reporting NULL in the water_leak attribute. 
I have removed, rejoined, removed, rejoined and still reports NULL. 

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 Dongle Plus

### Setup

Add-on P4

julianfs, 2024-05-02:
If you get the device to change state (either dry > wet or wet > dry), does it then display correctly?

mojaki8286, 2024-05-03:
It just doesn't change. It just stated NULL. However, overnight it somehow reverted to "Clear" so status changed by itself.
The problem now is even if it gets in contact with water it doesn't change to "Leaking", it just stays "Clear". 

I'll remove it again from the network and rejoin it and will report back. Until then I will have to manually kick off the sump pump

mojaki8286, 2024-05-04:
Removed it, rejoined again. Water_leak = NULL and it just doesn't change the Water_Leak attribute status at all if it gets into contact with water. So this is a problem

mojaki8286, 2024-05-04:
> If you get the device to change state (either dry > wet or wet > dry), does it then display correctly?

Removed it from the network, rejoined. Water_Leak attribute value is NULL. This status does not change when the sensor gets in contact with water. So this is definitely something that worked prior to the update.

mojaki8286, 2024-05-05:
Hi so this is my summary,

After numerous testing. 

Removing, rejoining, resetting the device (Aqara SJCGQ11LM), 
adding to different router (I have 3 routers + 1 coordinator in my house). 
Restoring to the previous zigbee2mqtt version 1.36.1-1.
Restoring to hass core_2024.4.1. 

I got this to work. The water_leak attribute initially stated NULL, and would not change at all even if dumped into water.
All of a sudden it worked and changed from NULL to Leaking. Weird.

It took a day of testing but eventually came right.

closed 2024-05-05 by mojaki8286

stigvi, 2024-05-06:
> Hi so this is my summary,
>
> After numerous testing.
>
> Removing, rejoining, resetting the device (Aqara SJCGQ11LM), adding to different router (I have 3 routers + 1 coordinator in my house). Restoring [...]

Do I understand you right that you got it working when you restored zigbee2mqtt to 1.36.1 and it is not working on 1.37.0?

AntoineGS, 2024-05-12:
I also had to revert to 1.36.1 as all four of my SJCGQ11LM sensors stopped working on my Multi Pan SkyConnect.