Koenkk/zigbee2mqtt #22392: LOG not available

Destroyer061090, 2024-05-02:
### What happened?

Updating to last z2m version I have on log page of addon following error. I have updated z2m with core version 2024.4.4 anche checked also with 2024.5.0.

Impossibile ottenere i registri del componente aggiuntivo, No systemd-journal-gatewayd Unix socket available
[image]

### What did you expect to happen?

I expect to see the log

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230220210310-if00

### Adapter

/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230220210310-if00

### Setup

Core2024.4.4 or core2024.5.0 Supervisor2024.04.4 Frontend20240404.2

### Debug log

Not available

SchoutenHans, 2024-05-02:
Same here version 1.37.0 does not log device payloads at all when log is set to all,info,warning or error!

mathiaswyss, 2024-05-06:
Same here, not getting any device log entries since 1.37.0 on all log levels (for both frontend and log file).

Destroyer061090, 2024-05-06:
I guess it is a general problem for all addon logs... can you check it?

waltzy34, 2024-05-10:
Same here, no logs for addons, zigbee is down, can't even open frontend for Z2M

LaurentChardin, 2024-05-10:
You might want to have a look as the documentation has been updated : 
https://www.zigbee2mqtt.io/guide/usage/debug.html and particularly this one: https://www.zigbee2mqtt.io/guide/usage/debug.html#publishing-to-mqtt-and-to-frontend

closed 2024-12-07 by github-actions[bot]