Koenkk/zigbee2mqtt #22393: After upgrading to 1.37.0 zigbee2mqtt can't connect via IPv4

ZOXSOCKS, 2024-05-02:
### What happened?

After upgrading add-on in Home Assistant Operating System on Raspberry Pi 4, Zigbee2mqtt can no longer connect to broker and exits with: 

`[2024-05-02 06:33:11] info: 	z2m: Connecting to MQTT server at http://homeassistant.local:1883
[2024-05-02 06:33:11] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:11] error: 	z2m: MQTT failed to connect, exiting...
[2024-05-02 06:33:11] info: 	z2m: Stopping zigbee-herdsman...
[2024-05-02 06:33:12] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:13] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:14] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:15] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:16] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:17] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:18] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:19] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:20] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:21] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-05-02 06:33:21] info: 	zh:zstack:znp: closing
[2024-05-02 06:33:21] info: 	zh:zstack:znp: Port closed
[2024-05-02 06:33:21] info: 	z2m: Stopped zigbee-herdsman`

### What did you expect to happen?

Z2M has been able to connect with no problem until updating to 1.37.0.

### How to reproduce it (minimal and precise)

Set-up Z2M with mosquito broker as normal with standard username and password 
Start add-on
Add-on fails to connect with above error and exits.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

N/A

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Pi 4, Home assistant OS 

### Debug log

`[2024-05-02 06:33:11] info: 	z2m: Connecting to MQTT server at http://homeassistant.local:1883
[2024-05-02 06:33:11] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:11] error: 	z2m: MQTT failed to connect, exiting...
[2024-05-02 06:33:11] info: 	z2m: Stopping zigbee-herdsman...
[2024-05-02 06:33:12] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:13] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:14] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:15] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:16] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:17] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:18] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:19] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:20] error: 	z2m: MQTT error: connect ECONNREFUSED /
[2024-05-02 06:33:21] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-05-02 06:33:21] info: 	zh:zstack:znp: closing
[2024-05-02 06:33:21] info: 	zh:zstack:znp: Port closed
[2024-05-02 06:33:21] info: 	z2m: Stopped zigbee-herdsman`

ZOXSOCKS, 2024-05-02:
Update:

I have resolved the issue by removed all of the server details in the configuration as I am using mosquitto.

For what it's worth, I've had those server details populated since I first installed the add-on over a year ago, but it appears to have started affecting it after updating to 1.37.0.

fdbdr1, 2024-05-02:
Addition:

The problem is the server protocol:
`[2024-05-02 06:33:11] info: 	z2m: Connecting to MQTT server at http://homeassistant.local:1883`

My issue was solved by changing that from `http` to `mqtt`