Koenkk/zigbee2mqtt #22394: FW update reappears but doesn't actually exist

Stooovie, 2024-05-02:
### What happened?

Z2M and subsequently HA keep showing an update for IKEA E1734 dimmer that apparently doesn't actually exist. This is been going on for months now. Ignoring it in HA only lasts until next HA update. Checking for updates is disabled in Z2M. 
[image]

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

20230507

### Adapter

Slae.sh CC2562RB

### Setup

Add-on in HA

Nerivec, 2024-05-02:
There is a firmware update for that device looks like (according to [ikea URL](http://fw.ota.homesmart.ikea.net/feed/version_info.json)).

```json
{"fw_binary_url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/tradfri_dimmer-24.4.5-prod.ota.ota.signed","fw_file_version_LSB":5,"fw_file_version_MSB":9220,"fw_filesize":214692,"fw_image_type":4554,"fw_manufacturer_id":4476,"fw_type":2},
```

Did you try the usual, pushing a button on the device to wake it up before starting the update?
Lots of Ikea stuff needs it: https://www.zigbee2mqtt.io/devices/E2201.html#troubleshooting-device-didn-t-respond-to-ota-request (just a random Ikea device that has the procedure described)

Stooovie, 2024-05-03:
You are, of course, completely correct. It's been some time since I had to activate a zigbee thing to update it. 

I still think it's a bug as the update reappears on every HA restart and does not stay ignored but okay. Thanks!

closed 2024-11-30 by github-actions[bot]