Koenkk/zigbee2mqtt #22396: Powertag "zh:controller:buffalozcl: Unknown attribute" repeatedly in log

rbswift, 2024-05-02:
### What happened?

Since upgrading to the latest version I have the following events repeatedly logged.

Seems to relate to this device: https://www.zigbee2mqtt.io/devices/A9MEM1570.html#schneider%2520electric-a9mem1570

`May 02 17:20:28 alarm npm[5128]: [2024-05-02 17:20:28] info:         zh:controller:buffalozcl: Unknown attribute 16652 in cluster seMetering
May 02 17:20:34 alarm npm[5128]: [2024-05-02 17:20:34] info:         zh:controller:buffalozcl: Unknown attribute 16654 in cluster seMetering
May 02 17:20:44 alarm npm[5128]: [2024-05-02 17:20:44] info:         zh:controller:buffalozcl: Unknown attribute 16403 in cluster seMetering
May 02 17:20:44 alarm npm[5128]: [2024-05-02 17:20:44] info:         zh:controller:buffalozcl: Unknown attribute 16655 in cluster seMetering
May 02 17:21:00 alarm npm[5128]: [2024-05-02 17:21:00] info:         zh:controller:buffalozcl: Unknown attribute 16652 in cluster seMetering
May 02 17:21:05 alarm npm[5128]: [2024-05-02 17:21:05] info:         zh:controller:buffalozcl: Unknown attribute 16654 in cluster seMetering
May 02 17:21:10 alarm npm[5128]: [2024-05-02 17:21:10] info:         zh:controller:buffalozcl: Unknown attribute 16384 in cluster seMetering
May 02 17:21:10 alarm npm[5128]: [2024-05-02 17:21:10] info:         zh:controller:buffalozcl: Unknown attribute 16653 in cluster seMetering
May 02 17:21:15 alarm npm[5128]: [2024-05-02 17:21:15] info:         zh:controller:buffalozcl: Unknown attribute 16403 in cluster seMetering
May 02 17:21:15 alarm npm[5128]: [2024-05-02 17:21:15] info:         zh:controller:buffalozcl: Unknown attribute 16655 in cluster seMetering
May 02 17:21:36 alarm npm[5128]: [2024-05-02 17:21:36] info:         zh:controller:buffalozcl: Unknown attribute 16654 in cluster seMetering
May 02 17:21:41 alarm npm[5128]: [2024-05-02 17:21:41] info:         zh:controller:buffalozcl: Unknown attribute 16384 in cluster seMetering
May 02 17:21:41 alarm npm[5128]: [2024-05-02 17:21:41] info:         zh:controller:buffalozcl: Unknown attribute 16653 in cluster seMetering
`

### How to reproduce it (minimal and precise)

just updated to 1.37

### Zigbee2MQTT version

1.37

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

plain - Raspberry Pi

jbouv55151, 2024-05-02:
I do have the same problem and the same product.

Nerivec, 2024-05-02:
This is a logging issue only. It will be lowered back to `debug` level, where it should have stayed 😉

rbswift, 2024-05-02:
> This is a logging issue only. It will be lowered back to `debug` level, where it should have stayed 😉

Thanks for that. I wonder what the additional data might be? Anything interesting?

Nerivec, 2024-05-02:
Lots of stuff that manufacturers intend for their hubs. Sometimes it proves useful to Z2M (after lengthy investigations into what they mean 😅), sometimes not.

Koenkk, 2024-05-04:
Moved to debug!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-04 by Koenkk