Koenkk/zigbee2mqtt #22397: aqara E1 blinds appears unknown after last z2m update 1.37.0

Prudax, 2024-05-02:
### What happened?

my aqara E1 blinds motors appears unknown after  last update, i still can control it but on dashboard shows "UNKNOWN", so i rolled back to previous update and all works ok.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20220125

### Adapter

sonoff dongle p

### Setup

mini pc bear metal

cromelex, 2024-05-02:
Same issue here. Not sure if this is with Zigbee2MQTT 1.37.0 or Home Assistant 2024.05.0

To be clear, Z2M seems to have all the info, but in Home Assistant the state of the cover.entity is "unknown"

{
    "battery": 43,
    "charging_status": false,
    "device_temperature": 23,
    "last_seen": "2024-05-02T11:14:40+01:00",
    "linkquality": 116,
    "motor_speed": "medium",
    "motor_state": "pause",
    "position": 36,
    "power_outage_count": 2,
    "running": false,
    "state": "OFF"
}

[image]

Looking at some automations I had, the attribute in HASS was previously "**position**", but now it is "**current_position**".

Was this changed in 1.37.0 ? That could be the cause of this issue.

balkonrad, 2024-05-02:
Same for me, in core log also warning occurs:

```
Recorder: homeassistant.components.mqtt.cover
Source: components/mqtt/cover.py:416
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [Issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First event: 12:45:26 (7 events)
Last logged: 12:45:41

Payload is not supported (e.g. open, closed, opening, closing, stopped): pause
Payload is not supported (e.g. open, closed, opening, closing, stopped): declining
```

Prudax, 2024-05-02:
well i rolled back to 1.36.1-1 and my rollers ok now.

cromelex, 2024-05-02:
> well i rolled back to 1.36.1-1 and my rollers ok now.

Can confirm the same, the issue is with zigbee2mqtt 1.37.0 rather than with Home Assistant 2024.05.0

gianlucasullazzo, 2024-05-02:
Hi Guys, same here.

Lezcarr, 2024-05-02:
same issue for me
2024-05-02 14:18:05] error: 	zh:controller:device: Read response to 0x54ef4410003e427f failed
[2024-05-02 14:18:38] error: 	z2m: Publish 'set' 'state' to 'Z2M Lounge Blind 2' failed: 'Error: ZCL command

Kazionauta, 2024-05-02:
Same here, First I rolled back to HA 2024.04.4 but there was still this problem. I updated HA to 2024.05 and dowgraded z2m and it works well.

Koenkk, 2024-05-02:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/22387

closed 2024-05-02 by Koenkk