Koenkk/zigbee2mqtt #22399: Z2M crashes every 30-60 minutes after 1.37.0 update

razor-4eg, 2024-05-02:
### What happened?

After 1.37.0 update z2m start crashing randomly once in 1-2 hours

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.3.0.0 build 131

### Adapter

EZSP v12 ZB Dongle-E

### Setup

Add-on on HA OS on Intel x86 server

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15187036/log.log)
[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15187037/log.log)
[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15187042/log.log)

supaeasy, 2024-05-02:
Did you try using ember instead of ezsp?

razor-4eg, 2024-05-02:
> Did you try using ember instead of ezsp?

I guess I will need to in future. But it will be great if ezsp will stay "as is" how it was promised. Restored back to 1.36 version, and it stable now.

supaeasy, 2024-05-02:
Also you could try updating your Adapter to 7.4.2

pkoz79, 2024-05-02:
I have the same issue with [SONOFF ZB Dongle-P].
Restored to 1.36.1 and its stable again.

closed 2024-06-27 by razor-4eg