Koenkk/zigbee2mqtt #22400: 4512707 Namron lost support

tore1980, 2024-05-02:
### What happened?

4512707 Namron lost support. The device has been working for some years.

### What did you expect to happen?

4512707 Namron support

### How to reproduce it (minimal and precise)

Update Z2m to 1.37.0-1 from 1.36.1-1 and updated serial configuration to "adapter: ember"

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.2

### Adapter

Sonoff 

### Setup

X86

tore1980, 2024-07-15:
4512707 Namron  working with Z2M version 1.36.1-1, using Iluminize 511.10 converter. Newer Z2M versions has removed the converter.
https://www.elektroimportoren.no/namron-zigbee-5-kanal-led-kontroller/4512707/Product.html
[image]

tore1980, 2024-07-15:
[image]

tore1980, 2024-07-15:
Still a problem 1.39.0-1

Koenkk, 2024-07-16:
Added!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `4512707.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-07-16 by Koenkk

tore1980, 2024-07-18:
Great now it works! Tnx. I have added photo, it`s my first time so hope its correct.

bjornarfjelldal, 2025-01-16:
@tore1980 can you confirm color and white balance control is working on this model? On the model page on z2m it does not say so, but the illuminaze one does