Koenkk/zigbee2mqtt #22401: ember driver on EFR32MG21

raptr55, 2024-05-02:
### What happened?

got this error:
z2m: Error: NCP EZSP protocol version of 8 does not match Host version 13. 

i think a have to update the firmware but in the the page --> https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

its unclear wich is the correct version

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EFR32MG21 

### Setup

add on on home assistant

### Debug log

[2024-05-02 14:13:15] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-02 14:13:16] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-02 14:13:16] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-02 14:13:16] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-02 14:13:16] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-05-02 14:13:16] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-02 14:13:16] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-02 14:13:17] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-02 14:13:17] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-02 14:13:17] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-02 14:13:17] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-02 14:13:17] error: 	z2m: Failed to start zigbee
[2024-05-02 14:13:17] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-02 14:13:17] error: 	z2m: Exiting...
[2024-05-02 14:13:17] error: 	z2m: Error: NCP EZSP protocol version of 8 does not match Host version 13
    at EmberAdapter.emberVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1651:19)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:802:9)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2660:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)

wastez, 2024-05-02:
No don’t use this, use the newest from darkxst. 
 https://github.com/darkxst/silabs-firmware-builder

Koenkk, 2024-05-02:
You are using an outdated firmware, see https://github.com/Koenkk/zigbee2mqtt/discussions/21462

closed 2024-05-02 by Koenkk

dev-2-4-h, 2024-05-22:
This is a very simple solution:
> You are using an outdated firmware, see https://github.com/Koenkk/zigbee2mqtt/discussions/21462

However, there is no solution for this error related to the Sonnoff ZigbeBridge with Tasmota.

raptr55, 2024-05-29:
just to notify a web page that explain how to flash the ember driver on adapters:

https://smarthomescene.com/guides/how-to-use-ember-driver-with-efr32mg21-zigbee-adapters/