Koenkk/zigbee2mqtt #22405: NO LOGGING after update to V1.37.0

SchoutenHans, 2024-05-02:
### What happened?

After update to V 1.37.0 there is no logging of device payload changes or any other device logging.

### What did you expect to happen?

To see Logging

### How to reproduce it (minimal and precise)

Revert back to 1.36.1

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

Latest 

### Adapter

CC2652RB

### Setup

Dockwer on PI and zigbee plugin homeseer

FriedCheese2006, 2024-05-02:
Seeing the same here. Looks like the changes are DEBUG level now.

`[2024-05-02 13:17:52] debug: 	z2m: Received Zigbee message from 'Office Button', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0`

Koenkk, 2024-05-02:
Things have changed in the logging, z2m is just less verbose now, so this is expected.

closed 2024-05-02 by Koenkk

Piscator74, 2024-05-06:
When I set the Log Level to debug in the "Log" tab of the front-end, it still shows nothing in the front-end.
It does show in the appropriate "log.log" file.

9901gorsh, 2024-05-06:
I have the same problem - no logs in Log tab with any Log level. z2m v1.37.0

magyarl89, 2024-05-25:
> Things have changed in the logging, z2m is just less verbose now, so this is expected.

Oh, what a pity. It was pretty useful to check how a new device works, what are the exact actions.
Now you have to tick "Log debug to MQTT and frontend" in Advanced options and set Log level to "debug". This gives you way more (confusing) data, hard to find the action lines.

It was easy to test, if a switch is working. Unfortunately the "Exposes" tab shows you only a blink of a switch action.
Is there a chance to get an user friendly filtered view of the actual actions of the devices in the frontend?