Koenkk/zigbee2mqtt #22408: Failed to configure Third Reality, Inc 3RSP02028BZ

haforme, 2024-05-02:
### What happened?

Third Reality, Inc device 3RSP02028BZ shows as failed to configure in logs.

### What did you expect to happen?

To configure correctly without errors.

### How to reproduce it (minimal and precise)

Installed 1.37.0-1 and saw error messages in log. Restart of Z2M also produces same results.

### Zigbee2MQTT version

1.37.0-1 and maybe 1.36.1-1

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Setup

Home Assistant Add-on running on Raspberry 4B

### Debug log

[2024-05-02 12:16:28] info: 	z2m: Configuring 'Cellar Temp-Humidity Display'
[2024-05-02 12:17:29] error: 	z2m: Failed to configure 'Cellar Temp-Humidity Display', attempt 3 (Error: Bind 0x282c02bfffe3a4bb/1 genPowerCfg from '0x00212effff065ea4/1' failed (waiting for response TIMEOUT)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:774:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:464:13)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:75:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:112:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:160:21)
    at Configure.configure (/app/lib/extension/configure.ts:121:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

Nerivec, 2024-05-03:
Oh, I was reading the logs, they are for a `Cellar Temp-Humidity Display` so I assumed it was battery-powered.

haforme, 2024-05-03:
My bad the model is really a Third Reality 3RTHS24BZ which is a battery operated end device.

haforme, 2024-05-03:
With the reformatting of the Z2M logs it became more evident that the Third Reality 3RTHS24BZ devices were failing to configure.  I looked at the previous Z2M logs and the errors were there also. I will try reconfigure them by keeping them awake.

closed 2024-05-03 by haforme

Horse2370, 2024-05-21:
haforme - did you ever resolve the error. I have the same sensor and it shows up a "main powered" and not battery as expected and fails to configure the genPowerCfg.
I've been unable to figure out how to wake the sensor up. re pairing does not help.