Koenkk/zigbee2mqtt #22409: Migrate 1.37 sonoff zbdongle-e ezsp to ember tutorial ?

kwetnico, 2024-05-02:
### What happened?

Hello with the new version z2m 1.37 it is requested to update the sonoff key to firmware 7.4 (my version is 6.10. 3) and to change in the configuration.xml file ezsp to ember. 
Is there a tutorial to show us how to do it  ?
All our devices must be repaired again, is there a backup to be made? 
With version 1.37 can we leave the configuration with ezsp or is this no longer recognized?
If so, how is it not clear?
@Koenkk @Nerivec 
Thanks 

### What did you expect to happen?

. 

### How to reproduce it (minimal and precise)

. 

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Pi4b

wastez, 2024-05-02:
You can update to 7.4.2 and don’t have to repair your devices even if you use ember. 
and yes you still can configure ezsp, it’s just not default anymore.

wastez, 2024-05-02:
But ember is working really fast, good and rock solid. I would give it a chance if you where you.

Koenkk, 2024-05-02:
See `How to start testing` from https://github.com/Koenkk/zigbee2mqtt/discussions/21462

closed 2024-05-02 by Koenkk