Koenkk/zigbee2mqtt #22412: lights not controlable after upgrade to 1.37 and skyconnect 7.4

remb0, 2024-05-02:
### What happened?

I updated zigbee2mqtt,  changed to ember and restarted.
after that al light als offline  and cannot be controled.  if I allow pairing a bulb can be paired,  i see state but cannot change.

I changed the firmware to: ncp-uart-hw-v7.4.1.0-skyconnect-115200 (1)   after the skyconnect version of 7.4 but  32 devices (all lights)  are offline

### What did you expect to happen?

Hoped that a mapping,  or repair would helped.

### Zigbee2MQTT version

1.37

### Adapter firmware version

7.4

### Adapter

skyconnect

### Setup

add-on home assistant on nuc

### Debug log

[2024-05-02 22:13:44] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-02 22:13:45] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-02 22:13:45] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-02 22:13:45] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-02 22:13:45] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-02 22:13:45] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-02 22:13:45] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-02 22:13:45] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-02 22:13:46] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-02 22:13:46] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-02 22:13:46] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-02 22:13:46] warning: 	zh:ember: [EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY. Firmware value will be used instead.
[2024-05-02 22:13:46] info: 	zh:ember: [STACK STATUS] Network up.
[2024-05-02 22:13:46] info: 	zh:ember: [INIT TC] NCP network matches config.
[2024-05-02 22:13:46] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-05-02 22:13:46] info: 	z2m: zigbee-herdsman started (resumed)
[2024-05-02 22:13:46] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":1,"revision":"7.4.1 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-05-02 22:13:46] info: 	z2m: Currently 58 devices are joined:
[2024-05-02 22:13:46] info: 	z2m: Dakkapel (0x00158d00044f28cc): SJCGQ11LM - Aqara Water leak sensor (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: kantoor1 (0x588e81fffe59e5d9): LED1837R5 - IKEA TRADFRI bulb GU10, warm white, 400 lm (Router)
[2024-05-02 22:13:46] info: 	z2m: kantoor2 (0x842e14fffe198488): LED1837R5 - IKEA TRADFRI bulb GU10, warm white, 400 lm (Router)
[2024-05-02 22:13:46] info: 	z2m: ikea_smart_outlet_switch (0xd0cf5efffe6d24eb): E1603/E1702/E1708 - IKEA TRADFRI control outlet (Router)
[2024-05-02 22:13:46] info: 	z2m: switch_trapgat_click (0x00158d00016d7c01): WXKG01LM - Xiaomi Mi wireless switch (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: kastkamer_sw (0x00158d00027c1640): WXKG02LM_rev2 - Aqara Wireless remote switch (double rocker), 2018 model (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: motion_badkamer_occupancy (0x00158d0001db083c): RTCGQ01LM - Xiaomi Mi motion sensor (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: badkamer (0x00158d0001ad350d): WSDCGQ01LM - Xiaomi Mi temperature and humidity sensor (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: temphum (0xa4c138dd81215bc4): LYWSD03MMC-z - Xiaomi Temp & RH Monitor Lite (pvxx/ZigbeeTLc) (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: kantoor (0x04cf8cdf3c77ca90): WXCJKG12LM - Aqara Opple wireless switch (double band) (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: smoke_detector_smoke (0x00158d0001d84325): JTYJ-GD-01LM/BW - Xiaomi Mijia Honeywell smoke detector (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: nachtkast (0x00158d0001e064dc): WXKG02LM_rev1 - Aqara Wireless remote switch (double rocker), 2016 model (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: trapgat_led_light (0x000b57fffe9b7381): LED1537R6/LED1739R5 - IKEA TRADFRI bulb GU10, white spectrum, 400 lm (Router)
[2024-05-02 22:13:46] info: 	z2m: kastkamer3 (0x842e14fffe47b499): LED1837R5 - IKEA TRADFRI bulb GU10, warm white, 400 lm (Router)
[2024-05-02 22:13:46] info: 	z2m: xiaomi_remote_3 (0x04cf8cdf3c77c79a): WXCJKG13LM - Aqara Opple wireless switch (triple band) (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: meterkast_contact (0x00158d000171e55b): MCCGQ01LM - Xiaomi Mi door and window sensor (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: hoekbol_light (0x000b57fffeac0bd9): LED1624G9 - IKEA TRADFRI bulb E12/E14/E26/E27, color/white spectum, globe, opal, 600 lm (Router)
[2024-05-02 22:13:46] info: 	z2m: hoeklamp_light (0xccccccfffe5f2de3): LED1624G9 - IKEA TRADFRI bulb E12/E14/E26/E27, color/white spectum, globe, opal, 600 lm (Router)
[2024-05-02 22:13:46] info: 	z2m: kastkamer2 (0xccccccfffe4cb4f1): LED1837R5 - IKEA TRADFRI bulb GU10, warm white, 400 lm (Router)
[2024-05-02 22:13:46] info: 	z2m: kastkamer1 (0x14b457fffeca0150): LED1837R5 - IKEA TRADFRI bulb GU10, warm white, 400 lm (Router)
[2024-05-02 22:13:46] info: 	z2m: sw_trapgat_zolder (0x00158d00016bdea2): WXKG01LM - Xiaomi Mi wireless switch (EndDevice)
[2024-05-02 22:13:46] info: 	z2m: zithoek_light (0x00178801089f4f68): 8718699688882 - Philips Hue white Filament bulb G93 E27 bluetooth (Router)
[2024-05-02 22:13:46] info: 	z2m: smoke_detector_2_smoke (0x00158d00028ee224): JTYJ-GD-01LM/BW - Xiaomi Mijia Honeywell smoke detector (EndDevice)
[... 56 lines cut ...]
[2024-05-02 22:15:39] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":0,"sequence":174} messageTag=255]
[2024-05-02 22:15:44] warning: 	z2m: Failed to ping 'Gang-1' (attempt 1/1, ZCL command 0x00178801105238ac/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":11,"options":4160,"groupId":0,"sequence":250}))
[2024-05-02 22:15:54] warning: 	z2m: Failed to ping 'Buitenverlichting' (attempt 1/1, ZCL command 0x0017880110523338/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":11,"options":4160,"groupId":0,"sequence":251}))
[2024-05-02 22:16:05] warning: 	z2m: Failed to ping 'Plantenbak' (attempt 1/1, ZCL command 0x00158d0004ae457c/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":253}))
[2024-05-02 22:16:14] warning: 	z2m: Failed to ping 'wc_light' (attempt 1/1, ZCL command 0x0017880110306d81/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":11,"options":4160,"groupId":0,"sequence":255}))
[2024-05-02 22:16:24] warning: 	z2m: Failed to ping 'wasmachine' (attempt 1/1, ZCL command 0x588e81fffed4aed3/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":0}))
[2024-05-02 22:16:34] warning: 	z2m: Failed to ping 'droger' (attempt 1/1, ZCL command 0x588e81fffed39e90/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transa [... cut ...]
[2024-05-02 22:18:41] warning: 	z2m: Failed to ping 'Human presence' (attempt 1/1, ZCL command 0xa4c13860583fcd1d/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":23}))

Nerivec, 2024-05-02:
Try to flash [this firmware](https://github.com/darkxst/silabs-firmware-builder/blob/ember-nohw/firmware_builds/skyconnect/ncp-uart-hw-v7.4.1.0-skyconnect-115200.gbl) from darkxst and then set `rtscts: false` in your `configuration.yaml` (or just remove the line if you have it set to `true` currently) [docs](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter)

Just in case, you can also try to shutdown Z2M, unplug your adapter for 30seconds, plug it back in, then try to start Z2M again.

remb0, 2024-05-02:
I flashed that firmware.   But because i have to sleep and all lights are on, I restored the zigbee2mqtt backup (but still same firmware) and everything works :)    Have to try your solution this sunday.

mitsumaui, 2024-05-04:
Same here with zzh! stick.

Other entities such as temp sensors were still getting updates, and the status was showing correctly in HA - but the ability to control was greyed out. 
Lights were still controllable via Z2M web UI. 

Worked fine again once I reverted to 1.36.1

remb0, 2024-05-05:
revert was ok.   but now I upgraded again. but stil in the config: ezsp
`Zigbee2MQTT versie
[1.37.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coördinator type
EZSP v13
Coordinator revisie
7.4.2.0 build 0
Coordinator IEEE adres
0x287681fffef07388
Versie frontend
0.6.165
zigbee_herdsman_converters_version
19.32.0
zigbee_herdsman_version
0.45.0`

raarts, 2024-05-12:
Same problem, after I upgraded my SkyConnect to `ncp-uart-hw-v7.4.2.0-skyconnect-115200.gbl` and changed to `ember` driver.

All fixed when I set the driver back to `ezsp`.

forgenator, 2024-05-13:
I just flashed 7.4.3.0 to my skyconnect and with the latest 1.37 zigbee2mqtt and everything works just fine. Driver as ember, and didn't add the rtscts configuration so it's nonexistant.

raarts, 2024-05-13:
Where did you find that version? It's not in the [download](https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee) directory.

forgenator, 2024-05-14:
It's in a different branch (ember-nohw), I have no idea why. But I was just, maybe it works, and it did? (https://github.com/darkxst/silabs-firmware-builder/blob/ember-nohw/firmware_builds/)

forgenator, 2024-05-14:
[image]

Here are the versions i'm using.

forgenator, 2024-05-14:
Tried with hardware flow control `rtscts: true` to test things in configuration.yaml and it broke everything, for those intrested.

remb0, 2024-05-19:
I use rtscts: false
falshed back to https://github.com/darkxst/silabs-firmware-builder/raw/ember-nohw/firmware_builds/skyconnect/ncp-uart-hw-v7.4.1.0-skyconnect-115200.gbl
and tested again with ember.

same result:  all powered router devices are offline.

remb0, 2024-05-19:
here is my log file  In case i missed a clue:
[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15368780/log.log)

remb0, 2024-05-19:
> @remb0 Can you confirm this with 1.37.1 with `ember`? If still an issue, please provide a `debug` log file of the behavior.

This is the log file you mean?

huotarih, 2024-05-23:
I flashed [ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl](https://github.com/darkxst/silabs-firmware-builder/blob/ember-nohw/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl) to my SonOFF ZigBee Dongle-E and it started working. Also worked fine after the update for a short time then stopped working. Also I deleted the backup for the coordinator, don't know if that had any effect though.

marcuscps, 2024-05-26:
What's the current status of this issue?

I was using the Ember firmware, but don't remember the exact firmware version. I'm sure it was 7.4.x, I thing 7.4.2.
It became pretty unstable, and I decided to revert to the ezsp driver. Everything started working again.

However, I can't add new devices to Z2M after 1.37.
I initially updated the firmware exactly for this reason.

Now I'm a bit unsure if I should stick to working (ezsp) without being able to add new devices or upgrading and risking breaking things again by using Ember.

Thanks

closed 2024-12-28 by github-actions[bot]