Koenkk/zigbee2mqtt #22413: Sonoff Dongle-E with firmware 7.4.2 and ZB2MQTT Version 1.37

NIcozozo57, 2024-05-02:
### What happened?

With version 1.36.1-1 and driver ember, everythin gis ok
With version 1.37 and driver ember, it won't start with message below 👍 
[2024-05-02 23:11:32] info: 	z2m: Connecting to MQTT server at mqtt://homeassistant.local/:1883
[2024-05-02 23:11:32] error: 	z2m: MQTT error: connect ENOENT /:1883
[2024-05-02 23:11:32] error: 	z2m: MQTT failed to connect, exiting...

### What did you expect to happen?

after upgrade to version 1.37

### How to reproduce it (minimal and precise)

no other change except the upgrade

### Zigbee2MQTT version

1.37

### Adapter firmware version

7.4.2 [GA]

### Adapter

EFR32MG21

### Setup

rpi4

Nerivec, 2024-05-02:
Make sure your MQTT instance is running properly. Looks like it's not.
Try a full reboot of the machine if it persists (and you haven't already).

qxzzxq, 2024-05-02:
I had the same issue after upgrading to 1.37.01. It seems that the url `mqtt://homeassistant.local/:1883` is incorrect. 

Removing the "/" before the port number in the configuration resolved the error.

The correct url should be:
```yml
server: mqtt://homeassistant.local:1883
```

However, I am not sure whether the slash was recently introduced with the upgrade or if it existed previously but went unnoticed because earlier versions managed the malformed URL appropriately.

NIcozozo57, 2024-05-03:
Yes [qxzzxq](https://github.com/qxzzxq) your trick fix tmy issue.
version before 1.37 accept the malformed URL, removing the slash before the port make it work.
Thx

closed 2024-05-03 by NIcozozo57

MarkCorazon, 2024-05-03:
I've got the same error after the latest update but a different problem. The configuration `server: mqtt://192.168.0.105:1883,` suddenly had a comma after the server address. This is solved after removing the comma.

oliversweb, 2024-05-03:
I have reverted to 1.36 ... rather than rolling forward and fixing can I wait for 1.38 where it will be fixed?

NIcozozo57, 2024-05-05:
Hi,
The issue is related to the URL which contain a slash at the end, and
version 1.37 didn't accept it, just remove it as below :
From :
server: mqtt://homeassistant.local/:1883
to :
server: mqtt://homeassistant.local:1883