Koenkk/zigbee2mqtt #22415: After upgrade to 1.37.0. Option to rename the devices don’t show to add to home assistant

aperezva, 2024-05-03:
### What happened?

After upgrade to 1.37. A can rename existing and new devices but option to rename in home assistant disappears 

### What did you expect to happen?

Add and change devices in home assistant

### How to reproduce it (minimal and precise)

After upgrade to 1.37 in lxc promox server

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20221226

### Adapter

Slz06

### Setup

Lxc proxmox

youplaymenot, 2024-05-03:
Same issue here on home assistant green. The blue rename button is gone.

jymorel, 2024-05-03:
Same here. Hope this very useful option will come back in next release ...

aperezva, 2024-05-03:
> Same here. Hope this very useful option will come back in next release ...

I did a rollback in proxmox, and is working fine in proviois 1.36 version.

shalak, 2024-05-03:
I hope this is a bug, and not "a feature". I didn't see mention about this in the release notes, unless I'm missing something.

Anybody got a workaround for this issue?

aperezva, 2024-05-03:
> I hope this is a bug, and not "a feature". I didn't see mention about this in the release notes, unless I'm missing something.
> 
> Anybody got a workaround for this issue?

Rollback to previous version till owner solve the problem

S33G, 2024-05-03:
The buttons were accidentally hidden on mobile browser width. Sorry about that.

A fix has been applied to the front end project.
All we need now is a Zigbee2MQTT release containing that fix.

https://github.com/nurikk/zigbee2mqtt-frontend/pull/2012

Koenkk, 2024-05-04:
Already fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2024-05-04 by Koenkk

aperezva, 2024-05-04:
> Already fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thanks,

Will you promote to a new release soon?

I m not using Dev.

Koenkk, 2024-05-04:
I will publish a hotfix release in the coming days.

aperezva, 2024-05-04:
Thanks a lot

RobertoCravallo, 2024-05-04:
I am a little confused. It seems there are two different problems in this thread. 

This is mine. No way to set the HA name anymore!!!
[image]

Ralf-Pfalz, 2024-05-05:
> I am a little confused. It seems there are two different problems in this thread.
> 
> This is mine. No way to set the HA name anymore!!! ![c7b96c8b564be0c4cc1911fe44065ac185cc10fb](https://private-user-images.githubusercontent.com/112649622/327967433-285cc9cd-b25d-405a-901b-c14b57c671de.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ4OTk1NzYsIm5iZiI6MTcxNDg5OTI3NiwicGF0aCI6Ii8xMTI2NDk2MjIvMzI3OTY3NDMzLTI4NWNjOWNkLWIyNWQtNDA1YS05MDFiLWMxNGI1N2M2NzFkZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwNVQwODU0MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZWU3NDM3ZmNhOTNjMTcxNGEwYmM2Mzc2OWEyYmQwZDU1YmM1MzQ5NzY0MjJhYzBmY2NlNmNmZmFjZjZjZjY2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.53RwFeT5oUAJ3n1A7z7JcxMweGSvHiy2PNRfi0TS96w)

Yes, this ist the problem. Same here...

Mark-Muc, 2024-05-05:
exact the same here ..