Koenkk/zigbee2mqtt #22416: 'commandTuyaAction' Duplicate messages cannot be determined

1020844469, 2024-05-03:
### What happened?

Tuya may send a repeated message after pressing the button once, but because this repeated data is exactly the same as the data received after pressing the button for the second time, two mqtt messages of action_1 will be published for the same button press.

Like this, these are two identical messages sent by Tuya with the same key press. Previously, hasAlreadyProcessedMessage could be used to process duplicate messages. But now because the data is the same, it cannot be processed this way:

Received Zigbee message from 'MY_BUTTON', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 2 with groupID 0

Received Zigbee message from 'MY_BUTTON', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 2 with groupID 0

### What did you expect to happen?

I hope that duplicate messages can be filtered out. By pressing the key once, no matter how many times tuya sends duplicate messages, only one mqtt message will be published.

### How to reproduce it (minimal and precise)

Use a wireless switch and keep pressing buttons. Sometimes due to delay, tuya will send multiple repeated messages for the same button press.

### Zigbee2MQTT version

1.36,1,  1.37.0  

### Adapter firmware version

20221226

### Adapter

SONOFF CC2652P Zigbee 3.0 USB Dongle Plus

### Setup

Tested with add-on and docker, both are the same

### Debug log

[2024-05-03 13:17:41] debug: 	z2m: Received Zigbee message from 'MY_BUTTON', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 2 with groupID 0
[2024-05-03 13:17:41] debug: 	z2m: Received Zigbee message from 'MY_BUTTON', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 2 with groupID 0
[2024-05-03 13:17:41] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt2/MY_BUTTON/action', payload 'scene_2'
[2024-05-03 13:17:41] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt2/MY_BUTTON/action', payload 'scene_2'

Koenkk, 2024-05-04:
@kirovilya could you take a look at this?

1020844469, 2024-06-15:
I found that there is a sequence number in **transactionSequenceNumber**, and the sequence number of repeated packets is the same. But when the front end receives Zigbee message from, the data is [], so there is no way to judge. The solution is to just put this transactionSequenceNumber as before
Just write it into the data field

[2024-06-15 19:56:29] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":55196,"srcendpoint":6,"dstendpoint":1,"wasbroadcast":0,"linkquality":142,"securityuse":0,"timestamp":5270905,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,60,253,0]}}
[2024-06-15 19:56:29] debug: 	zh:controller: Received payload: clusterID=6, address=55196, groupID=0, endpoint=6, destinationEndpoint=1, wasBroadcast=false, linkQuality=142, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,**"**transactionSequenceNumber":60**,**"commandIdentifier":253},"payload":{"value":0,"data":{"type":"Buffer","data":[]}},"command":{"ID":253,"parameters":[{"name":"value","type":32},{"name":"data","type":1008}],"name":"tuyaAction"}}
[2024-06-15 19:56:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-06-15 19:56:29] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-06-15 19:56:29] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-06-15 19:56:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-15 19:56:29] debug: 	z2m: Received Zigbee message from 'testswitch', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{**"data":[],**"type":"Buffer"},"value":0}' from endpoint 6 with groupID 0
[2024-06-15 19:56:29] debug: 	z2m: Received Zigbee message from 'testswitch', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 6 with groupID 0

Koenkk, 2024-06-20:
> and the sequence number of repeated packets is the same.

That's good right? We can just filter on the sequence number to detected repeated packages.

Could you provide the data/database.db entry of your device?

1020844469, 2024-06-21:
> > and the sequence number of repeated packets is the same.并且重复数据包的序列号相同。
> 
> That's good right? We can just filter on the sequence number to detected repeated packages.这很好，对吧？我们可以只过滤序列号以检测到重复的包。
> 
> Could you provide the data/database.db entry of your device?您能否提供设备的数据/database.db条目？

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0029e8a58f","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1718908229497}
{"id":2,"type":"EndDevice","ieeeAddr":"0x84fd27fffe8d2974","nwkAddr":8720,"manufId":4098,"manufName":"_TZ3000_wkai4ga5","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"锟絓u0003\u0016-\u0011\u001a>E-\u0012\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0011","65506":31,"65508":1,"65534":0,"modelId":"TS0044","manufacturerName":"_TZ3000_wkai4ga5","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":66,"powerSource":3}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1718949676194}
{"id":3,"type":"Router","ieeeAddr":"0x0c4314fffe08ba52","nwkAddr":23211,"manufId":4098,"manufName":"_TZ3000_dfl9kueg","powerSource":"Mains (single phase)","modelId":"TS0726","epList":[1,2,3,4,5,6],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65503":"@k锟?-\u0011","65506":31,"65508":0,"65534":0,"modelId":"TS0726","manufacturerName":"_TZ3000_dfl9kueg","zclVersion":3,"appVersion":69,"powerSource":1}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53280":0,"powerOnBehavior":2,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8a58f","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53280":1,"powerOnBehavior":2,"switchMode":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8a58f","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53280":1,"powerOnBehavior":2,"switchMode":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8a58f","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"epId":4,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53280":1,"powerOnBehavior":2,"switchMo [... cut ...] ificTuya_3":{"attributes":{"53280":1,"powerOnBehavior":0,"switchMode":1}}},"binds":[],"configuredReportings":[],"meta":{}},"6":{"epId":6,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53280":1,"powerOnBehavior":0,"switchMode":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"zclVersion":3,"interviewCompleted":true,"meta":{"configured":889996980},"lastSeen":1718950854502}

Koenkk, 2024-06-22:
Could you provide the debug log when pressing the button once which outputs to MQTT actions?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

1020844469, 2024-06-27:
> Could you provide the debug log when pressing the button once which outputs to MQTT actions?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

[2024-05-03 13:17:41] debug: z2m: Received Zigbee message from 'MY_BUTTON', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 2 with groupID 0
[2024-05-03 13:17:41] debug: z2m: Received Zigbee message from 'MY_BUTTON', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 2 with groupID 0
[2024-05-03 13:17:41] debug: z2m: MQTT publish: topic 'zigbee2mqtt2/MY_BUTTON/action', payload 'scene_2'
[2024-05-03 13:17:41] debug: z2m: MQTT publish: topic 'zigbee2mqtt2/MY_BUTTON/action', payload 'scene_2'

Koenkk, 2024-06-27:
I would expect multiple `incomingMsg` here. Make sure to start capturing the debug log before you click the button until you see the multiple events.

1020844469, 2024-07-19:
> I would expect multiple `incomingMsg` here. Make sure to start capturing the debug log before you click the button until you see the multiple events.

**I pressed the 5th button of a wireless switch 3 times, but sent mqtt messages 4 times:**
[2024-07-19 16:33:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/二楼工作室开关电源/action', payload 'scene_5'
[2024-07-19 16:33:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/二楼工作室开关电源/action', payload 'scene_5'
[2024-07-19 16:33:12] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/二楼工作室开关电源/action', payload 'scene_5'
[2024-07-19 16:33:12] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/二楼工作室开关电源/action', payload 'scene_5'

**This is the incomingMsg:**
_You can see that the last two messages are [1,42,253,0]. This "42" is the sequence number of the message. Tuya is used to identify duplicate messages and should be filtered. If not filtered, you will receive 2 duplicate messages. news_
[2024-07-19 16:33:09] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":32543,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":14737765,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,40,253,0]}}
[2024-07-19 16:33:10] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":32543,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":112,"securityuse":0,"timestamp":14820669,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,41,253,0]}}
[2024-07-19 16:33:12] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":32543,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":120,"securityuse":0,"timestamp":14901944,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,42,253,0]}}
[2024-07-19 16:33:12] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":32543,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":156,"securityuse":0,"timestamp":14910961,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,42,253,0]}}

**Complete log ( "srcaddr":32543, "srcendpoint":5,) :**
[2024-07-19 16:32:58] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,224,54,0,84,254,35,68,129,0,0,0,0,224,54,1,1,0,160,0,240,233,213,0,0,15,8,196,10,1,0,32,91,226,255,32,48,228,255,32,0,224,54,29,18]
[2024-07-19 16:32:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,224,54,0,84,254,35,68,129,0,0,0,0,224,54,1,1,0,160,0,240,233,213,0,0,15,8,196,10,1,0,32,91,226,255,32,48,228,255,32,0,224,54,29,18]
[2024-07-19 16:32:58] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [224,54,0] - 84
[2024-07-19 16:32:58] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":14048,"relaycount":0,"relaylist":[]}
[2024-07-19 16:32:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,35,68,129,0,0,0,0,224,54,1,1,0,160,0,240,233,213,0,0,15,8,196,10,1,0,32,91,226,255,32,48,228,255,32,0,224,54,29,18]
[2024-07-19 16:32:58] debug: 	zh:zstack:unpi:parser: --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,224,54,1,1,0,160,0,240,233,213,0,0,15,8,196,10,1,0,32,91,226,255,32,48,228,255,32,0,224,54,29] - 18
[2024-07-19 16:32:58] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":14048,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":160,"securityuse":0,"timestamp":14019056,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,196,10,1,0,32,91,226,255,32,48,228,255,32,0]}}
[2024-07-19 16:32:58] debug: 	zh:controller: Received payload: clusterID=0, address=14048, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=160, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":196,"commandIdentifier":10},"payload":[{"attrId":1,"dataType":32,"attrData":91},{"attrId":65506,"dataType":32,"attrData":48},{"attrId":65508,"dataType":32,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-07-19 16:32:58] debug: 	zh:controller:endpoint: ZCL command 0xa4c13824ee0f4daf/1 genBasic.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":196,"writeUndiv":false})
[2024-07-19 16:32:58] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c13824ee0f4daf:14048/1 (0,0,3197)
[2024-07-19 16:32:58] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":14048,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":128,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,196,11,10,0]}}
[2024-07-19 16:32:58] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,224,54,1,1,0,0,128,0,30,5,24,196,11,10,0,186]
[2024-07-19 16:32:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-19 16:32:58] debug: 	z2m: Received Zigbee message from '二楼影音室右上射灯1', type 'attributeReport', cluster 'genBasic', data '{"65506":48,"65508":0,"appVersion":91}' from endpoint 1 with groupID 0
[2024-07-19 16:32:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt2/二楼影音室右上射灯1', payload '{"brightness":127.5,"color":{"h":null,"hue":null},"color_mode":"color_temp","color_temp":153,"do_not_distu [... cut ...]
[2024-07-19 16:33:33] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,69,106,33,0,30,49]
[2024-07-19 16:33:33] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,69,0,33]
[2024-07-19 16:33:33] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,69,0,33]
[2024-07-19 16:33:33] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 69 - [0] - 33
[2024-07-19 16:33:33] debug: 	zh:zstack:znp: SRSP: <-- ZDO - extRouteDisc - {"status":0}
[2024-07-19 16:33:33] debug: 	zh:zstack:unpi:parser: --- parseNext []

Koenkk, 2024-07-21:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/4e1c3f21186739038b42632e6eefddda): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

1020844469, 2024-07-22:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/4e1c3f21186739038b42632e6eefddda):
>
> * save this as file next to `configuration.yaml` as [...]

@Koenkk 

Hello, the issue is not fixed. I have tested 4 different tuya wireless switches (Zigbee2MQTT version1.39.0 commit: 0326926), and the problem is the same. The bug found in z2m is described as follows:

In ext.js:
﻿
if (utils.hasAlreadyProcessedMessage(msg, model, msg.data[0])) return; <--As I said above, the length of this msg.data is 0,-> "data":[], so it cannot be used Judgment is as follows:
z2m: Received Zigbee message from 'xxx switch', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0 }' from endpoint 2 with groupID 0
﻿
But in actual debugging, there is data ->[1,40,253,0]:
debug: zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":32543,"srcendpoint":5,"dstendpoint":1,"wasbroadcast ":0,"linkquality":105,"securityuse":0,"timestamp":14737765,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data" :[1,40,253,0]}}

1020844469, 2024-07-22:
The issue has been fixed, thank you.

Koenkk, 2024-07-23:
Integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-23 by Koenkk