Koenkk/zigbee2mqtt #22418: Cant start z2m  after upgrade to version: 1.37.0-1

mickt34, 2024-05-03:
### What happened?

Upgraded to 1.37.0-1 nothing else has been changed for months

### What did you expect to happen?

To start correctly after the upgrade.

### How to reproduce it (minimal and precise)

Just upgraded thats all

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

 Core 2024.5.0 Supervisor 2024.04.4 Operating System 12.2 Frontend 20240501.0

### Adapter

/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

### Setup

Add-on via hacs on raspi 8gb ssd disk

### Debug log

```
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [20,0,0,0] - 67
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":20}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SREQ: --> SYS - nvRead - {"sysid":1,"itemid":4,"subid":199,"offset":0,"len":20}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:writer: --> frame [254,8,33,51,1,4,0,199,0,0,0,20,204]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: <-- [254,22,97,51,0,20,97,10,0,0,178,0,0,0,43,0,27,254,255,129,246,140,2,0,9,0,72]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,22,97,51,0,20,97,10,0,0,178,0,0,0,43,0,27,254,255,129,246,140,2,0,9,0,72]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 22 - 3 - 1 - 51 - [0,20,97,10,0,0,178,0,0,0,43,0,27,254,255,129,246,140,2,0,9,0] - 72
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvRead - {"status":0,"len":20,"value":{"type":"Buffer","data":[97,10,0,0,178,0,0,0,43,0,27,254,255,129,246,140,2,0,9,0]}}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SREQ: --> SYS - nvLength - {"sysid":1,"itemid":4,"subid":200}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:writer: --> frame [254,5,33,50,1,4,0,200,0,219]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: <-- [254,4,97,50,0,0,0,0,87]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,0,0,0,0,87]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [0,0,0,0] - 87
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":0}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-03 08:45:58] debug: 	zh:zstack:backup: fetched adapter tclk table (capacity=20, used=20)
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SREQ: --> SYS - nvLength - {"sysid":1,"itemid":7,"subid":0}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:writer: --> frame [254,5,33,50,1,7,0,0,0,16]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: <-- [254,4,97,50,12,0,0,0,91]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,12,0,0,0,91]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [12,0,0,0] - 91
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":12}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SREQ: --> SYS - nvRead - {"sysid":1,"itemid":7,"subid":0,"offset":0,"len":12}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:writer: --> frame [254,8,33,51,1,7,0,0,0,0,0,12,16]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: <-- [254,14,97,51,0,12,16,44,77,0,163,18,227,38,0,75,18,0,12]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,14,97,51,0,12,16,44,77,0,163,18,227,38,0,75,18,0,12]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 14 - 3 - 1 - 51 - [0,12,16,44,77,0,163,18,227,38,0,75,18,0] - 12
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvRead - {"status":0,"len":12,"value":{"type":"Buffer","data":[16,44,77,0,163,18,227,38,0,75,18,0]}}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SREQ: --> SYS - nvLength - {"sysid":1,"itemid":7,"subid":1}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:writer: --> frame [254,5,33,50,1,7,0,1,0,17]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: <-- [254,4,97,50,12,0,0,0,91]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,12,0,0,0,91]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [12,0,0,0] - 91
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":12}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[... 38 lines cut ...]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,12,0,0,0,91]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [12,0,0,0] - 91
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":12}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SREQ: --> SYS - nvRead - {"sysid":1,"itemid":7,"subid":4,"offset":0,"len":12}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:writer: --> frame [254,8,33,51,1,7,0,4,0,0,0,12,20]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: <-- [254,14,97,51,0,12,0,0,0,0,0,0,0,0,0,0,0,0,80]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,14,97,51,0,12,0,0,0,0,0,0,0,0,0,0,0,0,80]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 14 - 3 - 1 - 51 - [0,12,0,0,0,0,0,0,0,0,0,0,0,0] - 80
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvRead - {"status":0,"len":12,"value":{"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,0]}}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SREQ: --> SYS - nvLength - {"sysid":1,"itemid":7,"subid":5}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:writer: --> frame [254,5,33,50,1,7,0,5,0,21]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: <-- [254,4,97,50,0,0,0,0,87]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,4,97,50,0,0,0,0,87]
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --> parsed 4 - 3 - 1 - 50 - [0,0,0,0] - 87
[2024-05-03 08:45:58] debug: 	zh:zstack:znp: SRSP: <-- SYS - nvLength - {"len":0}
[2024-05-03 08:45:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-03 08:45:58] debug: 	zh:zstack:backup: fetched adapter network security material table (capacity=1, used=1)
[2024-05-03 08:45:58] debug: 	zh:zstack:backup: Following devices with link key are missing from new backup but present in old backup and database, adding them back: 
[2024-05-03 08:45:58] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2024-05-03 08:45:58] info: 	zh:zstack:znp: closing
[2024-05-03 08:45:58] info: 	zh:zstack:znp: Port closed
[2024-05-03 08:45:58] info: 	z2m: Stopped zigbee-herdsman
```

oliversweb, 2024-05-03:
Same issue ... so reverted back to v1.36

oliversweb, 2024-05-03:
Found my issue .. the mqtt url contained a `/` before the port no. which is a malformed url ... so removed and then reinstalled 1.37 and all is ok so far 👍

evoncken, 2024-05-03:
After upgrade to 1.37.0, get “502 bad gateway”. Still on HA 2024.04 latest, zigbee2mqtt update was last change on system. Will provide more information as needed. Rolling back to 1.36 to see if HA recovers.

wbeymar, 2024-05-03:
Same problem here. I tried updating the firmware of the zbdongle-e, changing the adapter: ezsp to ember, but nothing happens, just that Z2M does not boot.

oliversweb, 2024-05-03:
Try going into add-ons/ZigBee2MQTT/configuration and check the mqtt server URL is not malformed in anyway, I think v1.37 is less forgiving with malformed URLs than 1.36

mickt34, 2024-05-03:
Looks OK
`
server: http://192.168.0.35:1883`

mickt34, 2024-05-03:
> Looks OK ` server: http://192.168.0.35:1883`

?

wbeymar, 2024-05-03:
server: mqtt://homeassistant.local/ Like this? 
How do I know which one is correct?

oliversweb, 2024-05-03:
Mine is like mqtt://homeassistant.local:1883 ... note, no '/' after host and before port no.

mickt34, 2024-05-03:
After several tries it appears that you cant use an IP address it has to be "homeassistant.local" I would have thought an IP address would be at least as good.

So im using mqtt://homeassistant.local:1883 and 1.37 works fine now.

WarC0zes, 2024-05-03:
> J'ai trouvé mon problème .. l'URL mqtt contenait un `/`avant le numéro de port. qui est une URL mal formée... donc supprimée puis réinstallée 1.37 et tout va bien jusqu'à présent 👍

Hi,
You have to read the doc:
[image]

jasperm22, 2024-05-07:
Same problem here, url seems fine

Koenkk, 2024-05-07:
Can you check if this is fixed in the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

wbeymar, 2024-05-07:
Finally I could find a solution and it was a dumb solutión. First, I upgraded the dongle-e firmware to the latest versión, later, upgraded Z2M. then, I changed, in Z2M configuration, the server from mqtt://homeassistant.local/ to mqtt://homeassistant.local:1883

Everything is working now

Koenkk, 2024-05-08:
@wbeymar does `mqtt://homeassistant.local` also work?

wbeymar, 2024-05-08:
@Koenkk Not for me. I had to write after local_:1883_. Just like mqtt://homeassistant.local does not work to me

closed 2024-12-05 by github-actions[bot]