Koenkk/zigbee2mqtt #22420: Aqara roller blinds cannot be controlled anymore

fir3drag0n, 2024-05-03:
### What happened?

I switched to ember and the device is not usable anymore

`2024-05-03 10:45:46Publish 'set' 'position' to 'Rollo-Balkon' failed: 'Error: ZCL command 0x54ef4410004d298f/1 genAnalogOutput.write({"presentValue":61}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":13,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":43})'`

### What did you expect to happen?

To control it regularly 

### Zigbee2MQTT version

11.37.0-dev commit: 3a61f0a

### Adapter firmware version

7.4.2 [GA]

### Adapter

SLZB-06M

### Setup

Z2M Docker

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15197839/log.log)

barrymossel, 2024-05-03:
Not sure if related, but my Aqara roller blinds can be controlled through the Z2M web interface, but are not working properly through HA. State is unknown, cannot test more (as I am not home ATM).

Koenkk, 2024-05-04:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/22387

closed 2024-05-04 by Koenkk

fir3drag0n, 2024-05-06:
This problem appeared again