Koenkk/zigbee2mqtt #22421: V1.37.0 Error (re)configuring Nedis Smoke Detector TS0205 in log after upgrade to V1.37.0

SchoutenHans, 2024-05-03:
### What happened?

After upgrade to V1.37.0 errors in logfile for Nedis Smoke Detectors, Three Reconfigure attempts after restart.
---------------------------------------------------------------------------------------------------------------------------
Z2M Server: Failed to configure 'Smoke-2', attempt 3 (Error: Bind 0x5c0272fffec63ad9/1 genPowerCfg from '0x00124b00239325b4/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\utils\waitress.ts:63:23) at C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\zStackAdapter.ts:691:28 at Queue.execute (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\utils\queue.ts:35:20) at Endpoint.bind (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:464:13) at setupAttributes (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\modernExtend.ts:75:13) at configure (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\modernExtend.ts:112:17) at Object.configure (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\index.ts:160:21) at Configure.configure (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\lib\extension\configure.ts:121:13) at Configure.onMQTTMessage (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\lib\extension\configure.ts:55:21) at EventEmitter.wrappedCallback (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\lib\eventBus.ts:174:17))
3-5-2024 11:08:22
ZigbeePlus
Error
Z2M Server: Failed to configure 'Smoke-2', attempt 2 (Error: ZCL command 0x5c0272fffec63ad9/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 11371 - 1 - 19 - 1 - 7 after 10000ms) at Object.start (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\utils\waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\zStackAdapter.ts:498:47) at Queue.execute (C:\Program Files (x86)\HomeSeer HS4\Data\ZigbeePlus\zigbee2mqtt\node_modules\zigbee-herdsman\src\utils\queue.ts:35:20) at Request.send

### What did you expect to happen?

No Errors

### How to reproduce it (minimal and precise)

Revert to previous version 1.35.1

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

C2652RB Sleash

### Setup

Windows

b2un0, 2024-07-22:
i had the same issue for 11 Smoke detectors.
a manually reconfigure or interview did not help (after pressing the button).

SchoutenHans, 2024-07-28:
@Koenkk,

I tried to upgrade from V 1.35.1 to V 1.39.0 to see if this problem is solved, but the errors are still there.
So I'm back to V 1.35.1.  again.  What happened since V 1.35.1 with the Nedis Smoke Detectors?
Can you solve this?

Regards,
Hans

SchoutenHans, 2024-07-29:
Attached the database.db file of the system running V 1.35.1 (With this version the Nedis Smoke Detectors don't generate the error)

[database.zip](https://github.com/user-attachments/files/16417085/database.zip)

b2un0, 2024-07-30:
here mine

```
cat /apps/zigbee2mqtt/data/database.db | grep TS0205 > /apps/zigbee2mqtt/data/database.cat.db
```

```json
{"id":20,"type":"EndDevice","ieeeAddr":"0xa4c138319dcb2cdc","nwkAddr":7904,"manufId":4417,"manufName":"_TZ3210_up3pngle","powerSource":"Battery","modelId":"TS0205","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,4,5,1,1280],"outClusterList":[3,4,6,4096,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0205","manufacturerName":"_TZ3210_up3pngle","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":1,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a2f51ad","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":180}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f51ad","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":65,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1722304118636}
{"id":32,"type":"EndDevice","ieeeAddr":"0x5c0272fffec04418","nwkAddr":54491,"manufId":0,"manufName":"_TYZB01_dsjszp0x","powerSource":"Battery","modelId":"TS0205","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0205","manufacturerName":"_TYZB01_dsjszp0x"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a2f51ad","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f51ad","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":64,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722327958454}
{"id":66,"type":"EndDevice","ieeeAddr":"0x0c4314fffef32a28","nwkAddr":20572,"manufId":0,"manufName":"_TYZB01_dsjszp0x","powerSource":"Battery","modelId":"TS0205","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,61185],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0205","manufacturerName":"_TYZB01_dsjszp0x","powerSource":3,"appVersion":66,"zclVersion":3}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a2f51ad","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f51ad","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722316061539}
{"id":71,"type":"EndDevice","ieeeAddr":"0x0c4314fffe7277b8","nwkAddr":25162,"manufId":0,"manufName":"_TYZB01_dsjszp0x","powerSource":"Battery","modelId":"TS0205","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,61185],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0205","manufacturerName":"_TYZB01_dsjszp0x","powerSource":3,"zclVersion":3,"appVersion":66}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a2f51ad","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f51ad","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722321794087}
{"id":72,"type":"EndDevice","ieeeAddr":"0x0c4314fffe6ad1da","nwkAddr":65447,"manufId":0,"manufName":"_TYZB01_dsjszp0x","powerSource":"Battery","modelId":"TS0205","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,61185],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0205","manufacturerName":"_TYZB01_dsjszp0x","appVersion":66,"powerSource":3,"zclVersion":3}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a2f51ad","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f51ad","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722327629925}
{"id":90,"type":"EndDevice","ieeeAddr":"0x50325ffffe16e838","nwkAddr":41059,"manufId":0,"manufName":"_TYZB01_dsjszp0x","powerSource":"Battery","modelId":"TS0205","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,61185],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0205","manufacturerName":"_TYZB01_dsjszp0x","appVersion":66,"powerSource":3,"zclVersion":3}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002a2f51ad","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f51ad","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":66,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1722320422279}
{"id":108,"type":"EndDevice","ieeeAddr":"0x50325ffffe0fb27a","nwkAddr":12681,"manufId":0,"manufName":"_TYZB01_dsjszp0x","powerSource":"Battery","modelId":"TS0205","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,61185],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0205","manufacturerName":"_TYZB0 [... cut ...] x00124b002a2f51ad","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2f51ad","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":65,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1722305615099}

Koenkk, 2024-07-31:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-31 by Koenkk