Koenkk/zigbee2mqtt #22424: Rename entity ID missing in version 1.37.0-1

YewTreeWeb, 2024-05-03:
### What happened?

When renaming a device in previous versions under friendly name was a toggle called 'Update Home Assistant entity ID'. However, since updating to 1.37.0-1 this option has disappeared along with the issue stated in Github issue [#22415](https://github.com/Koenkk/zigbee2mqtt/issues/22415).

[image]

### What did you expect to happen?

[image]
When renaming devices for an option to rename the id to be included

### How to reproduce it (minimal and precise)

On a none mobile display click rename device icon.

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

20240501.0

### Adapter

Combee II

### Setup

Home Assistant OS

TeHashX, 2024-05-03:
https://github.com/Koenkk/zigbee2mqtt/issues/22384#issuecomment-2091447905

tauka84, 2024-05-03:
I can't change friendly name. Missing button.
[image]

traqnski89, 2024-05-04:
Same here, the button is there but no longer an option to update the entity ID.
From Mobile app in portrait mode, the button is not shown at all, in landscape mode is there but still missing the update entity ID

Koenkk, 2024-05-04:
This should be fixed in the  [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

TeHashX, 2024-05-04:
This will be fixed in the next release, but until then use manual rename https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request
Go to Settings / Devices / MQTT / Configure and input
Topic:
`zigbee2mqtt/bridge/request/device/rename`
Payload:
`{"from": "old_name", "to": "new_name","homeassistant_rename":true}
`
Example:
`{"from": "0xa4c1383ffd1c89c1", "to": "Bath Weather", "homeassistant_rename":true}`
or
`{"from": "old_name", "to": "Bath Weather", "homeassistant_rename":true}`

closed 2024-11-01 by Koenkk