Koenkk/zigbee2mqtt #22425: Custom Cluster defintion using new OnEvent hook : Custom clusters are not loaded when z2m restarts, until one device is reconfigured or added

LaurentChardin, 2024-05-03:
### What happened?

The custom clusters can now be added directly from the device definition file in ZHC.
It uses the OnEvent hook to register the clusters when called.

However, when z2m is restarting, it does not reload those cluster definitions at start (when a device has been previously added). 

One device need to be reconfigured or added in order to reload again the custom clusters definitions.

### What did you expect to happen?

I would expect the custom clusters to be reloaded at launch when a device that needs them already exists.

### Zigbee2MQTT version

1.37

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SonOff Dongle

### Setup

Manual installation (github)

Koenkk, 2024-05-04:
Could you add a `console.log("LOADED DEFINITIONS")` to the `onEvent()` of that device, restart z2m and see when it is logged?

LaurentChardin, 2024-05-04:
So when i restart z2m, the clusters are actually added at launch.

`
[2024-05-04 08:20:04] debug:    z2m: Received MQTT message on 'zigbee2mqtt-yokis/bridge/groups' with data '[]'
[2024-05-04 08:20:04] debug:    z2m: Received MQTT message on 'zigbee2mqtt-yokis/bridge/extensions' with data '[]'
[2024-05-04 08:20:05] debug:    zhc:yokis: Custom Clusters for MTR500E-UP
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisDevice' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisInput' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisEntryConfigurator' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisSubSystem' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLoadManager' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLightControl' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisStats' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Custom Clusters for MTR500E-UP
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisDevice' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisInput' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisEntryConfigurator' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisSubSystem' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLoadManager' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLightControl' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisStats' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Custom Clusters for MTV300E-UP
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisDevice' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisInput' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisEntryConfigurator' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisSubSystem' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLoadManager' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLightControl' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisDimmer' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisStats' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Custom Clusters for MVR500E-UP
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisDevice' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisInput' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisEntryConfigurator' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisSubSystem' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLightControl' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisWindowCovering' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisStats' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisDevice' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisInput' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisEntryConfigurator' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisSubSystem' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLoadManager' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLightControl' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisStats' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Custom Clusters for TLM1-UP
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisDevice' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisInput' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisLightControl' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisDimmer' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisWindowCovering' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Adding custom cluster 'manuSpecificYokisChannel' : [object Object]
[2024-05-04 08:20:05] debug:    zhc:yokis: Custom Clusters 'manuSpecificYokisChannel' already defined. Skipping
[... cut ...]
**But** they do not appear on the web UI:
[image]

Until a *reconfigure* is done, then they are properly shown:
[image]

**Another finding:**
If you look at the logs, the devices seems to be added in parallel and therefore custom clusters are added multiple times. Only the last device was actually benefiting for the cache. At startup, there might be a race condition on the cache building. I don't know if the OnEvent are executed in parallel, but they could eventually step on each other if this is the case on the cache entry.

Koenkk, 2024-05-04:
It's indeed a race condition, maybe we should tackle this a bit differently, instead of doing it in the `onEvent`, add a new property to `meta` called `customClusters`, which are then added synchronously during the z2m startup. 

CC: @sjorge @mrskycriper

sjorge, 2024-05-04:
It should be pretty trivial to just update the current modernExtend to write to the meta attribute.

LaurentChardin, 2024-05-05:
Another point besides the onEvent race condition is still about the situation that is happening in the UI.

I believe the frontend is basically listening the state in MQTT through WebSocket : therefore I suspect that at launch time, the initial state of the devices that is published (or updated) in the `bridge/devices` topic does not take into account yet the Custom Clusters.
And it seems that the devices are not updated in MQTT `bridge/devices` topic when the custom cluster cache is getting populated.

But only when we do a *reconfigure*, it actually updates the devices definition in the MQTT topic.

And this one is getting lost/refreshed when z2m is restarting.

We might want to either finish to populate the custom devices cache before updating the MQTT state at startup.

sjorge, 2024-05-05:
It should probably have the customClusters loaded before anything else is done with the device.As both fz and tz converters will most likely be using those added clusters. And MQTT/ws should not call them before they are usable.~ sjorge

LaurentChardin, 2024-05-05:
@sjorge I can confirm that the `bridge/devices` topic is always updated in MQTT when Z2M is starting (timestamp of the retained message is updated), and that it does not contains the custom clusters details. 
The publishDevices() method from the bridge.ts extension is firing *before* the custom clusters are actually processed.

Koenkk, 2024-05-09:
> Could we move the bridge extension loading at the end of the list eventually ?

I'm not sure if this solves it, could you check that?

LaurentChardin, 2024-05-10:
I can confirm that calling the **OnEvent** extension _before_ the **Bridge** extension properly show the right custom clusters names in the frontend UI, as the `bridge/devices` topic is properly updated at startup.

But there could be some side effects, as a device could need to interact with the event bus and some extensions would not be declared yet.

Koenkk, 2024-05-10:
@LaurentChardin could you check if it's OK now: https://github.com/Koenkk/zigbee2mqtt/commit/04e88baf6076defe9abe59c8ea88265bde807163 ?

LaurentChardin, 2024-05-10:
@Koenkk Yes this is the exact same change i did locally, and it works for me. But theoretically, i could create regressions at startup for some devices (for some people). 

If we move the custom clusters registration outside of the onEvent, then this fix is probably not necessary and could be reverted.

LaurentChardin, 2024-05-11:
Closing the issue : works for my use case ,but will keep a look if any side effects in the issue list.

Eventually will revert if we change the OnEvent hook !

closed 2024-05-11 by LaurentChardin

sjorge, 2024-05-11:
If this somehow turns out not to be sufficient, perhaps we can split out only the `start` event and have that called first and have the other events in the old position.

The modernExtend doesn't need any changes then either, on start existing devices will have the cluster pushed by the start event before everything else. And new device will have it pushed during the interview event.