Koenkk/zigbee2mqtt #22429: bituotechnik-spm01-u01 not reporting right value

Patricen1, 2024-05-03:
### What happened?

It looks like the following device
https://www.zigbee2mqtt.io/devices/SPM01-U01.html#bituo%2520technik-spm01-u01
is not reporting the right apparent power, maybe a 10expX factor issue.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Raspberry pi 3

Patricen1, 2024-05-04:
[image]

On the above screenshot, voltage is 236.19V, current is 0.19A, apparent power should be 44.876VA instead of 0.047VA

jymannob, 2024-05-18:
Same issue with Z2M _7.3.1.0 build 0_

sdalu, 2024-05-28:
Slightly different issue for me, voltage and current are not correctly reported.
And it appears that energy is not updated

Z2M version: 1.37.1 
SPM01 firmware: 003.00.03

[image]

sdalu, 2024-05-29:
I don't know if it is related, but I have the following errors reported in my logs:
~~~
[2024-05-29 18:01:43] error:    z2m: Failed to configure 'electivire', attempt 3 (Error: ZCL command 0xbc026efffea7f998/1 seMetering.
configReport([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], 
{"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"rese
rvedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')
    at Endpoint.checkStatus (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28)
    at Endpoint.zclCommand (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26)
    at Endpoint.configureReporting (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:5
56:9)
    at Object.currentSummDelivered (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:13
1:5)
    at Object.configure (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/bituo_technik.ts:21:13
)
    at Configure.configure (/home/home-automation/zigbee2mqtt/lib/extension/configure.ts:121:13)
    at EventEmitter.wrappedCallback (/home/home-automation/zigbee2mqtt/lib/eventBus.ts:174:17))
~~~

script0803, 2024-05-31:
In my test, the energy and produced energy seem to be reported normally? It's just that when it is initially connected, it will only report when the data changes, otherwise it will show null.
You can try to reconfigure the device to refresh.
[image]

sdalu, 2024-05-31:
@script0803, I'm always drawing at least 140W, so energy data should change quite enough

Performing a reconfigure (yellow button) lead to this error message:
~~~
error 2024-05-31 09:23:07Failed to configure 'electivire', attempt 5 (Error: ZCL command 0xbc026efffea7f998/1 seMetering.configReport([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE') at Endpoint.checkStatus (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28) at Endpoint.zclCommand (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26) at Endpoint.configureReporting (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:556:9) at Object.currentSummDelivered (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:131:5) at Object.configure (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/bituo_technik.ts:21:13) at Configure.configure (/home/home-automation/zigbee2mqtt/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/home/home-automation/zigbee2mqtt/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/home/home-automation/zigbee2mqtt/lib/eventBus.ts:174:17))
~~~

Also, every 2 days the device stops working, despite a fixed green light on the LED, which seems to mean according to the documentation that everything is ok. I need to press the button 5 to 10 seconds to have it working again.
~~~
[2024-05-31 07:04:30] warning:  z2m: Failed to ping 'electivire' (attempt 1/1, ZCL command 0xbc026efffea7f998/1 genBasic.read(["zclVersion"], {"timeout":10000,"dis
ableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequ
~~~

Patricen1, 2024-06-06:
Hello,

I mentioned that every time I do restart Z2M, this device is recon

Patricen1, 2024-06-26:
> In my test, the energy and produced energy seem to be reported normally? It's just that when it is initially connected, it will only report when the data changes, otherwise it will show null. You can try to reconfigure the device to refresh. <img alt="image" width="166" src="https://private-user-images.githubusercontent.com/116477970/335419451-8dbb8d9a-e280-4133-8f03-954dd5e6a0e5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkzODIyNjcsIm5iZiI6MTcxOTM4MTk2NywicGF0aCI6Ii8xMTY0Nzc5NzAvMzM1NDE5NDUxLThkYmI4ZDlhLWUyODAtNDEzMy04ZjAzLTk1NGRkNWU2YTBlNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyNlQwNjA2MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MDlmNGU0YTVlYzMyNTRlMmYyZTlkODM2NmQ1MzhlNmU1MzQ5NGFiZWRhNDgzOGNiYWNjOWMwZDkwNGU1ZDFkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ceM_tYEtNvUgOS6BfwQPrftaoj0dv296XGoDbZe3ZUQ">

Hello,
It looks nevertheless that apparent power calculation is wrong with a factor of 1,000.

sdalu, 2024-07-07:
I tried the 13.9.0 version.  As mentioned by @Patricen1 I've got the reconfiguration error:

~~~
[2024-07-07 14:00:32] error:    z2m: Failed to configure 'electivire-old', attempt 3 (Error: ZCL command 0xbc026efffea7f998/1 seMe
tering.configReport([{"attribute":"currentSummDelivered","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange"
:0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":n
ull,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')
    at Endpoint.checkStatus (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:316:2
8)
    at Endpoint.zclCommand (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:771:26
)
    at Endpoint.configureReporting (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.t
s:557:9)
    at Object.currentSummDelivered (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts
:128:5)
    at Object.configure (/home/home-automation/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/bituo_technik.ts:31
:13)
    at Configure.configure (/home/home-automation/zigbee2mqtt/lib/extension/configure.ts:126:13)
    at EventEmitter.wrappedCallback (/home/home-automation/zigbee2mqtt/lib/eventBus.ts:174:17))

~~~

Also values reported are incorrect, frequency got a x100 applied, voltage got a x100 applied, current got a x100 applied, power reactive is stuck to 0.

[image]

Patricen1, 2024-07-15:
Seems that this module has been updated to report energy usage with V1.39, but the 1000 factor for power value issue is still present.

Patricen1, 2024-09-04:
> @script0803 could you maybe take a look at this?

Hello @script0803, do you mention the issue we are facing?
It looks like some of the reported value needs to be extended by 1000.

Patricen1, 2024-11-02:
up...

script0803, 2024-11-04:
Hi, @Patricen1 
Sorry for the late reply.
Is the problem you encountered that the apperant power and reactive power are wrongly divided by 1000? 

Does active power have this problem?

 I remember that I adjusted the power divisor to 1 in the converter. 

In theory, both apperant power and reactive power should be calculated using this factor.

[image]

Patricen1, 2024-11-04:
> ![Screenshot_20240504-090130](https://private-user-images.githubusercontent.com/34392672/327930951-0ab5fe41-1b3c-4dd3-98eb-743f5e3b8fda.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA2OTk1NzcsIm5iZiI6MTczMDY5OTI3NywicGF0aCI6Ii8zNDM5MjY3Mi8zMjc5MzA5NTEtMGFiNWZlNDEtMWIzYy00ZGQzLTk4ZWItNzQzZjVlM2I4ZmRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA0VDA1NDc1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc4NjcxMDBmOGM2MTA0ODE4ZDZhNGFlMTUxOTZkZjI4ZGVjYmIxMjdmOGIxM2JjZmY5NzU1NDA2ZGMyNTdjOTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.xhE_PhpUl5XJQTJHlMy38oUhAsbSu4L_kIwtfyI4YKA)
> 
> On the above screenshot, voltage is 236.19V, current is 0.19A, apparent power should be 44.876VA instead of 0.047VA

Hello,
To my opinion all powers (active, réactive and power) should be extended by 1000 according to the above screenshot. Some people say energy is ok but i could not verify. Is there a way to test a sample code?

Patricen1, 2024-11-13:
Hello @script0803 ,

Did you find anything relevant?

Patricen1, 2024-12-02:
Do you think it workaround can be found on next release?
ie add a factor within ste "settings (specific) page"
[image]

script0803, 2024-12-18:
Hi, @Patricen1 
Sorry for the long wait, I have submitted a pull request, but the manager has not approved it yet. You can use this external converter first.

SPM01
```javascript
const {electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const {
    precisionRound, postfixWithEndpointName,
} = require('zigbee-herdsman-converters/lib/utils');
const utils = require('zigbee-herdsman-converters/lib/utils');

const bituo_fz = {
    electrical_measurement: {
        /**
         * When using this converter also add the following to the configure method of the device:
         * await readEletricalMeasurementConverterAttributes(endpoint);
         */
        cluster: 'haElectricalMeasurement',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (utils.hasAlreadyProcessedMessage(msg, model)) return;
            const getFactor = (key) => {
                const multiplier = msg.endpoint.getClusterAttributeValue('haElectricalMeasurement', `${key}Multiplier`);
                const divisor = msg.endpoint.getClusterAttributeValue('haElectricalMeasurement', `${key}Divisor`);
                const factor = multiplier && divisor ? multiplier / divisor : 1;
                return factor;
            };
    
            const lookup = [
                {key: 'rmsCurrent', name: 'current', factor: 'acCurrent'},
                {key: 'rmsCurrentPhB', name: 'current_phase_b', factor: 'acCurrent'},
                {key: 'rmsCurrentPhC', name: 'current_phase_c', factor: 'acCurrent'},
                {key: 'rmsVoltage', name: 'voltage', factor: 'acVoltage'},
                {key: 'rmsVoltagePhB', name: 'voltage_phase_b', factor: 'acVoltage'},
                {key: 'rmsVoltagePhC', name: 'voltage_phase_c', factor: 'acVoltage'},
                {key: 'acFrequency', name: 'ac_frequency', factor: 'acFrequency'},
                {key: 'dcPower', name: 'power', factor: 'dcPower'},
                {key: 'dcCurrent', name: 'current', factor: 'dcCurrent'},
                {key: 'dcVoltage', name: 'voltage', factor: 'dcVoltage'},
            ];
            const powerLookup = [
                {key: 'activePower', name: 'power'},
                {key: 'activePowerPhB', name: 'power_phase_b'},
                {key: 'activePowerPhC', name: 'power_phase_c'},
                {key: 'apparentPower', name: 'power_apparent'},
                {key: 'apparentPowerPhB', name: 'power_apparent_phase_b'},
                {key: 'apparentPowerPhC', name: 'power_apparent_phase_c'},
                {key: 'reactivePower', name: 'power_reactive'},
                {key: 'reactivePowerPhB', name: 'power_reactive_phase_b'},
                {key: 'reactivePowerPhC', name: 'power_reactive_phase_c'},
                {key: 'totalActivePower', name: 'total_power'},
                {key: 'totalApparentPower', name: 'total_power_apparent'},
                {key: 'totalReactivePower', name: 'total_power_reactive'},
            ]
    
            const payload = {};
            for (const entry of lookup) {
                if (msg.data[entry.key] !== undefined) {
                    const factor = getFactor(entry.factor);
                    const property = postfixWithEndpointName(entry.name, msg, model, meta);
                    const value = msg.data[entry.key] * factor;
                    payload[property] = value;
                }
            }
            for (const entry of powerLookup) {
                if (msg.data[entry.key] !== undefined) {
[... 155 lines cut ...]
    extend: [electricityMeter({fzMetering: fz.metering, fzElectricalMeasurement: bituo_fz.electrical_measurement,
        acFrequency: true,
        powerFactor: true,
        configureReporting: false,
        producedEnergy: true,
        threePhase: true,
        power: {divisor: 1, multiplier: 1},
        })],
    meta: {},
    exposes: [
        e.power_reactive(),
            e.power_reactive_phase_b(),
            e.power_reactive_phase_c(),
            e.power_apparent(),
            e.power_apparent_phase_b(),
            e.power_apparent_phase_c(),
            e.power_factor_phase_b(), e.power_factor_phase_c(),
            e.numeric('total_power', ea.STATE).withUnit('W').withDescription('Total active power'),
            e.numeric('total_power_reactive', ea.STATE).withUnit('VAR').withDescription('Total active power'),
            e.numeric('total_power_apparent', ea.STATE).withUnit('VA').withDescription('Total active power'),
    ],
};

module.exports = definition;
```

Patricen1, 2025-01-04:
Hello,
Thanks for this update. I have never used external scripts, how should I proceed?

Best

Patricen1, 2025-02-01:
> Hello, Thanks for this update. I have never used external scripts, how should I proceed?
> 
> Best

@script0803 Can you please give me some clues?
@Koenkk is it possible to integrate the fix within next release ?

Koenkk, 2025-02-02:
@script0803 could you make a PR?

Patricen1, 2025-03-04:
Up...

closed 2025-05-11 by github-actions[bot]

Loic691, 2025-07-14:
Hi all
Another issue with this device. Have you got the same (bad) behaviour ?
When AC power off and on of the device, the consumed and produced energy glitch to 0...

[image]

[image]

This is a bad behaviour because utility meter using this device as source have in each AC power off/on a big error difference step (when switching from 0 to energy)...

Is there a workaround ? May I have to create a new issue ?

Otherwise, I'm searching a Z2M command to reset counter of this device ? Where is it hidden ?