Koenkk/zigbee2mqtt #22434: Z2M Frontend ist empty - Devices still connected and working

errormastern, 2024-05-03:
### What happened?

After a restart the frontend no longer shows any devices or configuration. 
In background all devices are still responding and reporting their state, etc..  

I updated the adapter short before to newest firmware, updated zigbee2mqtt and changed driver to ember. 
After that everything was fine, up an running like a charm. 

Things stopped working after machine was stopped unplanned because the power supply was temporarily unavailable

Frontend now looks like this:

[image]

[image]

Tried several times of restart, revert to previous z2m version. Nothing helped. In the logs I can't see anything strange - but I'm no expert. 

Any chance to get this back running?

### What did you expect to happen?

Frontend shows devices and configuration

### How to reproduce it (minimal and precise)

The machine was stopped unplanned because the power supply was temporarily unavailable.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.2.0

### Adapter

Sonoff Dongle E

### Setup

Homeassistant running on x86 with Z2M Integration

### Debug log

[45df7312_zigbee2mqtt_2024-05-03T20-26-22.371Z.log](https://github.com/Koenkk/zigbee2mqtt/files/15204795/45df7312_zigbee2mqtt_2024-05-03T20-26-22.371Z.log)

flyer331, 2024-05-04:
Hi !
I got the same problem 2 weeks ago.
Finally, I found a solution this morning after many tries like :

- data backup import
- delete mosquito and Z2M plugins and all directories
- modified many cache options in the Z2M configuration.yaml file 
- Sonoff dongle E firmware changes (I tried all FW from 7.x.x up to the latest)

and this morning I tried to delete all lines concerning devices only.

This action recreated database.db file but I lost all friendly names, so I imported the latest backup of configation.yaml (only lines of devices) and it worked.

Juste one tip, do not click here otherwise you will loose everything again ... -> 
[image]

If you got the latest firmware on the dongle (7.4.1), do not forget to change the adapter option : 
`serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_xxxxxxxxxxxxxx-yyyy
  adapter: ember`

Hope it will help you !

Damien.

errormastern, 2024-05-05:
Hey Damien,

Thanks for your response. It's great to have a solution approach.

If I understood you correctly, you've performed the following steps:

1. Removed all devices from the configuration.yaml
2. Restarted - DB gets regenerated
3. Added devices again from the backup of configuration.yaml
4. Restarted, and devices are back

Is that correct?

> Juste one tip, do not click here otherwise you will loose everything again ... ->

Damn, you're probably right with your tip. I might have actually clicked that button the other day :o/ The issue likely arose on the next restart.

BIGede, 2024-05-05:
I also gave my Dongel-E a new firmware and had exactly the same problem. Even after a downgrade to the old firmware, nothing has changed. 

I did not understand where the lines should be deleted so that the database is rebuilt. I don't want to break something like this with 90 devices :)

errormastern, 2024-05-06:
I'm so glad, I got it running again. 
Thanks @flyer331, I got onto it through your hint and solution approach.

The button in the German translation has the label "Bild lokalisieren" (Translate image), which led to all 55 devices receiving an entry in the configuration with their images as Base64 png. 

I removed the icon: entries from all devices in configuration.yaml. Upon the next start of zigbee2mqtt, everything was back to normal.

@BIGede 
If you haven't icon entries for all or many of your devices, it seems to be another problem. After Flashing and Update I'd first no problems..

BIGede, 2024-05-07:
No, I actually have exactly the same problem. So you have completely deleted the Icon: line and restarted Z2M?

[image]

errormastern, 2024-05-07:
Yes, exactly .. Delete all icon entries and restart.

BIGede, 2024-05-07:
It works. Thanks for the help.

closed 2024-05-14 by errormastern

hksthff, 2024-10-13:
Alternative: simply download your latest HA Backup and navigate to the file "homeassistant.tar.gz" > data > zigbee2mqtt > configuration.yaml -> copy & paste it in the file editor inside HA -> save & restart zigbee2mqtt