Koenkk/zigbee2mqtt #22436: Can't update IKEA E2001/E2002

Tymon3310, 2024-05-03:
### What happened?

Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)

And 
Unparsed frame Oxc4. Skipped

[image]

### What did you expect to happen?

Update succesed

### How to reproduce it (minimal and precise)

1. Update home assistant to 2024.5.1
2. Try to update IKEA E2001/E2002

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Addon, rpi 4

busywait, 2024-05-05:
I have been able to update an E2002. I have Z2M as an Addon in HA 2024.5.1 on a Pi 4 
Zigbee2MQTT version 1.37.0
Coordinator type zStack3x0
Coordinator revision 20230922

My coordinator is a USB Sonoff Dongle 3.0 Plus

I didn't see any logged errors during the update.

busywait, 2024-05-05:
FWIW, the May 2024 firmware for the E2002 does not seem helpful: the left and right arrow click, hold and release are no longer recognised.

jonm802, 2024-05-06:
I had this exact same error after updating one of my ikea remotes
the left and right arrows stopped working and i was seeing those debug messages

I fixed by force removing the remote and i re-paired it, i also switched to the ember driver so im not sure if it was a combination of both
using sonoff usb stick

hanzwo, 2024-05-12:
It doesn't work for me.

I have to  remotes, one with the 1.x firmware, the other with the 2.x firmware. The first one works fine after deleting the device and re-pair it. The second one re-pairs with no issues, but doesn't work at all. If you reconfigure the last one, you will get the errors described above.

closed 2024-12-28 by github-actions[bot]