Koenkk/zigbee2mqtt #22437: SonOff SNZB-01P issue

aurelien-ridet, 2024-05-03:
### What happened?

This is a followup of https://github.com/Koenkk/zigbee2mqtt/issues/22032

I think this issue is still not fixed as of 1.37.0.

I have 2 of thoses buttons and they still display a question mark on the Devices page and in the about page.
These devices also doesn't appear on the Dashboard page, I don't get why.

Reading related issue, it should have been fixed in zigbee-herdsman-converters 19.21.0 (https://github.com/Koenkk/zigbee-herdsman-converters/pull/7396), and z2m 1.37.0 is based on 19.32.0, so everything should add-up.

### What did you expect to happen?

I expected a battery icon with power level in Device page, and the device to appear in Dashboard page.

### How to reproduce it (minimal and precise)

Pair SonOff SNZB-01P with z2m version 1.36.1 and update to 1.37.0
Also tried to pair them again with z2m 1.37.0 without success (I didn't removed them before)

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

zStack3x0 (Sonoff-P)

### Setup

Add-on on Jeedom

DetectiveDarklight, 2024-05-03:
Ping @VikeDragon : Is it fixed for you with latest version ?

VikeDragon, 2024-05-10:
> Ping @VikeDragon : Is it fixed for you with latest version ?

Thanks for the correction! Everything is done perfectly.
[image]

DetectiveDarklight, 2024-05-12:
After pairing again in 1.37.1, I can't reproduce the issue anymore (I also force removed the device this time)

closed 2024-05-12 by DetectiveDarklight