Koenkk/zigbee2mqtt #22443: Repeating Javasript errors thrown after 1.37.0 update (HA 2024.5.0)

rinne, 2024-05-04:
### What happened?

After updating via HA UI zigbee2mqtt to 1.37.0 periodic Javascript errors started to appear in the log. This is obviously a bug, but I don't know what specifically triggers it, because I really can't debug it in my environment without getting the wrath of my wife onto me :).

In two separate installations, identical errors are spawn as a burst exactly every 5 minutes.  Each burst always include the same number of errors (6 in other and 7 in other) but I couldn't easily link that number to any specific amount of certain devices or number of routes visible in the map etc.

On the surface, the system seems to work though.

```
[2024-05-04 12:12:50] error: 	zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt16LE (node:internal/buffer:245:5)
    at Buffer.readUIntLE (node:internal/buffer:182:17)
    at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67)
    at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23
    at Array.every (<anonymous>)
    at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15)
    at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)
[2024-05-04 12:12:50] error: 	zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
```

### What did you expect to happen?

Not seeing Javascript exceptions thrown.

### How to reproduce it (minimal and precise)

I just updated the 1.36 to 1.37.0

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff zbDongle-E

### Setup

Home Assistant under VirtualBox in Intel based mini-PC

rinne, 2024-05-04:
That particular error hasn't reappeared after I flashed 7.3.1 to the Zigbee stick.  The only remaining visible issue is weird log entry when enabling/disabling pairing.  I don't remember seeing that before:

```
[2024-05-04 13:14:23] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-04 13:14:23] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-04 13:14:53] info: 	z2m: Zigbee: allowing new devices to join.
[2024-05-04 13:14:53] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-04 13:14:55] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-04 13:14:55] error: 	zh:controller:greenpower: Received undefined command from '0'
```

Koenkk, 2024-05-05:
This probably also happened before, but pre 1.37.0 didn't log these errors. Since `ezsp` is deprecated, I recommend to switch to ember instead: https://github.com/Koenkk/zigbee2mqtt/discussions/21462

closed 2024-12-04 by github-actions[bot]