Koenkk/zigbee2mqtt #22444: V1.37.0: Frame changeSourceRouteHandler parsing error

uwen70, 2024-05-04:
### What happened?

After updating zigbee2mqtt to version 1.37.0 in HA I fond repeated frame parsing error in the log:

`[2024-05-04 11:50:24] error: 	zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt16LE (node:internal/buffer:245:5)
    at Buffer.readUIntLE (node:internal/buffer:182:17)
    at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67)
    at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23
    at Array.every (<anonymous>)
    at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15)
    at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)
[2024-05-04 11:50:24] error: 	zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
`

The error repeats itself sporadically (6 times in 24h).

### How to reproduce it (minimal and precise)

I don't know any way to provoke the error.

### Zigbee2MQTT version

1.37.0 commit: unknown

### Adapter firmware version

EZSP v8  6.10.3.0 build 297

### Adapter

ZB-GW04

### Setup

Add-on on Homeassistant on Debian, VM x86-64, VMware

Laguite14, 2024-05-04:
Hello,
I have exactly the same error in chain.... but Zigbee devices seems to work.

joaquinvacas, 2024-05-04:
Same here, my adapter (Tuya/LIDL converted LAN coordinator) doesn't have a compatible EmberZnet firmware AFAIK so I can't switch to the new driver.

Nerivec, 2024-05-04:
re https://github.com/Koenkk/zigbee2mqtt/issues/22443

baskient80, 2024-05-05:
same here... everything seems to work correctly though.

uwen70, 2024-05-05:
After updating Zigbee firmware from EZSP v8 6.10.3.0 build 297 to EZSP v13 7.4.2.0 the parsing error no longer occurred. 
An easy way to update is via WebFlasher (https://darkxst.github.io/silabs-firmware-builder/). It then also makes it possible to switch from the EZSP driver, which is now referred to as deprecated, to the new Ember. Going back is no problem either.

Laguite14, 2024-05-05:
> After updating Zigbee firmware from EZSP v8 6.10.3.0 build 297 to EZSP v13 7.4.2.0 the parsing error no longer occurred. An easy way to update is via WebFlasher [...]

Thanks, I think I saw that when flashing the new firmware on the coordinator you should have to redo all the pairing of the devices. Did you get this issue after flashing your dongle or Z2MQTT started again with all the devices ? 
Want to be sure before flashing.

Nerivec, 2024-05-05:
@Laguite14 Most people reported having no issues with upgrading from 6.10.3 to 7.x.x, but a few had to re-pair devices.

thombec, 2024-05-05:
today I flashed my Dongle-E from 6.10.3.0 to the current version. I have >60 devices already paired. 

All devices are paired after flashing. I have no issues with the upgrade - no dev was "lost".

Laguite14, 2024-05-06:
Thanks for your feedback, I will flash mine to see if it resolves Z2M issues.

sergeyshaklein, 2024-05-06:
Hello
After updating the firmware the errors disappeared. All devices have been preserved.

Ultraboss77, 2024-05-06:
And for users who does not have a compliant key ? Eg chip efr32mg13
What do you recommand @Koenkk for those users ? 😟

joaquinvacas, 2024-05-06:
I'm on the same boat, using a 256k chip which doesn't support newer
versions, not even 6.10 (stable)

Nerivec, 2024-05-06:
Silabs is splitting support for those old chips (Series 0 & 1) in the next firmware release. darkxst mentioned that support will be dropped in new firmware releases because of this. You can continue to use `ezsp`, but you will remain on old software with no possibility for upgrade, and that will likely break as soon as devices supporting the latest Zigbee revision (R23) come out (6.10.x are already showing signs of trouble with newer devices). Best option is to migrate to a more recent adapter. With the MG24 now coming out, it should provide good options, first tests on these are very promising 😉

Ultraboss77, 2024-05-06:
What about Zstack chips from TI ? They will be compliant with Zigbee R23, will be upgradable ?
If I do buy an other usb coordinator, which driver should I choose to be "future" proof ??

MattWestb, 2024-05-08:
EMx and EFR32MG1X is only supported until 6.10.X and EFR32MG2X is still full supported so its only splitting the code then its no use implanting functions that cant being used in the hardware.
Some MG32MG1 chips is still getting Thread / RCP firmware (EZSP 4.4.x is working OK) like IKEAs first gen module that using WFR32MG1P and tuya snot that is using EFR32MG1B chip.

I cant see some use ditching V8 then its not much changes only some commands / rev and better keep the backward compatibility (ZHA is still having support from early V5.X).

djrm05, 2024-05-13:
curiously I have 2 instances running with 2 zb-gw03 and only one of those have this issue, prompting a lot of red errors in the gui even all devices seems to work properly.
Anyone know how to stop this error flodding the gui and the log? I have check the loggin options but "error" seems to be the lower loggin option.

uwen70, 2024-05-14:
> curiously I have 2 instances running with 2 zb-gw03 and only one of those have this issue

Probably one of the sticks has a newer firmware?
> Anyone know how to stop this error flodding the gui and the log?

Update the stick's firmware. https://github.com/Koenkk/zigbee2mqtt/issues/22444#issuecomment-2094788152
On zb-gw03 the latest version 7.4.3.0 works for me. With Sonoff-E you should try 7.4.1.0.
After that you can also switch to the ember driver.

djrm05, 2024-05-14:
@uwen70 , thanks I would say it was the same version as I flashed both together, but maybe.... 
where did you get the 7.4.3.0 for the zb-gw03?

uwen70, 2024-05-14:
@djrm05 Firmware: https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds
Check whether you have version 1.1 or 1.2 of zb-gw03

djrm05, 2024-05-14:
@uwen70  sorry but do you mean to use the existing for gw04? I don't see gw03 there

uwen70, 2024-05-14:
> @uwen70 sorry but do you mean to use the existing for gw04? I don't see gw03 there
Sorry, I mixed up the gw03 and gw04.

dev-2-4-h, 2024-05-22:
Same issue here. I can not switch to ember as I use the Sonoff Zigbee Bridge with tasmota and there is no update available.

pioneer-01, 2024-05-27:
> Same issue here. I can not switch to ember as I use the Sonoff Zigbee Bridge with tasmota and there is no update available.

here as well. Using Sonoff ZBBridge Zigbee Gateway (ZBBridge) with Tasmota

uwen70, 2024-06-01:
New firmware 7.4.3.0 for ZB-GW03 available: https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/ZigbeeBridge_SM-011-signed

didius62, 2024-06-07:
Hi all
I have the same error !

Z2m : 1.38.0
plugin : jeezigbee
coordinator : Elelabs ELR023 fw : 6.10.3-41

jaspweowaale, 2024-06-19:
Same issue, Unparsed frame 0xc4. Skipped

And this 

Error 2024-06-19 08:14:53Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)

fransyozef, 2024-06-20:
ah .. this explains why my frontend doesn't work anymore?

`
[2024-06-20 18:23:51] error: 	zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:86:11)
    at Buffer.readUInt16LE (node:internal/buffer:245:5)
    at Buffer.readUIntLE (node:internal/buffer:182:17)
    at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67)
    at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23
    at Array.every (<anonymous>)
    at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15)
    at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35)
`

jaspweowaale, 2024-06-21:
I'm not sure if helps but below are my USB's  

Im new to this and running both ZAH and Zigbee2Mqtt, just to make it more fun Zwave  I used to work with GPS modems and Use teraterm-4.106 and see the boot, but then crickets, all info and videos related to Version E and P 

Perhaps my error is not related but its the same error on different models of USB sticks, to me it sounds like it is not a firmware issue, when possible to replicate the problem with Sonoff USB's  

ttyACM0
/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240220163931-if00
Subsystem:
tty
Device path:
/dev/ttyACM0
ID:
/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240220163931-if00
Attributes:
DEVLINKS: >-
  /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240220163931-if00
  /dev/serial/by-path/platform-xhci-hcd.1-usb-0:2:1.0
  /dev/serial/by-path/platform-xhci-hcd.1-usbv2-0:2:1.0
DEVNAME: /dev/ttyACM0
DEVPATH: >-
  /devices/platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb3/3-2/3-2:1.0/tty/ttyACM0
ID_BUS: usb
ID_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
ID_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
ID_MODEL_ID: 55d4
ID_PATH: platform-xhci-hcd.1-usb-0:2:1.0
ID_PATH_TAG: platform-xhci-hcd_1-usb-0_2_1_0
ID_PATH_WITH_USB_REVISION: platform-xhci-hcd.1-usbv2-0:2:1.0
ID_REVISION: '0442'
ID_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240220163931
ID_SERIAL_SHORT: '20240220163931'
ID_TYPE: generic
ID_USB_DRIVER: cdc_acm
ID_USB_INTERFACES: ':020201:0a0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
ID_USB_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
ID_USB_MODEL_ID: 55d4
ID_USB_REVISION: '0442'
ID_USB_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240220163931
ID_USB_SERIAL_SHORT: '20240220163931'
ID_USB_TYPE: generic
ID_USB_VENDOR: ITEAD
ID_USB_VENDOR_ENC: ITEAD
ID_USB_VENDOR_ID: 1a86
ID_VENDOR: ITEAD
ID_VENDOR_ENC: ITEAD
ID_VENDOR_ID: 1a86
MAJOR: '166'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '1365461'

ttyACM1
/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240123115137-if00
Subsystem:
tty
Device path:
/dev/ttyACM1
ID:
/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240123115137-if00
Attributes:
DEVLINKS: >-
  /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240123115137-if00
  /dev/serial/by-path/platform-xhci-hcd.0-usb-0:2:1.0
  /dev/serial/by-path/platform-xhci-hcd.0-usbv2-0:2:1.0
DEVNAME: /dev/ttyACM1
DEVPATH: >-
  /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-2/1-2:1.0/tty/ttyACM1
ID_BUS: usb
ID_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
ID_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
ID_MODEL_ID: 55d4
ID_PATH: platform-xhci-hcd.0-usb-0:2:1.0
ID_PATH_TAG: platform-xhci-hcd_0-usb-0_2_1_0
ID_PATH_WITH_USB_REVISION: platform-xhci-hcd.0-usbv2-0:2:1.0
ID_REVISION: '0442'
ID_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240123115137
ID_SERIAL_SHORT: '20240123115137'
ID_TYPE: generic
ID_USB_DRIVER: cdc_acm
ID_USB_INTERFACES: ':020201:0a0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
ID_USB_MODEL_ENC: SONOFF\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus\x20V2
ID_USB_MODEL_ID: 55d4
ID_USB_REVISION: '0442'
ID_USB_SERIAL: ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240123115137
ID_USB_SERIAL_SHORT: '20240123115137'
ID_USB_TYPE: generic
ID_USB_VENDOR: ITEAD
ID_USB_VENDOR_ENC: ITEAD
ID_USB_VENDOR_ID: 1a86
ID_VENDOR: ITEAD
ID_VENDOR_ENC: ITEAD
ID_VENDOR_ID: 1a86
MAJOR: '166'
MINOR: '1'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '1420613'

uwen70, 2024-06-21:
I think the "changeSourceRouteHandler parsing error" occurs with all Zigbee USB sticks that work with an EFR32MGxx chip, provided the firmware of the stick is still version 6.x.x.x. An update to 7.4.x.x should fix the problem. However, there don't seem to be any more firmware updates for old chips like EFR32MG13P on Elelabs ELR023.
@jaspweowaale: Your ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2 are with EFR32MG21 and should be able to be updated via WebFlasher (https://darkxst.github.io/silabs-firmware-builder/).

uwen70, 2024-06-21:
> ah .. this explains why my frontend doesn't work anymore?
> 
> v 1.38.0-1

@fransyozef: No, I don't think so. Despite this error, my web frontend continued to work without any problems.
After updating z2m from 1.37.x to 1.38.0-1, my web frontend only worked after another restart.

jaspweowaale, 2024-06-22:
> I think the "changeSourceRouteHandler parsing error" occurs with all Zigbee USB sticks that work with an EFR32MGxx chip, provided the firmware of the stick is still version 6.x.x.x. An update to 7.4.x.x should fix the problem. However, there don't seem to be any more firmware updates for old chips like EFR32MG13P on Elelabs ELR023. @jaspweowaale: Your ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2 are ZBDongle-E with EFR32MG21 and should be able to be updated via WebFlasher (https://darkxst.github.io/silabs-firmware-builder/).

Yeps, testing with one of my spares  thanks for the suggestion
,  
[image]

uwen70, 2024-06-23:
@jaspweowaale: You should update to version 7.4.2.0 right away. The firmware (ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl) can be found at https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee. With this version you can also switch to the Ember driver in z2m.

joaquinvacas, 2024-06-23:
The issue with MG13 chips is that those do not support newer firmware
builds than 6.10.x, I think mainly because of the chip only allowing 256k
firmwares.

jaspweowaale, 2024-06-24:
> @jaspweowaale: You should update to version 7.4.2.0 right away. The firmware (ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl) can be found at https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee. With this version you can also switch to the new Ember driver in z2m.

Yah I did do the update and all my 4 devices,   what do you mean ember driver?  I will need to google a bit related to that I think

uwen70, 2024-06-24:
@joaquinvacas:  You can try the firmware here: https://github.com/darkxst/silabs-firmware-builder/tree/main
It should support Elelabs Zigbee USB Adapter ELU013 and Elelabs Zigbee Raspberry Pi Shield ELR023 (both based on EFR32MG13P). The file is at least smaller than 256kb.
I recommend the ncp-uart-hw-v7.4.2.0-elelabs-elx0x3-115200.gbl, which also supports the Ember driver in z2m.

uwen70, 2024-06-24:
> @jaspweowaale: ...what do you mean ember driver? ...

The new Ember driver has been available since z2m 1.37.0. The release notes state: _The ezsp driver is now deprecated in favor of the new ember driver. The ember driver will become the officially supported driver for EmberZNet adapters in the future._
In the z2m configuration, `ezsp `must be replaced by `ember `to change the driver. However, firmware 7.4.0.0 or higher is required.

didius62, 2024-06-24:
Hi all , 

the Elelabs ELR023 is very stable, i am very satisfied with this adaptor. 
I have 4 routers CC2530+CC2591 and 45 equipments , all is runing fine . 

change the adapter is a very big work ! 
To day what adapter is recommanded ? 

@uwen70 : do you have tested the ncp-uart-hw-v7.4.2.0-elelabs-elx0x3-115200.gbl ? is it possible to downgrade the fw withot any lose ? 

many thanks 

DD 

----- Mail original -----

uwen70, 2024-06-25:
@didius62: There is no compelling reason to update a running system with new firmware at this point in time. I think an update is quite tricky. I don't have an Elelabs adapter, but an EasyIot ZB-GW04. My first attempt at an update failed because I chose the faster baud rate of 230400 and I first had to learn how to access the adapter at this speed. However, you can install a different, even older firmware image. There is also firmware for OpenThread and for Zigbee and OpenThread at the same time. However, only the Zigbee firmware is supported under z2m.
Some adapters save some configuration variables in addition to the firmware image. Sometimes these have to be reinitialized after an update. I don't know if this is necessary for your adapter and how to do it.
I have used versions 6.10.x, 7.3.1.0, 7.4.2.0 and have been using version 7.4.3.0 for a few weeks.

jaspweowaale, 2024-06-25:
Yah that ember thing did not work for, swapped out ezsp in the config part of the addon , and made a restart and and crickets zigbee2mqtt never came back,  bit late in the evening to play with this so have look tomorrow

didius62, 2024-06-25:
Thank you . 
DD 

----- Mail original -----

jaspweowaale, 2024-06-26:
I did get a UZG-01 today and Im so early in my HA setup, that I wonder what is best,  drop the Sonoff's  use it as a coordinator or POE it on a different floor as a router, I see the [Berded Thinking ](https://www.youtube.com/@BeardedTinker)  he review it but also at some point suggest running more than one zigbee2mqtt 

I think I will open a treat on HA forum to get some general feedback,  but the ember thing did not work despite the updates made to Sonoff USB stick

didius62, 2024-06-26:
Hi 

Ok I think i will keep my Popp Elelabs key. 
i have a sonoff ZBDongle-E but never tested ! (For that i need to buy another Raspberry to avoid problem !) 

DD 

----- Mail original -----

SashaAV, 2024-07-18:
Hi!

Same error. Here is stack trace:

[image]

FloMeyer, 2024-08-19:
I get the same error:

```
error 2024-08-19 12:35:23zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:177:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:151:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:145:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:430:35)
error 2024-08-19 12:35:23zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
error 2024-08-19 12:35:23zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:177:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:151:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:145:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:430:35)
error 2024-08-19 12:35:23zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
error 2024-08-19 12:35:23zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:177:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:151:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:145:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:430:35)
error 2024-08-19 12:35:23zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
error 2024-08-19 12:35:23zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:177:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:151:23 at Array.every (<anonymous>) at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:145:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:430:35)
error 2024-08-19 12:35:23zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
```

jerome-73, 2024-08-28:
Hi 

I get the same error.
tested with Version 1.38.0, 1.39.0 and 1.39.1.
Error message : 
ezsp: Unparsed frame 0xc4. Skipped
Killed

Controller Sonoff ZB USB dongle-E.
 
It crash after some minutes 2-3.

I tried to unplug and replug the USB dongle. message 0xc4 appear again, but without crashing z2m. now running since 15 minutes without crash.

Hope this comment can help a little.
Regards

didius62, 2024-10-01:
Hi ,
Could it be possible that the cc2530 routers are responsible for these errors?

DD

Uschix2000, 2024-12-04:
Hi! I have the same Error with an Aeotec Zi-Stick [https://aeotec.com/aeotec-z-stick-zigbee/](url)

Is it possible to do a firmware update for this Stick?

`2024-12-03 20:54:44zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:54:44zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
2024-12-03 20:54:48zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:54:48zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
2024-12-03 20:54:55zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:54:55zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
2024-12-03 20:55:36zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:55:36zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
2024-12-03 20:55:46zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:55:46zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
2024-12-03 20:55:47zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:55:47zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
2024-12-03 20:55:48zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:55:48zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
2024-12-03 20:55:51zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:55:51zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped
2024-12-03 20:55:54zh:ezsp:ezsp: Frame changeSourceRouteHandler parsing error: RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
2024-12-03 20:55:54zh:ezsp:ezsp: Unparsed frame 0xc4. Skipped`

didius62, 2024-12-04:
Hi ,

Sorry i dont't known . 
I solved this problem by switching to a Sonoff Zbdongle-E key and also my routers with this same USB key (with a router fw)
and for 3 months everything is stable no more errors. i use Z2m 1.42.0.

Didier

Uschix2000, 2024-12-04:
I am using on my Homeassistant the Zi Stick.

[image]

Can somebody tell me if it is possible to update this stick to solve the issues?

uwen70, 2024-12-05:
@Uschix2000: In principle, the stick can be updated. But I didn't see Coordinator version 7.x (or 8.x) on the [manufacturer's website](https://aeotec.freshdesk.com/support/solutions/articles/6000270951-update-zi-stick-to-thread-or-zigbee-with-python-3-9-and-windows). 
However, there is an [updater](https://github.com/darkxst/silabs-firmware-builder) for various sticks that also supports the ZI. After that, you have to switch z2m to Amber Driver so that the error goes away.

closed 2025-02-04 by uwen70