Koenkk/zigbee2mqtt #22445: ZBDongle-E: ember driver, broadcast errors at startup

Ricc68, 2024-05-04:
### What happened?

During startup of Z2M the following broadcast errors happened:

[2024-05-04 14:36:53] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":184} messageTag=255]
[2024-05-04 14:36:53] error: 	zh:ember: Delivery of BROADCAST failed for "65532" [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":185} messageTag=1]

### What did you expect to happen?

No broadcast errors.

### How to reproduce it (minimal and precise)

This is a first installation, the Zigbee network at the moment is empty except the coordinator ZBDongle-E.
The errors happened during startup of Home Assistant OS host.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff ZBDongle-E

### Setup

Add-on on Home Assistant OS, host is a VM on x86-64, dongle uses ember driver

### Debug log

[14:35:58] INFO: Preparing to start...
[14:36:01] INFO: Socat not enabled
[14:36:07] INFO: Starting Zigbee2MQTT...
[2024-05-04 14:36:29] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-04 14:36:49] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-04 14:36:49] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-04 14:36:50] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-04 14:36:50] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-04 14:36:50] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-04 14:36:50] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-05-04 14:36:50] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-04 14:36:50] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-04 14:36:51] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-04 14:36:51] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-04 14:36:51] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-04 14:36:52] info: 	zh:ember: [STACK STATUS] Network up.
[2024-05-04 14:36:52] info: 	zh:ember: [INIT TC] NCP network matches config.
[2024-05-04 14:36:52] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1246ms until next broadcast.
[2024-05-04 14:36:52] info: 	z2m: zigbee-herdsman started (resumed)
[2024-05-04 14:36:52] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":2,"revision":"7.4.2 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-05-04 14:36:52] info: 	z2m: Currently 0 devices are joined:
[2024-05-04 14:36:52] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-04 14:36:52] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-05-04 14:36:53] info: 	z2m: Connected to MQTT server
[2024-05-04 14:36:53] info: 	z2m: Started frontend on port 8099
[2024-05-04 14:36:53] info: 	z2m: Zigbee2MQTT started!
[2024-05-04 14:36:53] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":184} messageTag=255]
[2024-05-04 14:36:53] error: 	zh:ember: Delivery of BROADCAST failed for "65532" [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":185} messageTag=1]

VladislavVesely, 2024-05-04:
Same setup, same error here

Nerivec, 2024-05-04:
Can you guys try to downgrade to 7.4.1 and see if it still happens?

Ricc68, 2024-05-05:
Closing this issue here as it is being discussed widely in the open issue #22453 so there's no point in keeping a duplicate.

closed 2024-05-05 by Ricc68