Koenkk/zigbee2mqtt #22446: Add installation code: Invalid code size

remidebette, 2024-05-04:
### What happened?

I have a Bosch Twinguard Zigbee device that I want to pair to z2m.
I am using the new `ember` adaptor with the Sonoff Dongle E

In the [documentation](https://www.zigbee2mqtt.io/devices/8750001213.html) I need to take a picture of the QR code on the device and Add an installation code before pairing.

My QR code: `RB01SG0D836591B3CC0010000000000000000000000D6F001910BCEDDLKDD7ED5CDAA8E2C708B67D2B1573DB684`

z2m refuses to pair with the Invalid Code size error.

### What did you expect to happen?

Continue to pairing

### How to reproduce it (minimal and precise)

With an Ember adaptor,

Settings > Tools > Add installation code 
RB01SG0D836591B3CC0010000000000000000000000D6F001910BCEDDLKDD7ED5CDAA8E2C708B67D2B1573DB684

The error will happen

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.2

### Adapter

Sonoff Zigbee Dongle E

### Setup

Home assistant OS on Raspberrypi 5

### Debug log

```
error: 	z2m: Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: '[ADD INSTALL CODE] Failed for "0x000D6F001910BCED"; invalid code size.'
```

Nerivec, 2024-05-04:
Looks like this code doesn't have the CRC appended to it. I tested it locally, it seems to still be accepted by the adapter, so I'll modify the code to bypass the check in this scenario. I will mention this issue in the PR when that's available in dev, if you can then confirm it works, that would be great. 😉

remidebette, 2024-05-05:
Thanks a lot, I will
I should install the dev version of z2m at that time?

Can you explain me what the CRC is? I am a bit of a newbie.

After the code is sucessfully accepted by the adapter, we might still get the issue described here: https://github.com/Koenkk/zigbee2mqtt/issues/20169 (the install code step works on the `ezsp` adapter but then the pairing of the Bosch Twinguard fails on Sonoff Dongle E) that I also experienced before migrating to ember. 

Could this lacking data be the rootcause of it all?

Nerivec, 2024-05-05:
It's an [error detection mechanism](https://en.wikipedia.org/wiki/Cyclic_redundancy_check) for data.
Actually, you should be able to use your code with current release, simply by adding the CRC to it.
If my math is right, that should be it:
`RB01SG0D836591B3CC0010000000000000000000000D6F001910BCEDDLKDD7ED5CDAA8E2C708B67D2B1573DB6843A5F`

Let me know if that works, and if you encounter the other problem you mentioned 😉

remidebette, 2024-05-05:
Thank you for the response,
I just followed the factory reset details described [here](https://community.bosch-smarthome.com/t5/technische-probleme/betreff-schwierigkeiten-mit-der-funktionalit%C3%A4t-von-bosch-twinguard-und-smart-controller-2/td-p/45649) (more thorough than explained in the z2m docs) 
And tried to add the device with the code you shared, but it is not going fine:

Maybe this belongs more to the other ticket, but I am not really sure what I could do from here

logs
```shell
[2024-05-05 14:58:31] info: 	z2m: Successfully added new install code
[2024-05-05 14:58:31] info: 	z2m: Zigbee: allowing new devices to join.
[2024-05-05 14:58:31] info: 	zh:ember: [STACK STATUS] Network opened.
[2024-05-05 14:58:48] info: 	zh:controller: Interview for '0x000d6f001910bced' started
[2024-05-05 14:58:48] info: 	z2m: Device '0x000d6f001910bced' joined
[2024-05-05 14:58:49] info: 	z2m: Starting interview of '0x000d6f001910bced'
[2024-05-05 14:58:57] warning: 	zhc:lixee: Failed to read zigbee attributes: Error: Cluster 'liXeePrivate' has no attribute 'motDEtat'
[2024-05-05 14:59:13] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "20262".
[2024-05-05 14:59:15] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "20262".
[2024-05-05 14:59:17] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "20262".
[2024-05-05 14:59:18] error: 	zh:controller: Interview failed for '0x000d6f001910bced with error 'Error: Interview failed because can not get node descriptor ('0x000d6f001910bced')'
[2024-05-05 14:59:18] error: 	z2m: Failed to interview '0x000d6f001910bced', device has not successfully been paired
[2024-05-05 14:59:22] info: 	zh:controller: Interview for '0x000d6f001910bced' started
[2024-05-05 14:59:22] info: 	z2m: Starting interview of '0x000d6f001910bced'
[2024-05-05 14:59:25] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "63572".
[2024-05-05 14:59:27] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "63572".
[2024-05-05 14:59:28] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "63572".
[2024-05-05 14:59:30] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "63572".
[2024-05-05 14:59:35] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "38021".
[2024-05-05 14:59:46] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "3342".
[2024-05-05 14:59:48] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "3342".
[2024-05-05 14:59:49] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "3342".
[2024-05-05 14:59:51] error: 	zh:controller: Interview failed for '0x000d6f001910bced with error 'Error: Interview failed because can not get node descriptor ('0x000d6f001910bced')'
[2024-05-05 14:59:51] error: 	z2m: Failed to interview '0x000d6f001910bced', device has not successfully been paired
[2024-05-05 14:59:55] info: 	zh:controller: Interview for '0x000d6f001910bced' started
[2024-05-05 14:59:55] info: 	z2m: Starting interview of '0x000d6f001910bced'
[2024-05-05 14:59:58] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "18900".
[2024-05-05 15:00:00] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "18900".
[2024-05-05 15:00:02] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "18900".
[2024-05-05 15:00:03] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "18900".
[2024-05-05 15:00:13] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "51606".
[2024-05-05 15:00:14] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "51606".
[2024-05-05 15:00:16] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "51606".
[2024-05-05 15:00:18] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "51606".
[2024-05-05 15:00:24] error: 	zh:controller: Interview failed for '0x000d6f001910bced with error 'Error: Interview failed because can not get node descriptor ('0x000d6f001910bced')'
[2024-05-05 15:00:24] error: 	z2m: Failed to interview '0x000d6f001910bced', device has not successfully been paired
[2024-05-05 15:00:28] info: 	zh:controller: Interview for '0x000d6f001910bced' started
[2024-05-05 15:00:28] info: 	z2m: Starting interview of '0x000d6f001910bced'
[2024-05-05 15:00:38] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "15596".
[2024-05-05 15:00:40] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "15596".
[2024-05-05 15:00:41] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "15596".
[2024-05-05 15:00:53] warning: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "15285".
```

Nerivec, 2024-05-05:
Indeed seems to be the same issue as that other thread.
Can you try switching to `debug` level for your logs, and send the log file after you tried to pair that device again?

remidebette, 2024-05-05:
I am not exactly sure where the logs start and stop, also most likely there are other zigbee devices in here but you can see one failure loop as follows:

```
[2024-05-05 23:12:26] debug: 	zh:ember:ezsp: <=== [FRAME: ID=36:"TRUST_CENTER_JOIN_HANDLER" Seq=116 Len=19]
[2024-05-05 23:12:26] debug: 	zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=26453], [newNodeEui64=0x000d6f001910bced], [status=STANDARD_SECURITY_UNSECURED_JOIN], [policyDecision=USE_PRECONFIGURED_KEY], [parentOfNewNodeId=18384]
[2024-05-05 23:12:26] debug: 	zh:controller: Device '0x000d6f001910bced' joined
[2024-05-05 23:12:26] debug: 	zh:controller: Device '0x000d6f001910bced' accepted by handler
[2024-05-05 23:12:26] debug: 	zh:controller: Device '0x000d6f001910bced' is already in database with different network address, updating network address
[2024-05-05 23:12:26] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-05-05 23:12:26] info: 	zh:controller: Interview for '0x000d6f001910bced' started
[2024-05-05 23:12:26] debug: 	zh:controller:device: Interview - start device '0x000d6f001910bced'
[2024-05-05 23:12:26] debug: 	zh:ember:queue: Status queue=0 priorityQueue=0.
[2024-05-05 23:12:26] info: 	z2m: Starting interview of '0x000d6f001910bced'
[2024-05-05 23:12:26] debug: 	zhc:lixee: zlinky config: historique, single_phase, false, BASE
[2024-05-05 23:12:26] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f001910bced","ieee_address":"0x000d6f001910bced","status":"started"},"type":"device_interview"}'
[2024-05-05 23:12:26] debug: 	zh:ember: ~~~> [ZDO NODE_DESCRIPTOR_REQUEST target=26453]
[2024-05-05 23:12:26] debug: 	zh:ember: ~~~> [ZDO UNICAST apsFrame={"profileId":0,"clusterId":2,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":0} messageTag=21]
[2024-05-05 23:12:26] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=117 Len=24]
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=5 frmRx=7]
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5]
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=7] Added to rxQueue
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0]
[2024-05-05 23:12:26] debug: 	zh:ember:ezsp: Skipping async callback while waiting for response to command.
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6]
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=0] Added to rxQueue
[2024-05-05 23:12:26] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1]
[2024-05-05 23:12:26] debug: 	zh:ember:ezsp: Skipping async callback while waiting for response to command.
[2024-05-05 23:12:26] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=117 Len=7]
[2024-05-05 23:12:26] debug: 	zh:ember: ~~~> [SENT ZDO type=DIRECT apsFrame={"profileId":0,"clusterId":2,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":254} messageTag=21 status=SUCCESS]
[2024-05-05 23:12:26] debug: 	zh:ember:ezsp: <=== [FRAME: ID=36:"TRUST_CENTER_JOIN_HANDLER" Seq=116 Len=19]
[2024-05-05 23:12:26] debug: 	zh:ember:ezsp: ezspTrustCenterJoinHandler(): callback called with: [newNodeId=26453], [newNodeEui64=0x000d6f001910bced], [status=STANDARD_SECURITY_UNSECURED_JOIN], [policyDecision=USE_PRECONFIGURED_KEY], [parentOfNewNodeId=18384]
[2024-05-05 23:12:26] debug: 	zh:controller: Device '0x000d6f001910bced' joined
[2024-05-05 23:12:26] debug: 	zh:controller: Device '0x000d6f001910bced' accepted by handler
[2024-05-05 23:12:26] debug: 	zh:controller: Not interviewing '0x000d6f001910bced', completed 'false', in progress 'true'
[2024-05-05 23:12:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-05 23:12:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6]
[2024-05-05 23:12:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=1] Added to rxQueue
[2024-05-05 23:12:29] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2]
[2024-05-05 23:12:29] debug: 	zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=117 Len=37]
[2024-05-05 23:12:29] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":1794,"sourceEndpoint":6,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":60}], [lastHopLqi=120], [lastHopRssi=186], [sender=39180], [bindingIndex=255], [addressIndex=255], [messageContents=18440a00002579c100000000]
[2024-05-05 23:12:29] debug: 	zh:controller: Received payload: clusterID=1794, address=39180, groupID=0, endpoint=6, destinationEndpoint=1, wasBroadcast=false, linkQuality=120, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":68,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":37,"attrData":[0,49529]}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[... cut ...]
[2024-05-05 23:12:55] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f001910bced","ieee_address":"0x000d6f001910bced","status":"failed"},"type":"device_interview"}'
[2024-05-05 23:12:56] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-05 23:12:56] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6]
[2024-05-05 23:12:56] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=4] Added to rxQueue
[2024-05-05 23:12:56] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5]
[2024-05-05 23:12:56] debug: 	zh:ember:ezsp: <=== [FRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=125 Len=37]
```

Nerivec, 2024-05-07:
Closing since the issue here (install code) should be fixed in dev/next release. Will reply in the aforementioned issue for the twinguard pairing troubles.

closed 2024-05-07 by Nerivec

remidebette, 2024-05-08:
So it should be fine providing only the code of the QR from now on?

Also do you need me to repost in the other issue?
Do you have enough info?

Nerivec, 2024-05-08:
> So it should be fine providing only the code of the QR from now on? (Without CRC)

After next release, it will be available yes, for now in dev.

It's okay, I couldn't find anything in your logs, it seems the device is just not replying to the interview request from Z2M. I requested more info from other users in the other issue.