Koenkk/zigbee2mqtt #22447: Ikea Parasoll and Vallhorn failed interview

Bruhildo, 2024-05-04:
### What happened?

Cannot get new Ikea sensors (Parasoll and Vallhorn) to connect to Z2M.

Tried many times with different level of success. The most I could get to was to pair once, but it got unresponsive after about a day. Sometimes I can get to recognize the device but does not pass the interview.

Using LADDA 750 batteries. Connecting very close to coordinator or router.

### What did you expect to happen?

Device should connect to Z2M with no issues and pass interview.

Other devices are able to pair without issues (including other older Ikea devices). Just happening to Parasoll and Vallhorn. 

### How to reproduce it (minimal and precise)

Pressed the config button on the device (Parasoll and Vallhorn) 4 times in 5 sec.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Home Assistant OS 2024.5.1

### Debug log

When trying to connect to the coordinator:
Info 2024-05-04 10:54:54Interview for '0x048727fffe5ff9ea' started
Info 2024-05-04 10:54:54Device '0x048727fffe5ff9ea' joined
Info 2024-05-04 10:54:54Starting interview of '0x048727fffe5ff9ea'
Warning 2024-05-04 10:55:14Device '0x048727fffe5ff9ea' left the network

When trying to connect to a router (Ikea Plug):
Info 2024-05-04 10:57:25Interview for '0x048727fffe5ff9ea' started
Info 2024-05-04 10:57:25Device '0x048727fffe5ff9ea' joined
Info 2024-05-04 10:57:25Starting interview of '0x048727fffe5ff9ea'
Error 2024-05-04 10:58:05Interview failed for '0x048727fffe5ff9ea with error 'Error: Interview failed because can not get node descriptor ('0x048727fffe5ff9ea')'
Error 2024-05-04 10:58:05Failed to interview '0x048727fffe5ff9ea', device has not successfully been paired
Error 2024-05-04 10:58:12Interview failed for '0x048727fffe5ff9ea with error 'Error: Interview failed because can not get node descriptor ('0x048727fffe5ff9ea')'
Error 2024-05-04 10:58:12Failed to interview '0x048727fffe5ff9ea', device has not successfully been paired
Info 2024-05-04 10:58:31Interview for '0x048727fffe5ff9ea' started
Info 2024-05-04 10:58:31Starting interview of '0x048727fffe5ff9ea'
Error 2024-05-04 10:58:59Interview failed for '0x048727fffe5ff9ea with error 'Error: Interview failed because can not get node descriptor ('0x048727fffe5ff9ea')'
Error 2024-05-04 10:58:59Failed to interview '0x048727fffe5ff9ea', device has not successfully been paired

Frank-1964, 2024-05-04:
Same issue here with 1.37

anergit, 2024-06-09:
I have the same issue, for all of my PARASOLL-devices.

```
2024-06-09 12:45:57 Interview failed for '0x3c2ef5fffe43a044 with error 'Error: Interview failed because can not get node descriptor ('0x3c2ef5fffe43a044')'
2024-06-09 12:45:57 Failed to interview '0x3c2ef5fffe43a044', device has not successfully been paired
```

Zigbee2MQTT v1.38

HomeAssistant:
Core 2024.5.5
Supervisor 2024.06.0
Operating System 11.4
Frontend 20240501.1

Dites33, 2024-10-30:
Same for me, any solution ? :(

Bruhildo, 2024-10-30:
Mine started to work after many attempts and has been rock solid since.

I purchased a new one and paired right away with no issues.

closed 2024-10-31 by Bruhildo

Majestic7979, 2025-03-29:
I paired almost all but one... There were two that were refusing to pair. I force deleted, then paired, it failed, then I just pressed the pair button four times without deleting and it paired successfully, but tried repeating the same process on the last one and it won't pair. I know it works because it was paired before I reset the network....