Koenkk/zigbee2mqtt #22449: HESZB-120 impossible to start siren/alarm

rafpigna, 2024-05-04:
### What happened?

I added this smoke and fire detector in my newtork.
[](https://www.zigbee2mqtt.io/devices/HESZB-120.html
)
The issue is that if I try to trigger the alarm start, clicking on the "start" button in the "values" dashboard, nothing happens.
I cant see any error in logs boht in the "log" in Z2M dashboard or in the addon registry.

Please note: If I put the detector near a heat source, or I press the test button on the device itself, the alarm starts loudly, so it's not defective or broken.

You can find the device in the log below as "firedetector"

### What did you expect to happen?

That the siren starts....

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

EZSP v12 - 7.3.1.0 build 176

### Adapter

Sonoff dongle-e

### Setup

Home Assistant OS addon, X86

### Debug log

[
[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15210217/log.log)
]

mamrai1, 2024-05-04:
[image]
Check that the sensor is on firmware 4.0.8

rafpigna, 2024-05-05:
> ! Check that the sensor is on firmware 4.0.8

You were right, thanks!
After upgrading firmware via OTA (that took at least 10 retries because was failing aroun 23% each time, now the alarm works fine both via the dashboard and also via automations, triggered by other sensors or switches. Thanks!

closed 2024-05-05 by rafpigna