Koenkk/zigbee2mqtt #22450: MiBoxer FUT039Z broken

ich777, 2024-05-04:
### What happened?

Since the last Z2M Docker update my MiBoxer FUT039Z are not usable anymore from HomeAssistant, it works fine from the Z2M WebInterface.

Here a screenshot from Z2M:
[image]

And here what is reported in HA:
[image]

Maybe I did something wrong but I just came home and saw that the host where Z2M is running on pulled an update.
Everything else is working just fine.

### What did you expect to happen?

Control the light from within HomeAssistant

### How to reproduce it (minimal and precise)

Update to latest Z2M version.

### Zigbee2MQTT version

1.37.0 commit: 46f34c8

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 TI CC2652P + CP2102N

### Setup

HomeAssitant Core on Debian

ich777, 2024-05-05:
Same as #22404

Koenkk, 2024-05-05:
Are you running at least HA `2024.4`

ich777, 2024-05-06:
Thank you, issue solved!

closed 2024-05-06 by ich777