Koenkk/zigbee2mqtt #22451: [New device support]: ZM24E2-1/33 Zemismart tubular motor

alex-v-fraser, 2024-05-04:
### Link

https://www.zemismart.com/products/zm24-zb

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0x0ceff6fffe651577","nwkAddr":60890,"manufId":4098,"manufName":"_TZE200_fzo2pocs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_fzo2pocs","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1714847433521}

### Comments

My new device appears in Zigbee2MQTT as ZM25TQ (but it's really ZM24E2, Only "Stop"/"Up"/"Down" buttons work. Motor limits don't work, position is always 88%, motor doesn't move when I change its position.
And I'm stupid in coding. Please help.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_fzo2pocs',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

alex-v-fraser, 2024-05-05:
I did some external converter which made my curtains work better. In button mode it works fine. Motor limits work but they give some errors when I push set or reset limit:

```
Error 2024-05-05 08:58:42 Exception while calling fromZigbee converter: Value 'true' is not allowed, expected one of [object Object],[object Object]}
Error 2024-05-05 08:58:42 Exception while calling fromZigbee converter: Value 'false' is not allowed, expected one of [object Object],[object Object]}

```
Position mode works now but there are also some bugs. It shows correct fully closed or opened position. But when it fully closed and I try to slide it up by 25% - it moves up by 75%. So when I need to fully close it - I move slider up by fully open position. Motor revers mode changes arrows buttons, but there's no changes in position mode. I suppose, I need to set position to 100% as fully closed, and 0% as fully open. Any ideas?

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_fzo2pocs']),
    model: 'ZM24E2',
    vendor: 'Zemismart',
    description: 'Tubular motor',
    extend: [],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        // e.cover(),
        // e.cover_position(),
        e.enum('reverse_direction', ea.STATE_SET, ['forward', 'back']).withDescription('Reverse the motor direction'),
        // e.enum('border', ea.STATE_SET, ['up', 'down', 'up_delete', 'down_delete', 'remove_top_bottom']),
        e.enum('click_control', ea.STATE_SET, ['up', 'down']).withDescription('Single motor steps'),
        e.binary('motor_fault', ea.STATE, true, false),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.enum('upper_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the upper stroke limit'),
        e.enum('middle_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the middle stroke limit'),
        e.enum('lower_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Reset / Set the lower stroke limit'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.raw],
            [5, 'reverse_direction', tuya.valueConverterBasic.lookup({'forward': tuya.enum(0), 'back': tuya.enum(1)})],
            [12, 'motor_fault', tuya.valueConverter.trueFalse1],
            [13, 'battery', tuya.valueConverter.raw],
            [16, 'border', tuya.valueConverterBasic.lookup({
                'up': tuya.enum(0), 'down': tuya.enum(1), 'up_delete': tuya.enum(2), 'down_delete': tuya.enum(3),
                'remove_top_bottom': tuya.enum(4)})],
            [20, 'click_control', tuya.valueConverterBasic.lookup({'up': tuya.enum(0), 'down': tuya.enum(1)})],
            [103, 'upper_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})],
            [104, 'middle_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})],
            [105, 'lower_stroke_limit', tuya.valueConverterBasic.lookup({'SET': tuya.enum(1), 'RESET': tuya.enum(0)})],
        ],
    },
    extend: [],
};

module.exports = definition;
```

alex-v-fraser, 2024-05-05:
Found `invert_cover: true` in devices.yaml. Changed to `invert_cover: false`. 
I don't remember if I wrote it there before. 
Now it works as expected, except for errors when setting/resetting motor limits.
If anyone competent can help with these errors - we could add a new device to Z2M.

alex-v-fraser, 2024-05-07:
Very unpleasant situation has occurred today: my curtain was fully opened spontaneously at 5:27. 
Next time it spontaneously was set to position 10%. (right now while I writing it!!!)
There's no initiator name in HA-logbook, I have no RC-unit for it.
Z2M log is attached.
Is it possible to find out
@Koenkk will you help me with it please?

```
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,218,237,0,181]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,218,237,0,181]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [218,237,0] - 181
[2024-05-07 05:27:29] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":60890,"relaycount":0,"relaylist":[]}
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: <-- [254,30,68,129,0,0,0,239,218,237,1,1,0,69,0,36,172,39,0,0,10,9,75,2,0,1,7,4,0,1,0,218,237,29,138]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,0,239,218,237,1,1,0,69,0,36,172,39,0,0,10,9,75,2,0,1,7,4,0,1,0,218,237,29,138]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,218,237,1,1,0,69,0,36,172,39,0,0,10,9,75,2,0,1,7,4,0,1,0,218,237,29] - 138
[2024-05-07 05:27:29] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":60890,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":69,"securityuse":0,"timestamp":2599972,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,75,2,0,1,7,4,0,1,0]}}
[2024-05-07 05:27:29] debug: 	zh:controller: Received payload: clusterID=61184, address=60890, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=69, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":75,"commandIdentifier":2},"payload":{"seq":256,"dpValues":[{"dp":7,"datatype":4,"data":{"type":"Buffer","data":[0]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2024-05-07 05:27:29] debug: 	zh:controller:endpoint: ZCL command 0x0************0/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":75,"writeUndiv":false})
[2024-05-07 05:27:29] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x0************0:60890/1 (0,0,1)
[2024-05-07 05:27:29] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":60890,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":102,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,75,11,2,0]}}
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,218,237,1,1,0,239,102,0,30,5,16,75,11,2,0,221]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-07 05:27:29] debug: 	z2m: Received Zigbee message from 'bedroom_curtain', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":7}],"seq":256}' from endpoint 1 with groupID 0
[2024-05-07 05:27:29] debug: 	zhc:tuya: Datapoint 7 not defined for '_TZE200_fzo2pocs' with value 0
[2024-05-07 05:27:29] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bedroom_curtain', payload '{"click_control":"down","last_seen":"2024-05-07T02:27:29.182Z","linkquality":69,"lower_stroke_limit":"SET","middle_stroke_limit":"RESET","motor_fault":false,"motor_speed":false,"position":0,"reverse_direction":"forward","running":false,"state":"CLOSE","upper_stroke_limit":"SET"}'
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-05-07 05:27:29] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,102,160]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,102,160]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,102] - 160
[2024-05-07 05:27:29] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":102}
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: <-- [254,30,68,129,0,0,0,239,218,237,1,1,0,65,0,82,183,39,0,0,10,9,76,2,0,1,1,4,0,1,0,218,237,29,226]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,0,239,218,237,1,1,0,65,0,82,183,39,0,0,10,9,76,2,0,1,1,4,0,1,0,218,237,29,226]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,218,237,1,1,0,65,0,82,183,39,0,0,10,9,76,2,0,1,1,4,0,1,0,218,237,29] - 226
[2024-05-07 05:27:29] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":60890,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":2602834,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,76,2,0,1,1,4 [... cut ...]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,103,161]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,103,161]
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,103] - 161
[2024-05-07 05:27:29] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":103}
[2024-05-07 05:27:29] debug: 	zh:zstack:unpi:parser: --- parseNext []

```

Mustang1966, 2024-10-13:
I got the tubular motor on Friday and made a converter. I haven't had a chance to fully test it since I still need to put in the motor and get the remote to try it out. No errors so far:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_fzo2pocs']),
    model: 'ZM24-ZB',
    vendor: 'Zemismart',
    description: 'Tubular motor',
    extend: [],
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.enum('remote_pair', ea.STATE_SET, ['on', 'off']).withDescription('Remote control pairing mode').withCategory('config'),
        e.enum('motor_direction', ea.STATE_SET, ['normal', 'reversed']).withDescription('Motor direction').withCategory('config'),
        e.binary('motor_fault', ea.STATE, true, false),
        e.enum('upper_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set / Reset the upper stroke limit').withCategory('config'),
        e.enum('middle_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set / Reset the middle stroke limit').withCategory('config'),
        e.enum('lower_stroke_limit', ea.STATE_SET, ['SET', 'RESET']).withDescription('Set / Reset the lower stroke limit').withCategory('config'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.raw],            
            [5, 'motor_direction', tuya.valueConverter.tubularMotorDirection],
            [12, 'motor_fault', tuya.valueConverter.trueFalse1],
            [101, 'remote_pair', tuya.valueConverterBasic.lookup({on: true, off: false})],
            [103, 'upper_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
            [104, 'middle_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
            [105, 'lower_stroke_limit', tuya.valueConverterBasic.lookup({SET: true, RESET: false})],
        ],
    },
    extend: [],
};

module.exports = definition;`

alex-v-fraser, 2024-12-28:
Still waiting and hoping for adding an item

closed 2025-03-09 by github-actions[bot]