Koenkk/zigbee2mqtt #22455: OTA Failed on 067774 Legrand Device Double Gang Wireless switch

Loic691, 2024-05-04:
### What happened?

OTA failed

### What did you expect to happen?

OTA update the device...

### How to reproduce it (minimal and precise)

Each time I try to update the device

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.1 [GA]

### Adapter

Dongle E Sonoff with new Ember driver

### Setup

Add-on Home Assistant

### Debug log

My device :
```
Date du firmware : 20231113
Version du firmware : 0024
Constructeur : [Legrand](https://www.zigbee2mqtt.io/supported-devices/#v=Legrand)
Modèle : [067774](https://www.zigbee2mqtt.io/devices/067774.html#legrand-067774)
Alimentation : Batterie 100%
```

```
Info 2024-05-04 23:54:31Updating 'Bouton Eclairage Cuisine' to latest firmware
Info 2024-05-04 23:54:34Update of 'Bouton Eclairage Cuisine' at 0.00%
Error 2024-05-04 23:54:34~x~> [ZCL to=22306] Failed to send request with status=MESSAGE_TOO_LONG.
```

Loic691, 2024-05-04:
@Koenkk  Can you label this thread as AMBER ?
Thx

Ra72xx, 2024-05-07:
Same with Ikea ICPSHC24-30EU-IL-1/ICPSHC24-10EU-IL-2. But I still use the old ezsp `driver`. And, like in #22463, Z2M has to be restarted after the update attempts. 

I now seemingy have managed to run a complete update, with many error messages, but the version number has not (yet?) been updated in the ui.

However, this Ikea device has not had an update available for months according to Z2M, although another device of the some kind had been updated long ago. So this is an improvement ;-).

Nerivec, 2024-05-08:
@Loic691 Can you generate a network map and identify the path from the coordinator to the device in question (just push irrelevant devices to the side or something) and send a screenshot of that?

Loic691, 2024-05-10:
> @Loic691 Can you generate a network map and identify the path from the coordinator to the device in question (just push irrelevant devices to the side or something) and send a screenshot of that?

Thanks to your response.
Here's a capture but the button isn't attached to a router node... The button works yet...
How to get a schema wich attach the wireless device to one of node ?

[image]

Loic691, 2024-05-10:
Another information following the update to 1.37.1
Here's the log when I try to update the double gangs wireless switch

```
Info 2024-05-10 23:56:17Updating 'Bouton Eclairage Cuisine' to latest firmware
Info 2024-05-10 23:56:25Update of 'Bouton Eclairage Cuisine' at 0.00%
Error 2024-05-10 23:56:25~x~> [ZCL to=22306] Failed to send request with status=MESSAGE_TOO_LONG. 
```

Loic691, 2024-05-12:
For information I will create another thread for the polling of the Legrand device..
Finaly I have one of my wireless legrand device which poll the zigbee network each hours.
It's the only device which this beahviour.
I have another one (same model same firmware) wich won't ping network...

Loic691, 2024-05-20:
Here's more detailed log
`[2024-05-20 23:16:16] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/Sonde Garage', payload '{"battery":100,"device":{"applicationVersion":0,"dateCode":"20230105","friendlyName":"Sonde Garage","hardwareVersion":0,"ieeeAddr":"0x842712fffe41791f","manufacturerID":4742,"manufacturerName":"SONOFF","model":"SNZB-02D","networkAddress":35093,"powerSource":"Battery","softwareBuildID":"1.0.9","type":"EndDevice","zclVersion":8},"humidity":56.8,"last_seen":"2024-05-20T23:16:16+02:00","linkquality":255,"temperature":20.7}'
[2024-05-20 23:16:16] debug: 	z2m: Device 'Sonde Garage' requested OTA
[2024-05-20 23:16:16] debug: 	zh:controller:endpoint: CommandResponse 0x842712fffe41791f/1 genOta.queryNextImageResponse({"status":152}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-20 23:16:16] debug: 	zh:controller:endpoint: Request Queue (0x842712fffe41791f/1): send request
[2024-05-20 23:16:16] debug: 	zh:ember:queue: Status queue=0 priorityQueue=0.
[2024-05-20 23:16:16] debug: 	zh:ember: ~~~> [ZCL to=35093 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"manufacturerCode":null,"transactionSequenceNumber":4,"commandIdentifier":2}]
[2024-05-20 23:16:16] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=162 Len=25]
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=2 frmRx=0]
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=0] Added to rxQueue
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1]
[2024-05-20 23:16:16] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=162 Len=7]
[2024-05-20 23:16:16] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=183 messageTag=112 status=SUCCESS]
[2024-05-20 23:16:16] debug: 	z2m: Responded to OTA request of 'Sonde Garage' with 'NO_IMAGE_AVAILABLE'
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=1] Added to rxQueue
[2024-05-20 23:16:16] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2]
[2024-05-20 23:16:16] debug: 	zh:ember:ezsp: <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=162 Len=26]
[2024-05-20 23:16:16] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [type=DIRECT], [indexOrDestination=35093], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":183}], [messageTag=112], [status=SUCCESS], [messageContents=19040298]
[2024-05-20 23:16:17] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-05-20 23:16:17] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3]
[2024-05-20 23:16:17] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=2] Added to rxQueue
[2024-05-20 23:16:17] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3]
[2024-05-20 23:16:17] debug: 	zh:ember:ezsp: <=== [FRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=162 Len=18]
[2024-05-20 23:16:17] debug: 	zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=59157], [sourceEui=0x001788010be842c1], [lastHopLqi=116], [lastHopRssi=185], [relayCount=0], [relayList=]
[2024-05-20 23:16:22] debug: 	z2m: Update of 'Bouton Eclairage Cuisine' failed (Error: OTA: Device didn't respond to OTA request)
[2024-05-20 23:16:22] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/Bouton Eclairage Cuisine', payload '{"action":null,"battery":0,"device":{"applicationVersion":0,"dateCode":" \u000020231113\u0000\u0000\u0000\u0000\u0000","friendlyName":"Bouton Eclairage Cuisine","hardwareVersion":8,"ieeeAddr":"0x0004740000961c83","manufacturerID":4129,"manufacturerName":" Legrand","model":"067774","networkAddress":22306,"powerSource":"Battery","softwareBuildID":"0024","stackVersion":67,"type":"EndDevice","zclVersion":2},"last_seen":"2024-05-20T23:16:10+02:00","linkquality":236,"update":{"installed_version":2376703,"latest_version":2966019,"state":"available"},"update_available":null,"voltage":2300}'
[2024-05-20 23:16:22] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Bouton Eclairage Cuisine"},"error":"Update of 'Bouton Eclairage Cuisine' failed (OTA: Device didn't respond to OTA request)","status":"error","transaction":"gwmbr-4"}'
[2024-05-20 23:16:22] error: 	z2m: Update of 'Bouton Eclairage Cuisine' failed (OTA: Device didn't respond to OTA request)
[2024-05-20 23:16:22] debug: 	z2m: Error: OTA: Device didn't respond to OTA request
    at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:314:15)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:436:21)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:273:41)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)`

Nerivec, 2024-06-10:
@Loic691 Can you try latest dev branch (will be in July release otherwise)? OTA was reworked, in particular, the start of update was optimized a bit, see if those Legrand devices like it better now?
CC: @cocapic #22498

Loic691, 2024-06-11:
Thx for your help !
No time for now... I spent much time on zigbee.... ZWave is still very very best at all points ;-)
I try this week end.
Since I migrate Z2M debian VM to HA add on, it's a little more complex to switch on dev branch.
What's is the best choice for this with HA. It's my production server...

Nerivec, 2024-06-11:
You can swap between regular and `edge` add-ons in HA (`edge` is using dev branch, simply uninstall/re-install the `edge` add-on to get latest dev branch).

cocapic, 2024-06-12:
Hi, cannot move to dev branch ... or i will be killed by someone
but in 1.38.0 no progress on legrand OTA devices

But i cannot explain yet, better return from 067646 since 1.37.1
https://github.com/Koenkk/zigbee2mqtt/issues/22744

regards

pipolaq, 2024-06-13:
Hi, 
I tried the latest dev branch, but I have trouble to get the OTA, same problems as @cocapic 

`Failed to check if update available for 'Volet parents' (OTA: Device didn't respond to OTA request)`

I tried to remove the battery, wait for 20sec, put back the battery and instantly click on "check for updates" while pressing a button every second for the remote not to fall asleep. No luck for now, tried 100 times ! 

When I check the current firmware i have "019��!��!#@8n#" and formware date "20181218"
[image]

Thanks !

Nerivec, 2024-06-13:
@Koenkk If those Legrand are indeed on 2018 firmware, that's old Zigbee spec (could even be pre-R21 😓). Remember anything changing in OTA in newer specs that could explain the fact they don't want to answer?
Looks like datecode/swbuildid values also need some tweaks for proper display on these devices.

@pipolaq Any chance that device is not connected directly to the coordinator? If it is not, try bringing it close to the coordinator and re-pairing to it specifically. Then try to OTA again (keeping it awake until it starts), see if that makes it happier.

cocapic, 2024-06-14:
> @pipolaq Any chance that device is not connected directly to the coordinator? If it is not, try bringing it close to the coordinator and re-pairing to it specifically. Then try to OTA again (keeping it awake until it starts), see if that makes it happier.

[067646](https://www.zigbee2mqtt.io/devices/067646.html#legrand-067646)
[067774](https://www.zigbee2mqtt.io/devices/067774.html#legrand-067774)
[412170](https://www.zigbee2mqtt.io/devices/412170.html#legrand-412170)
[067771](https://www.zigbee2mqtt.io/devices/067771.html#legrand-067771)

some of them are line powered and some just near the controller same issue

[image]

pipolaq, 2024-06-14:
> @Koenkk If those Legrand are indeed on 2018 firmware, that's old Zigbee spec (could even be pre-R21 😓). Remember anything changing in OTA in newer specs that could explain the fact they don't want [...]

Hi @Nerivec and thanks for the help,
So any way to update this old firmware switch ?
I tried to switch off every line powered device (zigbee router device) and to re-pair my wall switch very close to the coordinator, but no luck with that. I even tried to change the battery and put a new one, without success, i am still not able to upgrade my wall switch.

Regards,

Koenkk, 2024-06-15:
It seems the OTA never starts at the device side. Does someone have the original bridge and can make a sniff when updating the device there?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Mincka, 2024-06-28:
I have the same issue with a more recent firmware and the Edge version of the addon.
`Device didn't respond to OTA request`

What's the exact procedure to help the device to initiate the OTA update?
I tried to keep it alive by pressing the button every second but it did not help.

**Zigbee Model**
Double gangs remote switch

**Zigbee Manufacturer**
Legrand

**Description**
Wireless double remote switch

**Support status**
Supported

**Network address**
0x88F6

**Firmware build date**
20220912

**Firmware version**
001e

cocapic, 2024-07-04:
still not working in 1.39.0

Nerivec, 2024-07-04:
Looks like [some of the firmware versions might have quirks](https://github.com/Koenkk/zigbee2mqtt/issues/19916#issuecomment-2204561938). Can someone with one of these devices see what's going on in the airwaves when OTA is requested? 
https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

jbulmers, 2024-07-29:
Hello all,

Any updates on the above ? I've got the same issue where i can not update my firmware that is from 2018.

Thank you for your help,

[image]

MichaelMure, 2024-08-18:
> It seems the OTA never starts at the device side. Does someone have the original bridge and can make a sniff when updating the device there?
> 
> https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

I have the original bridge, happy to ship it if someone wants to try that.

apydo, 2024-09-25:
> > It seems the OTA never starts at the device side. Does someone have the original bridge and can make a sniff when updating the device there?
> > https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html
> 
> I have the original bridge, happy to ship it if someone wants to try that.

Are you able to try yourself to avoid sending it, @MichaelMure ?

If not, I'm located in France and have two outdated Legrand 412171, maybe I can try to sniff.

beauby, 2024-10-16:
I am having the same issue with a Legrand teleruptor (412170, with firmware version 2441731 ). Rebooting the device before starting the update [as suggested by the docs](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#troubleshooting) seems to improve the situation slightly, as the update does start:
```
Info 2024-10-16 02:14:39z2m: Updating 'Teleruptor Legrand' to latest firmware
Info 2024-10-16 02:14:46z2m: Update of 'Teleruptor Legrand' at 0.02%, ≈ 415 minutes remaining
Info 2024-10-16 02:14:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Teleruptor Legrand', payload '{"device_mode":"auto","linkquality":144,"power":0,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":2441731,"latest_version":6636035,"progress":0.02,"remaining":24916,"state":"updating"},"update_available":null}'
Info 2024-10-16 02:24:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Teleruptor Legrand', payload '{"device_mode":"auto","linkquality":148,"power":0,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":2441731,"latest_version":6636035,"progress":0.02,"remaining":24916,"state":"updating"},"update_available":null}'
Info 2024-10-16 02:34:16z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Teleruptor Legrand', payload '{"device_mode":"auto","linkquality":156,"power":0,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":2441731,"latest_version":6636035,"progress":0.02,"remaining":24916,"state":"updating"},"update_available":null}'
```
However, it doesn't seem to be making any progress past the initial 2%.

Spark3127, 2024-10-21:
Hi ! Same problem here with Legrand 412170 (Télérupteur connecté Drivia) and zigbee2mqtt 1.40.2, would be happy to help !

Dim3333, 2024-11-01:
Hi,

Have you found a solution ?
Same problem with Legrand 412171.

Thanks.

titou4307, 2024-11-04:
Hello
Does anyone can be says the last firmware on this Legrand 412170 ?
I am under Z2M 1.41.0 and the 3 Legrand I have are under : 0065

Thanks

CodeForLove83, 2024-11-04:
> Hello
> 
> Does anyone can be says the last firmware on this Legrand 412170 ?
> 
> I am under Z2M 1.41.0 and the 3 Legrand I have are under : 0065
> 
> 
> 
> Thanks  (From France....Sorry for the poor English)
> 
> 
> 
> 

Hi, I've recently uploaded all the latest firmware files of Legrand devices (latest update 12th Sep. 2024). Please refresh the OTA section to verify if any update is available.

titou4307, 2024-11-04:
Hi @CodeForLove83 

I have all the time "Faill to check if update available...."

CodeForLove83, 2024-11-05:
> Hi @CodeForLove83 
> 
> 
> 
> I have all the time "Faill to check if update available...."
> 
> 

Hi, your device is classified as NLP device (connected outlet family). The latest version available is 0x65.

Loic691, 2024-11-05:
On most on my Legrand Device, I have a message that firmware is more recent that online firmware...
All my battery one's can't have OTA update... The process fails...

CodeForLove83, 2024-11-05:
Yes, I acknowledge the issue with remote/wireless Legrand OTA update. I need to sniff the OTA traffic between Legrand official Gateway and remote/wireless Legrand to identify where the issue is. 

The message returned from z2m when you're on the latest firmware version is misleading. I'll check it out.

titou4307, 2024-11-05:
> Hi, your device is classified as NLP device (connected outlet family). The latest version available is 0x65.

Hi @CodeForLove83 

Sorry I am frenchee so the translation may not sure for me....

do you say I have an old material and so it cant be updated ?

CodeForLove83, 2024-11-05:
> > Hi, your device is classified as NLP device (connected outlet family). The latest version available is 0x65.
> 
> 
> 
> Hi @CodeForLove83 
> 
> 
> 
> Sorry I am frenchee so the translation may not sure for me....
> 
> 
> 
> do you say I have an old material and so it cant be updated ? 

You've already installed the latest version for 412170. 0065 is the latest.

Dim3333, 2024-11-05:
I come back with my update issue about Legrand device 412171.
An update is available:
[image]

But the update doesn't works:
`error 2024-11-05 11:23:43z2m: Update of 'Contacteur Cumulus' failed (Device didn't respond to OTA request)`

Have you a solution to update this device ?

Thanks.

Loic691, 2024-11-05:
Hi
No issues for me... Your firmware looks old. Maybe have an upgrade first with Legrand gateway...
[image]

Legrand Zigbee Device are very strange issues especially Battery's ones...

Dim3333, 2024-11-05:
Thanks.
Legrand devices are very expensive but with a very capricious working. 
I don't have Legrand gateway... 😩

CodeForLove83, 2024-11-05:
> Hi
> 
> No issues for me... Your firmware looks old. Maybe have an upgrade first with Legrand gateway...
> 
> [image]
> 
> 
> 
> Legrand Zigbee Device have very strange issues especially Battery's ones...

Yes, this is typical for this device. You need to power off your electric panel, then retry. Conversely unpair your device and pair it again.

Dim3333, 2024-11-05:
I tried several times to power off the device and then to retry the update.... without success.
Sometimes, the update progress but stop for unknonw reason.

CodeForLove83, 2024-11-05:
> I tried several times to power off the device and then to retry the update.... without success.
> 
> Sometimes, the update progress but stop for unknonw reason.

Ok, what's your coordinator?
I've been having several issues with ember. It seems that most of the issues related to OTA update have been fixed with zstack (Sonoff Zigbee Dongle-P).

Dim3333, 2024-11-05:
I use Sonoff Zigbee Dongle-E with Ember 7.4.3.

CodeForLove83, 2024-11-05:
> I use Sonoff Zigbee Dongle-E with Ember 7.4.3.

Same coordinator I had before. OTA updates were a nightmare with Dongle-E. I'm repairing all my devices to Dongle-P because seems to be a way more stable. 

You could give a try :)

Dim3333, 2024-11-05:
Yes but a lot of works to just update one device .. and i don't want to break my installation. 😉

Loic691, 2024-11-05:
Upgrade OTA with Sonoff E dongle works for me.
But I have to get update before with legrand gateway

Dim3333, 2024-11-05:
> > I use Sonoff Zigbee Dongle-E with Ember 7.4.3.
> 
> Same coordinator I had before. OTA updates were a nightmare with Dongle-E. I'm repairing all my devices to Dongle-P because seems to be a way more stable.
> 
> You could give a try :)

I tried with an CC2531 coordinator. Same issue.

Dim3333, 2024-11-06:
> Hi No issues for me... Your firmware looks old. Maybe have an upgrade first with Legrand gateway... [...]

When you click on "Rechercher les MAJ", what'is the installed version ?

Thanks.

Loic691, 2024-11-06:
> When you click on "Rechercher les MAJ", what'is the installed version ?
>
> Thanks.

When I click on the button, I obtain this..

`zhc:ota:common: Firmware on '0x0004740000a1a14c' ( Contactor) is newer than latest firmware online.`

Very very strange because, I made several updates with Z2M OTA since few months. The last one this summer...
The update didn't update or the information of the firmware Date is bad...

Not understand the behaviour between Legrand devices and Z2M...

Loic691, 2024-11-06:
For information, here's a view of all my Legrand devices...
[image]

Dim3333, 2024-11-06:
> > > Hi No issues for me... Your firmware looks old. Maybe have an upgrade first with Legrand gateway... ![image](https://private-user-images.githubusercontent.com/60837526/383111624-77cccb9f-3b4d-4c18-b5ad-76002d66d96d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA5MTQ1OTYsIm5iZiI6MTczMDkxNDI5NiwicGF0aCI6Ii82MDgzNzUyNi8zODMxMTE2MjQtNzdjY2NiOWYtM2I0ZC00YzE4LWI1YWQtNzYwMDJkNjZkOTZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA2VDE3MzEzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE0OWM1Njk5MzQzNzA0NjNmZWY2NzcyYmNiZjc1MmMzYTJhZGExN2E2OTMxOTRlMDcwYzdkNmNiNmRjMDA4MjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9BXrAV1YItDIFmClwzpyHndBFrBXmcZOaRntCKplE0s)
> > > Legrand Zigbee Device have very strange issues especially Battery's ones...
> > 
> > 
> > When you click on "Rechercher les MAJ", what'is the installed version ?
> > Thanks.
> 
> When I click on the button, I obtain this..
> 
> `zhc:ota:common: Firmware on '0x0004740000a1a14c' ( Contactor) is newer than latest firmware online.`
> 
> Very very strange because, I made several updates with Z2M OTA since few months. The last one this summer... I'm using the Legrand Gateway for upgrading only my Legrand Battery devices which can't updates with Z2M... The update didn't update or the information of the firmware Date is bad...
> 
> Not understand the behaviour between Legrand devices and Z2M...

Thanks but I would like to know the installed version in the log (journal) when you click on "Rechercher les MAJ"
Example :
`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Contacteur Cumulus', payload '{"auto_mode":null,"device_mode":"auto","linkquality":140,"power":0,"power_apparent":0,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":6637055,"latest_version":6636035,"state":"idle"},"update_available":null}'`

CodeForLove83, 2024-11-06:
@Dim3333 what's the device you're looking for firmware update?

I'll check it out on developer Legrand webpage the latest firmware version. I recently uploaded the latest firmware files available. 

Moreover, I guess, there is a little difference between the file version readable on the update file and the file version retrieved by the ZigBee stack and most of the time you run the update you get the error "... is newer than..."

Loic691, 2024-11-07:
> Thanks but I would like to know the installed version in the log (journal) when you click on "Rechercher les MAJ"

Oh sorry
Heres' version for my 2 Legrand contactors

`info 2024-11-07 08:14:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Pompe Piscine', payload '{"auto_mode":null,"current":0,"device":{"applicationVersion":0,"dateCode":" \u000020200123\u0000\u0000\u0000\u0000\u0000","friendlyName":"Pompe Piscine","hardwareVersion":2,"ieeeAddr":"0x0004740000833775","manufacturerID":4129,"manufacturerName":" Legrand","model":"412171","networkAddress":52521,"powerSource":"Mains (single phase)","softwareBuildID":"0065","stackVersion":67,"type":"Router","zclVersion":8},"device_mode":"switch","last_seen":"2024-11-07T08:14:07+01:00","led_when_off":"OFF","led_when_on":"OFF","linkquality":244,"power":0,"power_apparent":0,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":6637055,"latest_version":6636035,"state":"idle"},"update_available":null,"voltage":0}'`

`info 2024-11-07 08:11:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Eclairage Piscine', payload '{"auto_mode":null,"current":0,"device":{"applicationVersion":0,"dateCode":" \u000020200219\u0000\u0000\u0000\u0000\u0000","friendlyName":"Eclairage Piscine","hardwareVersion":2,"ieeeAddr":"0x0004740000a1a14c","manufacturerID":4129,"manufacturerName":" Legrand","model":"412171","networkAddress":42751,"powerSource":"Mains (single phase)","softwareBuildID":"0065","stackVersion":67,"type":"Router","zclVersion":8},"device_mode":"switch","last_seen":"2024-11-07T08:11:46+01:00","led_when_off":"OFF","led_when_on":"OFF","linkquality":244,"power":0,"power_apparent":0,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":6637055,"latest_version":6636035,"state":"idle"},"update_available":null,"voltage":0}'`

I always had strange dateCode with special character... I saw you haven't this... Strange

CodeForLove83, 2024-11-07:
@Loic691 you're running the latest available software version: "softwareBuildID":"0065"

Please read my previous comment regarding the issue with "... is newer than..."

Loic691, 2024-11-07:
> what's the device model you're looking for firmware update?
> 
> I'll check it out on developer Legrand webpage the latest firmware version. I recently uploaded the latest firmware files available.
> 
> Moreover, I guess, there is a little difference between the file version readable on the update file and the file version retrieved by the ZigBee stack and when you run the update you get the error "... is newer than..."

What is your opinion about the special character in the firmware date in the Z2M GUI ? @Dim3333 don't have same characters...

CodeForLove83, 2024-11-07:
I can’t find the log files from @Dim3333 that report this attribute. However, this is Unicode encoding. I’ll need to check my log files, but I’m fairly confident the date code is reported the same as yours.

Loic691, 2024-11-07:
as explained by @CodeForLove83 this message is normal 
`Moreover, I guess, there is a little difference between the file version readable on the update file and the file version retrieved by the ZigBee stack and when you run the update you get the error "... is newer than..."`

Dim3333, 2024-11-07:
Yes yes, I saw.
It's not a real issue juste a remark.
My device is now up to date...not without difficulties.
My first Legrand Zigbee device ..... Maybe the last....

Loic691, 2024-11-07:
Hey hey !
Legrand powered devices are OK. I haven't issues with them.
My big interrogation was about Battery'ones...
Battery drain with Z2M, no OTA updates working, working strangly,...
A franco french manufacturer.... But the only alternative if we want to keep the Legrand Neptune Style for enclosure...
For powered devices there is other solutions with micro modules behind classic Neptune switch. But when we haven't eletric cables connection...hard...
An other possibility for me, less flexible, is to use a  Hue switch battery module behind a physical Neptune switch

CodeForLove83, 2024-11-07:
@Loic691 I didn't get your point. Are you suggesting a solution for replacing battery device with another device? I'm struggling with shutter remote control that it doesn't even seem to work flawlessly with official Legrand Gateway. Sometimes you press the button and it doesn't work. I've already replaced all the devices.

CodeForLove83, 2024-11-07:
@Koenkk I guess we need your help here to speed things up if possible. 

I sniffed the traffic between a remote Legrand device and official Legrand gateway and I gather the messages exchanged to start the OTA update. Can you help us here to figure out how to implement the fix into z2m? 

At first glance it seems that is the end device (0xb2fc) that asks for update (Query Next Image Request, Image Block Request) and then the Gateway replies with the firmware Image. 
z2m doesn't send the firmware images because it's up to the user decide when to upgrade the device, but remote Legrand device seems to behave like that.

Attached you can find the sniff.
e4:a6:98:2e:47:28:bd:e7:d5:db:92:97:d5:61:2d:3f
[sniffing - setting mansarda with option 1 2 1.pcapng.zip](https://github.com/user-attachments/files/17660822/sniffing.-.setting.mansarda.with.option.1.2.1.pcapng.zip)

apydo, 2024-11-08:
> Hi, Legrand device 412171. After a lot of try, I follow this procedure to force the update of my device. [https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/](url)
> 
> It didn't work immediately but the next day the device finally updated.

I have two of them to update.
Could you please be more specific on this update procedure.
The content of different config files for this OTA would be helpful.

Dim3333, 2024-11-09:
1- Open configuration.yaml (found in /config/zigbee2mqtt/configuration.yaml)
2- Add the following lines at the end:
`ota:
  zigbee_ota_override_index_location: my_index.json`
3- Navigate to /config/zigbee2mqtt/
4- Create an empty file and name it my_index.json
5- Copy/Paste the code below in my_index.json
`[
  {
    "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-00654203-NLP.zigbee",
    "force": true
  }
]`
5- Restart HA
6- Try to update your device

From my side, it didn't work immediately but the next day the device finally updated. Why ? How ? I don't know.

Good luck.

[39 more replies]

closed 2025-09-01 by github-actions[bot]