Koenkk/zigbee2mqtt #22457: error: 	z2m: Entity 'homeassistant/select' is unknown

cloudbr34k84, 2024-05-05:
### What happened?

In my logs im seeing this list of errors
```
[2024-05-04 12:17:18] info: 	z2m: Zigbee2MQTT started!
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:19] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:20] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:21] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:21] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:21] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:21] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:21] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:21] error: 	z2m: Entity 'homeassistant/button' is unknown
[2024-05-04 12:17:21] error: 	z2m: Entity 'homeassistant/select' is unknown
[2024-05-04 12:17:21] error: 	z2m: Entity 'homeassistant/button' is unknown
```

### Zigbee2MQTT version

1.37.0-dev commit: 23e97d0

### Adapter firmware version

20240321

### Adapter

Zigstar UZG-01

### Setup

HA Core on NUC Z2M Add ON

### Debug log

[log (1).log](https://github.com/Koenkk/zigbee2mqtt/files/15211352/log.1.log)

Koenkk, 2024-05-06:
Let's merge this with https://github.com/Koenkk/zigbee2mqtt/issues/22258

closed 2024-05-06 by Koenkk

renoye, 2024-05-10:
this is still be found in 1.37.1-1

Mik145, 2024-05-10:
+1
`[2024-05-10 17:29:56] error: 	z2m: Entity 'homeassistant/binary_sensor' is unknown
[2024-05-10 17:29:56] error: 	z2m: Entity 'homeassistant/binary_sensor' is unknown
[2024-05-10 17:29:56] error: 	z2m: Entity 'homeassistant/binary_sensor' is unknown
[2024-05-10 17:29:56] error: 	z2m: Entity 'homeassistant/binary_sensor' is unknown
[2024-05-10 17:29:56] error: 	z2m: Entity 'homeassistant/binary_sensor' is unknown
[2024-05-10 17:29:56] error: 	z2m: Entity 'homeassistant/binary_sensor' is unknown`

svenf01, 2024-05-12:
> +1 `[2024-05-10 17:29:56] error: z2m: Entity 'homeassistant/binary_sensor' is unknown [2024-05-10 17:29:56] error: z2m: Entity 'homeassistant/binary_sensor' is unknown [2024-05-10 17:29:56] error: [...]

Same for me. Multiple of error: 	"z2m: Entity 'homeassistant/binary_sensor' is unknown"

Koenkk, 2024-05-21:
Should be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

rwb196884, 2025-01-10:
Seeing dozens of these for every sub-topic of every device.
For example 
```
[2025-01-09 21:30:31] error:    z2m: Entity 'Home/Hall' is unknown
```
where the actual devices are `Home/Hall/Thermostat` and `Home/Hall/Socket`