Koenkk/zigbee2mqtt #22458: IKEA Remotes (Tradfri 5 button E1524/E1810 and Styrbar) lost arrow left and right hold actions.

Opontios, 2024-05-05:
### What happened?

Don't know if it has been reported before, but didn't find anything related.  IKEA Remotes (Tradfri 5 button E1524/E1810 and Styrbar 4 button) lost arrow left and right hold actions. Instead they show "None" when holding. Don't want to mention how many automations it broke for me.

### What did you expect to happen?

To work normally as it did before for a very long time.

### How to reproduce it (minimal and precise)

Just use arrow_left_hold and arrow_right_hold actions.

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee 3.0 Dongle Plus

### Setup

HAOS

DavyRoswinkel, 2024-05-05:
I have the same issue with:IKEA E2001/E2002

Updated the firmware of the switches yesterday since then this issue happens.

The arrow buttons don't work anymore.
Is there an option to revert the firmware of the switches?

Opontios, 2024-05-05:
My remotes are all updated since long time, there were no new firmware update lately. The issue is not in the firmware, reverting back to v1.36.1-1 brings back the normal functionality.

TB-SE, 2024-05-05:
+1

Luckily I decided to only update one of several remotes I own. To be clear, I use the E2001/E2002.

v6ak, 2024-05-05:
IMHO there are two separate issues with Styrbar:

1. After some Zigbee2mqtt update, left/right **hold** actions don't work.
2. After the recent firmware update, left/right **tap** actions don't work.

Not sure about other IKEA remotes.

Opontios, 2024-05-05:
> > My remotes are all updated since long time, there were no new firmware update lately. The issue is not in the firmware, reverting back to v1.36.1-1 brings back the normal functionality.
> 
> Yesterday the remotes worked for me with the latest zigbee2mqtt, after I updated the ikea firmware they stopped working. The 2 Normal, buttons work, but the arrow buttons stopped working.
> 
> Reverting back zigbee2mqtt didn't solve the issue for me

I haven't done any firmware update lately because there was no recent update. My firmware version for 5 button remote is 24.4.5 with creation date 2019-04-01, and for Styrbar 1.0.024  2020-07-08. If you say that you updated recently that means that you never updated the firmware before. AFAIK, there hasn't been any change in those firmwares, so the problem is clearly in the Z2M software. In any case, reverting back  to  1.36.1-1 fixes the issue. Probably you didn't reboot the whole machine after downgrading, try it.

v6ak, 2024-05-05:
Few days ago, zigbee2mqtt has started offering me updates for all my IKEA
devices (Styrbar, Rodret, Somrig and Tretakt). The old version of Styrbar
firmware was 65572, the new version is 33816593.

For some reason, they haven't appeared at once. Even various Styrbars had
the updates at somewhat different time.

IetIesAai, 2024-05-05:
I have the issue OP describes with the tradfri remote E1524/E1810 - on 1.37 long press on left or right doesn't work (action produces "none"), downgrading to 1.36.1 immediately solves it. 

This particular issue has nothing to do with any recent firmware/ota updates.

dwsnb, 2024-05-05:
Glad to see that I'm not the only one experiencing this issue. Was surprised to see that those actions just stopped working all of a sudden. Don't think its tied to any OTA updates as mentioned above.

tossbek, 2024-05-05:
What I changed:  IKEA E2001/E2002 firmware update to Firmware-Date: 20240408 Firmware-Version: 2.4.11

Zigbee2MQTT-Version 1.37.0-1
Coordinator-Typ zStack3x0
Coordinator-Version 20230507

Device: IKEA E2001/E2002 
Issue: Right & Left Hold as well as Right & Left Relese do not work any longer. 
These triggers are also missing the HA automation Device trigger section.

As of now, I found these errors after holding the right / left buttons
--- RIGHT ----
[2024-05-05 21:42:54] debug: 	z2m: Received Zigbee message from '0x003c84fffee35e9b', type 'commandTradfriArrowHold', cluster 'genScenes', data '{"value":3328}' from endpoint 1 with groupID 0
[2024-05-05 21:42:54] debug: 	z2m: No converter available for 'E2001/E2002' with cluster 'genScenes' and type 'commandTradfriArrowHold' and data '{"value":3328}'
[2024-05-05 21:42:56] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,237,112,1,53,117,88]
[2024-05-05 21:42:56] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,237,112,1,53,117,88]
[2024-05-05 21:42:56] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [237,112,1,53,117] - 88
[2024-05-05 21:42:56] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":28909,"relaycount":1,"relaylist":[30005]}
[2024-05-05 21:42:56] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-05 21:42:56] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,5,0,237,112,1,1,0,142,0,225,247,64,0,0,7,21,124,17,76,9,4,7,53,117,28,251]
[2024-05-05 21:42:56] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,5,0,237,112,1,1,0,142,0,225,247,64,0,0,7,21,124,17,76,9,4,7,53,117,28,251]
[2024-05-05 21:42:56] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,5,0,237,112,1,1,0,142,0,225,247,64,0,0,7,21,124,17,76,9,4,7,53,117,28] - 251
[2024-05-05 21:42:56] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":5,"srcaddr":28909,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":142,"securityuse":0,"timestamp":4257761,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[21,124,17,76,9,4,7]}}
[2024-05-05 21:42:56] debug: 	zh:controller: Received payload: clusterID=5, address=28909, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=142, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4476,"transactionSequenceNumber":76,"commandIdentifier":9},"payload":{"value":1796},"command":{"ID":9,"parameters":[{"name":"value","type":33}],"name":"tradfriArrowRelease"}}
[2024-05-05 21:42:56] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-05 21:42:56] debug: 	z2m: Received Zigbee message from '0x003c84fffee35e9b', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":1796}' from endpoint 1 with groupID 0
--- Left ---
[2024-05-05 21:44:05] debug: 	z2m: Received Zigbee message from '0x003c84fffee35e9b', type 'commandTradfriArrowHold', cluster 'genScenes', data '{"value":3329}' from endpoint 1 with groupID 0
[2024-05-05 21:44:05] debug: 	z2m: No converter available for 'E2001/E2002' with cluster 'genScenes' and type 'commandTradfriArrowHold' and data '{"value":3329}'
[2024-05-05 21:44:08] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,237,112,1,53,117,88]
[2024-05-05 21:44:08] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,237,112,1,53,117,88]
[2024-05-05 21:44:08] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [237,112,1,53,117] - 88
[2024-05-05 21:44:08] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":28909,"relaycount":1,"relaylist":[30005]}
[2024-05-05 21:44:08] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-05 21:44:08] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,5,0,237,112,1,1,0,145,0,104,95,134,0,0,7,21,124,17,81,9,155,11,53,117,28,141]
[2024-05-05 21:44:08] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,5,0,237,112,1,1,0,145,0,104,95,134,0,0,7,21,124,17,81,9,155,11,53,117,28,141]
[2024-05-05 21:44:08] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,5,0,237,112,1,1,0,145,0,104,95,134,0,0,7,21,124,17,81,9,155,11,53,117,28] - 141
[2024-05-05 21:44:08] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":5,"srcaddr":28909,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":145,"securityuse":0,"timestamp":8806248,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[21,124,17,81,9,155,11]}}
[2024-05-05 21:44:08] debug: 	zh:controller: Received payload: clusterID=5, address=28909, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=145, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":true,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4476,"transactionSequenceNumber":81,"commandIdentifier":9},"payload":{"value":2971},"command":{"ID":9,"parameters":[{"name":"value","type":33}],"name":"tradfriArrowRelease"}}
[2024-05-05 21:44:08] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-05 21:44:08] debug: 	z2m: Received Zigbee message from '0x003c84fffee35e9b', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":2971}' from endpoint 1 with groupID 0

FYI:
Directly after the firmware update, I received errors like this; I had perform described by [TB-SE]:
2024-05-05 20:39:26Configuring '0x003c84fffee35e9b'
[... cut ...] t Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

---
[TB-SE]
(https://github.com/Koenkk/zigbee2mqtt/issues/22458#issuecomment-2094703800) • 
+1

Luckily I decided to only update one of several remotes I own. To be clear, I use the E2001/E2002.

[UPDATE] I finally got everything working again after force removing the remote, restarting Z2M and then pairing it again. Fingers crossed it keeps working without issues. 🤞🏻

io53, 2024-05-06:
> Pushed a fix, let me know if it works!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I can confirm that it now works again in the dev branch. 👍

v6ak, 2024-05-06:
I've tried the dev version (1.37.0-dev commit: 358e2ee) with Styrbars:

* Styrbars with old firmware (firmware version 1.0.024 / build date 20200708 / HA shown version 65572) seem to work well – I get actions `arrow_{left,right}_{click,hold,release}` events. (I've fully tested just one Styrbar, but I don't think others would behave differently.)
* Styrbar with new firmware (firmware version 2.4.11 / build date 20240408 / HA shown version 65572) – arrow events (`arrow_{left,right}_{click,hold,release}`) don't work, I just get `on` event when I hold arrow_left. But maybe it is a separate issue. Is there a debug info that would be useful?

JohnBull-1975, 2024-05-06:
> Pushed a fix, let me know if it works!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I updated one of my STYRBAR E2001/E2002 on Friday from 1.0.024 to 2.4.11 and realized yesterday that left-right didn't worked.
I tried to remove and re-pair it and after that up/down stopped working as well!
I have deleted, forced deleted, rebooted etc but no response at all from any button :-(
I can still remove and re-join it and on every re-join it updates the "last seen" status and it also updates last seen if I powercylcle it. But no response from any button.
For quite some time yesterday I it was Offline until I gave it a new name...then it came back Online, but still no response from any button.
Tried to bind functions via MQTT direct to a bulb late yesteray and could first not bind OnOff, but LevelCtrl worked and after that I also managed to bind OnOff and then up/down worked for on/off and dim up/down. 
Unbinded and then nothing worked again. No repsonse from buttons in MQTT console

Planned to take it back to IKEA and tell them it stopped working and see if they exchange it and maybe give me a unit with 2.4.5 and see if that works...

Do you think your fix will solve my problem? When will it be implemented in production version of MQTT? I'm not fully comfortable with trying to change to dev branch...but I really want my STYRBAR back...

v6ak, 2024-05-06:
@JohnBull-1975  I guess the Z2M update will not help with issues with the new firmware (see my previous post), but you can try it.

I've also noticed that Reconfigure doesn't work with Styrbars. It works neither for new nor for old firmware. The error message is very similar, but slightly different. Newlines are mine (I've added them for easier comparison), the last line is the same:

old firmware:

```
Failed to configure '<id>',
attempt 2
(Error: Bind <id>/1 genOnOff from '<some 3-digit number>' failed
(AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:691:28 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:464:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:147:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

new firmware:
```
Failed to configure '<id>',
attempt 4
(Error: Bind <id>/1 genOnOff from '<some long 0xhexa>/1' failed
(AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:691:28 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:464:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:147:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

JohnBull-1975, 2024-05-06:
> @JohnBull-1975 I guess the Z2M update will not help with issues with the new firmware (see my previous post), but you can try it.
>
> I've also noticed that Reconfigure doesn't work with Styrbars. It [...]

Ahhh...I hoped so...
Do you think it will be solved in Z2MQTT or is it to purchase a few new STYRBAR with older firmware and never update...?

Opontios, 2024-05-06:
> Ahhh...I hoped so... Do you think it will be solved in Z2MQTT or is it to purchase a few new STYRBAR with older firmware and never update...?

I guess that question is for @Koenkk

metbril, 2024-05-06:
For the record. My STYRBAR updated to FW 2.4.11 yesterday (new from box with 1.x) and doesn't register left and right clicks. 
I have the latest Z2M (1.37.0) running with HA.

marthubner, 2024-05-06:
> For the record. My [STYRBAR E2001/2002](https://www.zigbee2mqtt.io/devices/E2001_E2002.html#ikea-e2001%252Fe2002) updated OTA to FW 2.4.11 yesterday (new from box with 1.x) and doesn't register left and right clicks. I have the latest Z2M (1.37.0) running with HA.

Just force remove the remote, restart Z2M and pair it again, worked for me...

JohnBull-1975, 2024-05-06:
> Just force remove the remote, restart Z2M and pair it again, worked for me...

Didn't worked for me....can pair it after force remove, but no button work. No reaction at all in Z2M console. Not even "last seen" gets updated when I press a button.

GermanDZ, 2024-05-06:
My control doesn't report `arrow_right_hold`, `arrow_right_release`, `arrow_left_hold`, `arrow_left_release` anymore. But that was happening even before the firmware upgrade (but after upgrading zigbee2mqtt in the recent days).

java-57, 2024-05-06:
> Just force remove the remote, restart Z2M and pair it again, worked for me...

Worked for me as well, but only for the left/right click actions. Hold actions still don't work.

stephdust, 2024-05-06:
> Just force remove the remote, restart Z2M and pair it again, worked for me...

it works for me too after upgrading from dev branch, removing the remote in z2m (no restart of z2m) , re-pair it again and give it back its previous name, thanks guys

JohnBull-1975, 2024-05-06:
I went to IKEA with my STYRBAR. The guy at customer service managed to pair it to one of their lamps...and then when he should show me how he did it (yes I know how it's supposed to work...) he couldn't, so he gave up and gave me a new one with firmware 1.0.024 and it just works again!!! :-)
I hope someone will fix it for the future, but I will wait a good while until I try to update any of my two STYRBAR...

v6ak, 2024-05-06:
I've digged slightly deeper to the issue with new Styrbar firmware:

1. When I press left on Styrbar with old firmware, I see raw messages and MQTT messages in logs. With new firmware, nothing appears.
2. I sometimes see  the following message (where `0xdeadbeefcafebabe` corresponds to the ID of the updated Styrbar):

```
zigbee2mqtt      | [2024-05-06 18:57:48] error: 	z2m: Failed to configure '0xdeadbeefcafebabe', attempt 2 (Error: Device 0xdeadbeefcafebabe has no output cluster genScenes
zigbee2mqtt      |     at getEndpointsWithCluster (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:30:15)
zigbee2mqtt      |     at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:131:31)
zigbee2mqtt      |     at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:27)
zigbee2mqtt      |     at Configure.configure (/app/lib/extension/configure.ts:121:13)
zigbee2mqtt      |     at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

Koenkk, 2024-05-06:
Does a re-configure of the device helps? Press the yellow refresh button on the device page, make sure to wakeup the device before doing that.

tossbek, 2024-05-06:
@Koenkk 
> Pushed a fix, let me know if it works!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I switched just some minutes to the dev branch and everything worked with  IKEA E2001/E2002 Firwmare 2.4.11.
This means, that all buttons are available again. 
Zigbee2MQTT-Version 1.37.0-dev commit: [b2e8b3e](https://github.com/Koenkk/zigbee2mqtt/commit/b2e8b3e)
Coordinator-Typ zStack3x0
Coordinator-Version 20230507

Detail
Stopped Zigbee2MQTT HomeAssistant Addon
Installed Zigbee2MQTT-EDGE 
Copied config
Started Zigbee2MQTT-EDGE 
Rebooted HomeAssistant  just to be sure to get the missing triggers
Right/Left Release and Hold buttons are available again
Thanks

kikher, 2024-05-06:
In my case, my styrbars do not report short or long actions. When I try to reconfigure it, I get the error that I attach in the image. 
Seeing the latest comments and that is patently solved with the dev version. There is a release date for the official version for those of us who are not comfortable with the edge version.
[image]

jhbruhn, 2024-05-06:
I can confirm the left/right tap button behaviour missing after Styrbar firmware update with Z2M1.37.0 and ember stack. Error on reconfigure is:
```
Failed to configure 'Wohnzimmer STYRBAR', attempt 4 (Error: Bind 0x50325ffffe2ac628/1 genOnOff from '0x040d84fffe401de5/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":188}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:96:31) at EmberAdapter.onMessageSentDeliveryFailed (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:558:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:3957:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:794:18) at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:448:22) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))
```

greiseknoue, 2024-05-06:
I haven't tried the dev branch yet and i also had the issue that the device didn't respond at all after device/firmware update, but i found out about a quick fix:

1. Go to your styrbar device in Z2M.
2. Go to the tab bind and look for the "Cluster" title
3. GenPollCtrl was the only one marked for me. So mark the other four in the image
[image]
4. Remove one battery from the device, press the blue "Bind" button and at the same time put in the battery and press any buttons on the device. 

You should get a blue message that the binding was successful.  All buttons should work, except "hold left/right".

v6ak, 2024-05-07:
@greiseknoue Is it Styrbar, or some other device?

In bind tab, I see two items, one for coordinator, one for default_bind_group. The settings are exactly the same for both old and new FW (I am sure I haven't changed it for those two Styrbars):

[image]

The list of offered clusters is a bit different:

* Most notably, I don't have Scenes there, so I can't enable it. And this might be the reason why I am getting `Device 0xdeadbeefcafebabe has no output cluster genScenes` when trying to reconfigure it.
* I don't have 64636 and Groups.
* I have additional manuSpecificAmazonWWAH (probably [“Works with All Hubs”](https://developer.amazon.com/en-US/blogs/alexa/device-makers/2020/07/introducing-new-works-with-alexa-requirements-for-zigbee-devices)).

jamesonuk, 2024-05-07:
I wonder if the missing actions is the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/22506

If someone uses MQTT explorer you can see the list of published device actions Vs the actual actions.   This has been happening to me since 1.36

v6ak, 2024-05-07:
@jamesonuk Likely a separate issue. In my case, I see the actions in HA, but they aren't triggered.

chrizzlibaer, 2024-05-07:
Having the same issue with all of my STYBAR E2001/E2002 remotes after updating them to Firmware 20240408 / 2.4.11, i figured, that the coordinator bind only has "genPollCtrl" active:
[image]

So I did the following:
- Opened the device, pushed button 4 times to repair
- Started Permit_Join in z2m to allow it to rejoin
- While joining, checked every Cluster in Bind besides "64636"  and "genBasic"
- It binds all clusters again and buttons are working as expected:

[image]

Maybe this helps :-)

v6ak, 2024-05-07:
You have Scenes cluster (the one you apparently need for arrows), I don't. Maybe just enabling the Scenes cluster would do the same job for you.

I've even tried publishing message `zigbee2mqtt/bridge/request/device/bind` with payload `{"from":"0xdeadbeefcafebabe/1","to":"Coordinator"}` or `{"from":"0xdeadbeefcafebabe/1","to":"Coordinator","clusters":["genPollCtrl","genPowerCfg","genOnOff","genLevelCtrl","genScenes"]}`, but it didn't add the cluster.

Then, I tried to remove and re-add the Styrbar. First, I couldn't enter the pairing mode (not sure why), but I succeeded after re-trying it multiple times. After re-pairing, it finally works (tested all actions, including hold and release).

Now, I have just one item in Bind tab and it has Scenes cluster, enabled by default.

smeen89, 2024-05-07:
So I've found that @chrizzlibaer solution works. But in my case I got the same options under bind tab as @v6ak like image below. Scenes missing...

[image]

What worked for me was to force remove the styrbar from Z2M. Then re-pair it. And when re-pairing I got some error that it couldn't bind some stuff. **That's fine.** By now the bind tab had the options mentioned in @chrizzlibaer image like below

[image]

I followed the rest of @chrizzlibaer solution and pulled out one battery, put it back in, pressed a button on the remote and followed that by pressing "Bind <3".

Lastly I had to go into home assistant and under devices look up my styrbar and manually change the device id of the action sensor so it matched my previous setup/automations.

Hope this helps someone.

v6ak, 2024-05-07:
Second Styrbar, similar story. Not sure if Z2M restart has helped me to re-pair the Styrbar.

Third Styrbar: I've tried to re-pair before FW update. It was smoother, but I didn't get the Scenes cluster. So, it continued with approximately the same story.

I have few more Styrbars to update. I might test a different approach there.

(Z2M version still 1.37.0-dev commit: 358e2ee)

Koenkk, 2024-05-07:
Pushed another fix, configure should succeed now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

v6ak, 2024-05-08:
Fourth Styrbar updated with older Z2M (1.37.0-dev commit: 358e2ee). Then, I updated Z2M to 1.37.0-dev commit: 6b886f6. It started working well **without** any further actions like reconfiguring.

Fifth Styrbar will update smoothly, I guess.

Koenkk, 2024-05-08:
I checked the following with the latest dev, all OK with:
- E1524/E1810 @ 2.3.080
- E2001/E2002 @ 1.0.024
- E2001/E2002 @ 2.4.11

Assuming this can be closed now (I will make a hotfix release ASAP)

closed 2024-05-08 by Koenkk

Opontios, 2024-05-08:
So the E1524/E1810 with the latest 24.4.5 won't have arrow hold functionality even after the fix?

v6ak, 2024-05-09:
Hmm, after update of the 5th Srytbar with Z2M 1.37.0-dev commit 358e2ee, arrows don't work and reconfigure fails, but the error message is a bit different. **This time, genOnOff fails,** not genScenes.

I've updated to the latest dev (1.37.0-dev commit: cce41c2), but reconfigure fails with the same outcome :(

There is the full error message (line breaks are mine):

```
Failed to configure '0xdeadbeefcafebabe', attempt 1 (
Error: Bind 0xdeadbeefcafebabe/1 genOnOff from '0x0011223344556677/1' failed (AREQ - ZDO - bindRsp after 10000ms)
 at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
 at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:691:28
 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
 at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:464:13)
 at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9)
 at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/ikea.ts:143:17)
 at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21)
 at Configure.configure (/app/lib/extension/configure.ts:121:13)
 at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21)
 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)
)
```

Koenkk, 2024-05-09:
It looks like the device is not awake during the re-configure, make sure to wake it up by clicking a button on it right before doing the re-configure.

v6ak, 2024-05-09:
I was trying to wake it, but probably in a wrong way. I assumed that any button will do it, and clicked arrows, because they didn't have any side effect. But it seems that they didn't have any effect at all. Waking it by `on` button succeeded.

Nezz, 2024-05-10:
I had the same issue where none of the actions worked on the Styrbar remote on firmware 2.4.11.

I updated to 1.37.1. I tried re-configuring the remote but it didn't help. I tried re-pairing but it fails to configure.
```
[2024-05-10 23:51:05] debug: 	zh:controller:device: Interview - active endpoints request failed for '0x04cd15fffebf52d6', attempt 2
[2024-05-10 23:51:05] debug: 	zh:controller:device: Interview - quirks check for 'Remote Control N2'-'IKEA of Sweden'-'EndDevice'
[2024-05-10 23:51:05] debug: 	zh:controller:device: Interview - quirks did not match
[2024-05-10 23:51:05] debug: 	zh:controller:device: Interview - failed for device '0x04cd15fffebf52d6' with error 'Error: Interview failed because can not get active endpoints ('0x04cd15fffebf52d6')
    at Device.interviewInternal (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:635:19)
    at Device.interview (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:489:13)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:597:17)'
[2024-05-10 23:51:05] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-05-10 23:51:05] error: 	zh:controller: Interview failed for '0x04cd15fffebf52d6 with error 'Error: Interview failed because can not get active endpoints ('0x04cd15fffebf52d6')'
[2024-05-10 23:51:05] error: 	z2m: Failed to interview '0x04cd15fffebf52d6', device has not successfully been paired
```

More logs:
[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/15279967/logs.txt)

Sabjul33, 2024-05-12:
It may be the same issue but i can't update my RODRET wireless dimmer Product ID: E2201. 
I have an error : 
```
Info <small>2024-05-12 21:36:45</small>`Updating ' xxx' to latest firmware`
Info <small>2024-05-12 21:37:16</small>`Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "918".`
Error <small>2024-05-12 21:37:50</small>`Update of ' xxx' failed (OTA: Device didn't respond to OTA request)`
```
and when i try to reconfigure the device i have this : 

`Failed to configure ' xxx ', attempt 1 (Error: Bind 0x287681fffe4b85d3/1 genPollCtrl from '0xe0798dfffef43c06/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":62}) at EmberOneWaitress.deliveryFailedFor (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:96:31) at EmberAdapter.onMessageSentDeliveryFailed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:540:30) at Ezsp.emit (node:events:519:28) at Ezsp.ezspMessageSentHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:3960:18) at Ezsp.callbackDispatch (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:797:18) at Ezsp.tick (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:451:22) at listOnTimeout (node:internal/timers:573:17) at processTimers (node:internal/timers:514:7))`

Thanks for your help

I have 1.37.1 Z2M and V7.4.2 sonoff Dongle E Firmware

chrizzlibaer, 2024-05-12:
> It may be the same issue but i can't update my RODRET wireless dimmer Product ID: E2201. I have an error :
>
> ```
> Info <small>2024-05-12 21:36:45</small>`Updating ' xxx' to latest firmware`
> Info [...]

Battery powered zigbee devices always need to be waken up when updating.

So issue the firmware then press the on or off buttons multiple times for around 5 seconds, then the firmware update should start.
You can now stop The button pressing. Just leave the remote near the coordinator in order so stabilize and speed up the upgrade progress as good as possible. It may take hours to update though.

Sabjul33, 2024-05-14:
> Battery powered zigbee devices always need to be waken up when updating.
>
> So issue the firmware then press the on or off buttons multiple times for around 5 seconds, then the firmware update should [...]

Thanks for your help Update done ..

Greendragoonjr, 2024-12-30:
Hello,

I stumbled on this post after struggling a lot with the E2001 remote after the 2.4.16 firmware update. 
After reading lots of post, tried rebinding the remote with three IKEA LED1924G9 bulbs, i finally managed to have the remote work for the up and down button. Unfortunately the left and right keys does not respond. The bulbs does not change between their 8 default colors.

What is weird is that during the process, I completely removed the bulbs and the remote of Z2M and Home Assistant. And to check that all worked all right I binded the remote with the bulbs outside of Z2M. 
There i was able to control the bulbs as intended with on/off, brightness change and color change between the 8 scenes of IKEA. 

After adding the bulbs and the remote in HA through Z2M, I was able to bind the three bulbs (NOT as a group, all individually) with the remote. Now I can finally turn the bulbs on and off and change the brightness using the remote. But I can't change the bulbs color with the remote.

When checking the logs, Z2M correctly detect the left and right arrow press but does not proceed in the change of color, while it does for the up and down buttons with birghtness 
[image]

Here's the binging config of the remote:
[image]

And the firmware version of the bulb and the remote: 
[image]

My HA Core is in 2024.12.4 version, and HA operating system in 14.0
My Z2M is in  1.42.0-2. 

Since the left and right buttons correcly works outside of Z2M, I guess there is something missing in Z2M. But i do not know what.

Does anyone finally managed to have the Ikea styrbar E2001/E2002 remote working fully properly in his 2.14.6 version?

Thanks for the help

Opontios, 2025-01-08:
Why are you using the binding function in Z2M? What's the purpose? To have them working without HA?

Greendragoonjr, 2025-01-09:
@Opontios Yes so that the remote and the bulbs can communicate outside of HA. So that if HA is down, then i can still light my rooms. But also the communication between the remote and bulbs is faster (normally). But i want to bind them through Z2M, once the devices are detected in HA, and not by using the Ikea procedure explained in their user manual because. Otherwise the devices are not found in HA and i can't use additional automations, specific light colors.

And in my specific case, it also allows to easily go between the 8 default light colors of the Ikea bulbs. I tried to make a template to do the same within HA but that is too advanced for me and i couldn't do it.

What is weird with the way i succesed to bind the remote and the bulbs in Z2M, is that now the battery levels of the remote is no more available. it remains in a "N/A" level

Greendragoonjr, 2025-01-16:
Heya @rohitpid, it is similar but not quite the same.

 For me both left and right buttons are detected by Z2M. And i'd say almost immediately the action appears in the Log. 
But these buttons will not change the colors of the bulbs within their 8 default color set.

Since Z2M V2.0 i was able to bind more clusters (Identify, OTA and manuSpecificIkeaUnkown). But it didn't help with my problem.

[image]

I'll answer in your thread as it this one is closed.