Koenkk/zigbee2mqtt #22459: Inovelli VZM36 reads attributes into incorrect property names

rpendleton, 2024-05-05:
### What happened?

After pairing an Inovelli VZM36 fan/light canopy module in Zigbee2Mqtt, I went to the Exposes tab and clicked the refresh button on each attribute. However, when refreshing attributes ending in a 1 or 2, nothing changed in the UI.

Looking at the logs and MQTT messages, it seems this is happening because querying the current value of attributes ending in 1 or 2 updates state properties such as `dimmingSpeedDownRemote`, regardless of which endpoint is being updated.

However, when changing those same parameters from the UI, the new values are saved in the state using property names that contain the endpoint, such as `dimmingSpeedDownRemote_1` for the light or `dimmingSpeedDownRemote_2` for the fan. This holds true for other attributes as well.

### What did you expect to happen?

When refreshing exposed attributes, I expect:
* the current value to be displayed in the UI
* the current value to be stored in the device state using a consistent property name that doesn't overlap with other properties of the device

### How to reproduce it (minimal and precise)

1. Pair an Inovelli VZM36
2. Navigate to the Exposes tab
3. Click the refresh button on "DimmingSpeedUpRemote 1" or any other property ending in 1 or 2

### Zigbee2MQTT version

1.37.0-dev commit: beedfae

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Plain on x86_64 Docker

### Debug log

Attached: [log.log](https://github.com/Koenkk/zigbee2mqtt/files/15211758/log.log)

I split the log into six sections that correlate to:
* Refreshing dimmingSpeedUpRemote_1
* Refreshing dimmingSpeedUpRemote_2
* Setting dimmingSpeedUpRemote_1
* Setting dimmingSpeedUpRemote_2
* Refreshing dimmingSpeedUpRemote_1
* Refreshing dimmingSpeedUpRemote_2

Notably, see how after refreshing both parameters, there's only a single property in the state for `dimmingSpeedUpRemote`.

However, after setting both parameters, there are suddenly three properties in the state for `dimmingSpeedUpRemote`, `dimmingSpeedUpRemote_1` and `dimmingSpeedUpRemote_2`.

And then after that, refreshing the two attributes results in the `dimmingSpeedUpRemote` property being changed to whichever property was refreshed most recently.

Koenkk, 2024-05-05:
@InovelliUSA could you take a look at this?

InovelliUSA, 2024-05-13:
I do know about this issue, but have not had time to look into resolving it yet.

closed 2025-02-02 by Koenkk