Koenkk/zigbee2mqtt #22460: Unavailable entities from operational device

dankarization, 2024-05-05:
### What happened?

I have a few devices throughout the house with this problem:
[image]

1) They are fully operational from Z2M frontend, have more than good coverage (first 4 in the list below are directly under the coordinator in the same room with it), report nicely and had no such problems before as i believe.
2) They have several entities, but some of them (of course, by the will of fate, not the ones I really need) are operational from home assistant
3) All entities were working lately, and just became unavailable for some reason 

[image]
[image]
[image]

### How to reproduce it (minimal and precise)

Idk, maybe turn off MQTT addon for some time.

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

20240315

### Adapter

TZT HamGeek POE (ZigStarGW) with cc2652p2 chip

### Setup

Home Assistant OS with 2 addons for each floor and coordinator

### Debug log

In these logs look for problems with devices:
- Island LED
- Island floor
- Подсветка кухня шкаф
- Kitchen. Working light
- Led light. Sauna
- Wine cellar chandelier (it's a group of 3 lights: Люстра 1, Люстра 2, Люстра 3)

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15212201/log.log)
[log1.log](https://github.com/Koenkk/zigbee2mqtt/files/15212202/log1.log)
[log2.log](https://github.com/Koenkk/zigbee2mqtt/files/15212215/log2.log)

Failed to ping for all '(отключен)' ones is _ok'ish_ as they are physically disconnected from power.

log1.log file, line 7892, I see MQTT connection problems that were quickly resolved by itself. Aproximately at this time entities I talk about became unavailable and never came back (but almost all others are ok)

dankarization, 2024-05-05:
UPD: restored 1.36.1-1 version backup and the problem is gone, but on the 1.36.1-1 i have a problem with brightness:1000 as many other people, why I updated to 1.37.0-1

cloudbr34k84, 2024-05-05:
Yep same problem

Koenkk, 2024-05-05:
What home assistant version are you using? Make sure to use at least 2024.4 as mentioned in the Z2M release notes.

craigwitter, 2024-05-05:
Hasos 2024.5.1 here.

Thanks,
Craig

cloudbr34k84, 2024-05-05:
I'm running beta

craigwitter, 2024-05-07:
@Koenkk does this help? https://github.com/home-assistant/core/issues/116914

Koenkk, 2024-05-07:
@craigwitter that's not about entities becoming unavailable as far as I can see

popy2k14, 2024-05-09:
Use EMQX broker instead of mosquito.
See here: https://github.com/Koenkk/zigbee2mqtt/issues/18862

MarcoGos, 2024-05-09:
@popy2k14 Thanks, that really helped. I used to use Mosquitto and had the same issue as described here. I swapped Mosquitto with EMQX, now a reboot of home assistant always restores all entities and no entities become unavailable anymore.

popy2k14, 2024-05-09:
@MarcoGos You're welcome. Nice that it also solved the issue for you. Running EMQX for over a month now without a issue.

popy2k14, 2024-11-06:
Running EMQX for 6 months now without facing this issue ever.
But i think this is not solved: https://github.com/Koenkk/zigbee2mqtt/issues/18862

PS.: Not tested mosquitto myself again

MarcoGos, 2024-11-06:
@popy2k14 I can second that. I use EMQX for a couple of months now without this issue, where as using mosquitto gave me this issue often.

dankarization, 2024-11-06:
I am still running mosquitto, on dozen different Zigbee coordinators and Z2M 1.39 and higher. The one exact device from my screenshot is connected to zStack3x0 with firmware 20240315. Problem is gone.

closed 2024-11-06 by dankarization

popy2k14, 2024-11-06:
@MarcoGos Did you tried recent versions of Z2M and mosquitto?
Maybe issue is solved?
I think the right place to discuss further is this issue: https://github.com/Koenkk/zigbee2mqtt/issues/18862