Koenkk/zigbee2mqtt #22462: [New device support]: Nous L6Z recognized as TS0001 should be TS0001_power

marcel1188, 2024-05-05:
### Link

https://www.amazon.de/ZigBee-schalter-Nous-L6Z-Stromverbrauch/dp/B0CMTZXZRQ?th=1

### Database entry

{"id":93,"type":"Router","ieeeAddr":"0xa4c1387a591f6f3c","nwkAddr":48846,"manufId":4417,"manufName":"_TZ3000_qaabwu5c","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u0001\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_qaabwu5c","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":234,"rmsCurrent":0,"activePower":0}},"manuSpecificBosch":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2828372,"53251":122,"53252":10522,"53253":1272,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002b48291a","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1714902705739}

### Comments

Dear all,

I recently installed two of the above mentioned Nous L6Z. They are advertised as powermonitoring and working with Z2M.
Unfortunately it only exposes the switch and the powermonitoring is not shown. I found that the devices are recognized as TS0001 but I think it should be TS0001_power. I tried to fix it using an external converter, but due to the fact that this is completely new for me, I was not able to make it work properly. I am using the HA-addon.

Could somebody help me or give me advice, please?

### External definition

```shell
const {identify, onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0001'],
    model: 'TS0001',
    vendor: '_TZ3000_qaabwu5c',
    description: 'Automatically generated definition',
    extend: [identify(), onOff({"powerOnBehavior":false}), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

Topherfx, 2024-05-23:
I have the same problem. Can't get it to work properly either. The nous L6Z switch gets recognized as TS0001 an doesn't expose the values of the power monitoring

Koenkk, 2024-05-27:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-27 by Koenkk

Koenkk, 2025-07-30:
no problem, added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.