Koenkk/zigbee2mqtt #22463: Z2M crash while running OTA update of Philips Hue bulb

Ricc68, 2024-05-05:
### What happened?

During the OTA update of a Philips Hue bulb z2m crashed.

### What did you expect to happen?

OTA update complete successfully.

### How to reproduce it (minimal and precise)

Run the OTA update of the Philips Hue bulb.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff ZBDongle-E

### Setup

Add-on on Home Assistant OS, host is a VM on x86-64, dongle uses ember driver

### Debug log

[2024-05-05 12:32:04] info: 	z2m: Update available for 'Lampada del soggiorno'
[2024-05-05 12:32:17] info: 	z2m: Updating 'Lampada del soggiorno' to latest firmware
[2024-05-05 12:32:18] info: 	z2m: Update of 'Lampada del soggiorno' at 0.00%
[2024-05-05 12:32:52] info: 	z2m: Update of 'Lampada del soggiorno' at 0.14%, ≈ 401 minutes remaining
[2024-05-05 12:33:25] info: 	z2m: Update of 'Lampada del soggiorno' at 0.27%, ≈ 411 minutes remaining
[2024-05-05 12:34:01] info: 	z2m: Update of 'Lampada del soggiorno' at 0.51%, ≈ 339 minutes remaining
[2024-05-05 12:34:36] info: 	z2m: Update of 'Lampada del soggiorno' at 0.68%, ≈ 337 minutes remaining
[2024-05-05 12:35:09] info: 	z2m: Update of 'Lampada del soggiorno' at 0.82%, ≈ 346 minutes remaining
[2024-05-05 12:35:11] error: 	zh:ember: Delivery of BROADCAST failed for "65532" [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":154} messageTag=70]
[2024-05-05 12:35:11] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":155} messageTag=48]
[2024-05-05 12:35:43] info: 	z2m: Update of 'Lampada del soggiorno' at 0.95%, ≈ 356 minutes remaining
[2024-05-05 12:36:18] info: 	z2m: Update of 'Lampada del soggiorno' at 1.17%, ≈ 339 minutes remaining
[2024-05-05 12:36:57] info: 	z2m: Update of 'Lampada del soggiorno' at 1.49%, ≈ 307 minutes remaining
[2024-05-05 12:37:31] info: 	z2m: Update of 'Lampada del soggiorno' at 1.66%, ≈ 309 minutes remaining
Error: Delivery failed for {"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":11,"options":4352,"groupId":0,"sequence":0}
    at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:96:31)
    at EmberAdapter.onMessageSentDeliveryFailed (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:558:30)
    at Ezsp.emit (node:events:517:28)
    at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:3957:18)
    at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:794:18)
    at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:448:22)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)

Koenkk, 2024-05-06:
With crash, do you mean that z2m completely stops after this?

@Nerivec could it be that we are missing an await somewhere? This is a command response `imageBlockResponse` which has a `catch`: https://github.com/Koenkk/zigbee-herdsman-converters/blob/0bdd6bf799d131a06a013dc3003eadcd760faf5d/src/lib/ota/common.ts#L499

Ricc68, 2024-05-06:
Yes, with crash I mean z2m completely stops after this.
To run z2m again I had to press the Start button.

mundschenk-at, 2024-05-29:
With the latest Hue FW updates becoming available, I have about 50 devices to update. Each is super slow and often times out after an hour or so, so you need to start the update again. Unfortunately, there is also no way to schedule multiple updates in series (marking several devices to update eventually brings down the network due to congestion).

Nerivec, 2024-05-29:
It must be something specific to Hue that is causing trouble. I had a report, not too long ago, of almost 20 Inovelli devices being updated (sequentially of course 😁) without issue on an ember network with over 300 devices; that would indicate the underlaying system (OTA & driver) is more than fine.
Of course, that kind of stress on the network is entirely dependent on the quality of the network, routes taken during OTA (any subpar router involved), anything spamming the network at the same time (like those bad Tuya devices), etc...

mundschenk-at, 2024-05-29:
To me it looks like the stability issue has something to with `last_seen` being enabled as per https://github.com/Koenkk/zigbee2mqtt/issues/19531 - Z2M publishes MQTT state message every few hundred milliseconds, several times per second (!) when `last_seen` is enabled. I can't see anything less than a second as a useful granularity, probably even 5 seconds would be enough.

With only one update at a time (if that's a hard requirement, Z2M should enforce it), updating the Hue lights seems to be stable for me today, but still quite slow (about an hour for one light, will have to check the logs for better statistics). What I did notice with a quick look at the debug logs is a consistent discrepancy between the requested and sent chunk size:

```
[2024-05-29 15:32:24] debug:    zhc:ota:common: Request offsets: fileOffset=179206 pageOffset=0                 dataSize=64
[2024-05-29 15:32:24] debug:    zhc:ota:common: Payload offsets: start=179206 end=179256 dataSize=50
```

mundschenk-at, 2024-05-29:
I also get this a lot in the logs, not sure if this is expected:

```
[2024-05-29 15:49:03] debug: 	zh:controller: Skipping command 'imageBlockRequest' because it is missing from the lookup
```

mundschenk-at, 2024-05-29:
Sorry @Nerivec, I may have muddied the waters here - I am using zStack, not Ember. Should I open a separate issue?

Nerivec, 2024-05-29:
No on the contrary, it confirms this is a more "generic" issue.

The first `dataSize` should be `maximumDataSize` (the value received), the second one is the value determined/used by Z2M.
The 'skipping' is because OTA is handled separately, it doesn't have any impact.

At ~50 bytes per message, and the average firmware being 250KB (_some Hue appear to be double that_), that's a lot of messages, plus it is throttled to avoid crashes, so timeframe for update is unfortunately pretty long; best scenario is unlikely to be less than 20 minutes.

mundschenk-at, 2024-05-29:
Would it be possible to increase the chunk size to the requested 64 bytes? I know there was a similar discussion in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6193 and it would a appear that the 50 bytes is more or less an arbitrary limit (i.e. it probably should not be significantly higher due to network congestion, but a power-of-two value like 64 would at least seem more natural too me). 

It's been some time since I had a Hue Bridge in use, but FW updates always seemed much quicker to me then. (Though it is possible that this only seemed that way because upgrading was "fire and forget", it would update all devices automatically once you gave it the go.)

Nerivec, 2024-05-29:
From the comments in code, it seems the 50 was chosen because higher values often result in instabilities. Also have to consider deeply nested routes and the cost involved.
The throttling that Z2M does probably is higher than what the Hue Bridge does, hence the possibly longer timeframe. It's a static value at the moment (250ms), of course a variable one would be better, but the problem is with supporting the myriad of setups/networks/devices... 😅

thargy, 2024-05-30:
I have 170 Hue Devices (FML), I would probably be willing to kill for the option to update all (even if it did them sequentially).

As it is, I often have 10-20 on the go at a time, and they usually take 2-3 hours each. I've not seen any real problems (though Z2m can respond slowly to clicks on the UI).

Nerivec, 2024-05-31:
@mundschenk-at Any chance you can run custom code to test an OTA refactor? If yes, find me on zigbee2mqtt Discord.

mundschenk-at, 2024-05-31:
@Nerivec Sure, I've pinged you on Discord.

Nerivec, 2024-06-10:
I know @mundschenk-at had some pretty good results with the new [OTA refactoring](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7585). Had some good feedback on large `ember` networks too. So I'm going to tag this.
@Ricc68 not sure if you can test latest dev branch, otherwise it will be in July release. Let me know then how it goes.

Ricc68, 2024-06-11:
@Nerivec can't easily test the dev branch as I'm on HAOS. Will report back in July with the next release.

mundschenk-at, 2024-06-11:
@Ricc68 You could switch to the Zigbee2MQTT Edge add-on.

Ricc68, 2024-06-15:
@Nerivec I have installed the Edge branch (latest dev) but the OTA update keeps failing. The good news is that z2m is not crashing anymore.
It says the device didn't respond to OTA request, but the strange thing is that this device always responded to OTA requests before and now is not responding anymore:

[2024-06-15 18:18:42] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Lampada del soggiorno","transaction":"rpjz7-1"}'
[2024-06-15 18:18:42] info: 	z2m: Updating 'Lampada del soggiorno' to latest firmware
[2024-06-15 18:18:42] debug: 	zh:controller:endpoint: ZCL command 0x001788010bd3aeb7/11 genBasic.read(["dateCode","swBuildId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false,"sendPolicy":"immediate"})
[2024-06-15 18:18:42] debug: 	zh:ember:queue: Status queue=0 priorityQueue=0.
[2024-06-15 18:18:42] debug: 	zh:ember: ~~~> [ZCL to=6105 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":11,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"manufacturerCode":null,"transactionSequenceNumber":7,"commandIdentifier":0}]
[2024-06-15 18:18:42] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=56 Len=28]
[2024-06-15 18:18:42] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=2]
[2024-06-15 18:18:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-15 18:18:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-15 18:18:42] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=2] Added to rxQueue
[2024-06-15 18:18:42] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3]
[2024-06-15 18:18:42] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=56 Len=7]
[2024-06-15 18:18:42] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=216 messageTag=6 status=SUCCESS]
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1]
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=3] Added to rxQueue
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4]
[2024-06-15 18:18:47] debug: 	zh:ember:ezsp: <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=56 Len=29]
[2024-06-15 18:18:47] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [type=DIRECT], [indexOrDestination=6105],
[apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":11,"options":4160,"groupId":0,"sequence":216}], [messageTag=6], [status=DELIVERY_FAILED], [messageContents=10070006000040]
[2024-06-15 18:18:47] debug: 	zh:controller:endpoint: Error: ZCL command 0x001788010bd3aeb7/11 genBasic.read(["dateCode","swBuildId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false,"sendPolicy":"immediate"}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":11,"options":4160,"groupId":0,"sequence":216})
[2024-06-15 18:18:47] debug: 	zhc:ota:common: Updating '0x001788010bd3aeb7' (LWA017) to latest
[2024-06-15 18:18:47] debug: 	zhc:ota:common: Using endpoint '11'
[2024-06-15 18:18:47] debug: 	zh:controller:endpoint: CommandResponse 0x001788010bd3aeb7/11 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false,"sendPolicy":"immediate"})
[2024-06-15 18:18:47] debug: 	zh:ember:queue: Status queue=0 priorityQueue=0.
[2024-06-15 18:18:47] debug: 	zh:ember: ~~~> [ZCL to=6105 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":11,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"manufacturerCode":null,"transactionSequenceNumber":8,"commandIdentifier":0}]
[2024-06-15 18:18:47] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=57 Len=26]
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=4]
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2]
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=4] Added to rxQueue
[2024-06-15 18:18:47] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5]
[2024-06-15 18:18:47] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=57 Len=7]
[2024-06-15 18:18:47] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=217 messageTag=7 status=SUCCESS]
[2024-06-15 18:18:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-06-15 18:18:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2]
[2024-06-15 18:18:48] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=5] Added to rxQueue
[... cut ...]
[2024-06-15 18:19:47] error: 	z2m: Update of 'Lampada del soggiorno' failed (Device didn't respond to OTA request)
[2024-06-15 18:19:47] debug: 	z2m: Error: Device didn't respond to OTA request
    at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:379:15)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:525:40)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:272:41)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

Nerivec, 2024-06-15:
@Ricc68 Looks like your issue lies elsewhere (not OTA). Try bringing the device in question close to the coordinator, and re-pair the device to it directly. Then see if the device behaves better, you may have a router in-between the two that's making troubles. _As long as you can't send simple commands to the device without encountering problems (stable network), no point even trying OTA, if it doesn't fail at first, it will likely fail at some point (it takes many, many messages to OTA successfully)._

Ricc68, 2024-06-15:
@Nerivec understand your point, but the bulb is 1.5m away from the dongle-e and additionally the bulb is the only router in my small ZigBee network. True that there's lot of WiFi pollution from others and in addition the dongle is near my WiFi router but it was working fairly well before installing the dev branch.

Ricc68, 2024-06-16:
@Nerivec I have moved the dongle 1m further away from the WiFi router and things greatly improved. In 30 mins it updated the bulb and I don't see errors anymore. The good thing, other than having found the way to make my network more reliable, is that with the dev build a failed update is not crashing z2m anymore which I think is the true goal of this topic.
And thanks for having me double check my setup which seems now to have solved the root cause of the failures (finger crossed).

closed 2024-07-01 by Nerivec