Koenkk/zigbee2mqtt #22464: DoubleSocket50AU: power usage is not updating in 1.37.0

andyblac, 2024-05-05:
### What happened?

I updated to 1.37.0-1, but since my power usage is stuck at a fixed usage at time of update (75w), Once I never to previous version it all working again.

### What did you expect to happen?

Power usage to continue to be updated.

### How to reproduce it (minimal and precise)

just update.

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

20221226

### Adapter

zStack3x0 -Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Home Assistant OS

### Debug log

could not get logs to show in Z2M,

andyblac, 2024-05-05:
[image]

updated Z2M on the 2nd May, noticed that my energy report was not working, found issue with Z2M on the 5th May, and reverted.

andyblac, 2024-05-05:
might be related to this https://github.com/Koenkk/zigbee-herdsman-converters/commit/68b6e087227a9419739ad1a986712e4d9a119209 commit ?

Koenkk, 2024-05-05:
Yes, fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-05 by Koenkk

bensuffolk, 2024-11-26:
I am using 1.41.0 via the Home Assistant add on, and I notice that my socket is no longer reporting changes in power. Its stuck on the last power reading it gave and even if I unplug the item from the socket, its still showing 245w, and 6w for the other socket.

I can't be sure when this started to happen.

Is there anything I should do to try and debug this?

julianfs, 2024-11-27:
> I am using 1.41.0 via the Home Assistant add on, and I notice that my socket is no longer reporting changes in power. Its stuck on the last power reading it gave and even if I unplug the item from [...]

What zigbee coordinator are you using? I've found a Sonoff P will stop picking up usage data after a while, but a SLZB-06 has been continuously reliable over a number of months.

bensuffolk, 2024-11-27:
I have the Sonoff Dongle-E which it's running with the ember firmware. Its been working fine for a year or so, I have only noticed the lack of power data in the last few weeks or so.

andyblac, 2024-11-27:
have you rejoined it to the zigbee network recently ?, I have found that if you join the Aurora AONE devices to the zigbee network with a client device attached and consuming power at the time it will fail to update the power readings.

Try, removing the socket from the zigbee network, unplug any client devices attached to the sockets, and then rejoin it to the zigbee network.

bensuffolk, 2024-11-27:
Ah that could be it. 

I have noticed that after a while the socket can stop routing some of my door sensors Aqara door sensors (MCCGQ11LM). Normally I can fix that by turning the power off and back on to the socket. But this time rather than having to tun the ring main off I just pressed the buttons to make it rejoin.

I will try when I get home later and let you know.

andyblac, 2024-11-27:
> Ah that could be it.
>
> I have noticed that after a while the socket can stop routing some of my door sensors Aqara door sensors (MCCGQ11LM). Normally I can fix that by turning the power off and back [...]

yeah, this is why I bought a 2nd SLZB-06M, and flashed to router mode, and forced all my downstairs Aqara clients to that.
There pretty cheap from AliExpress https://www.aliexpress.com/item/1005004942648430.html

bensuffolk, 2024-11-27:
Deleting it and reading it did the trick. I have power monitoring working again now.

Thanks