Koenkk/zigbee2mqtt #22465: 'Change name in HA' disappeared in the dialog box for naming devices.

Wolk9, 2024-05-05:
### What happened?

There used to be an option in de naming device dialog box to change the name in HA accordingly, but it seems disappeared with one of the latests updates.

### What did you expect to happen?

Appearance of the option to change the name in HA accordingly. 

### How to reproduce it (minimal and precise)

Add a device and rename it. You will miss the option to rename the entitiy in HA accordingly.

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

 20210120

### Adapter

 zStack3x0 Slaesh' 

### Setup

HAOS Z2M Add-On

bikeymouse, 2024-05-05:
I also ran into this. 
The way it works now is just changed:
If you change the name in  Z2M device "About" page it will be default also change the HomeAssistant name.

If you want a different name in HA you can go to the settings page for the device and find on the bottom of the page a Home Assistant section where you can set the name.
Once you have done that, this name will always be used even if you change the name in Z2M. 

Unfortunately you can not blank out the name in the HA setting section again.

Wolk9, 2024-05-05:
I’m sorry,

But if I use the method you describe here, the name will NOT be changed or set in HA. You’ll get a sensor.203948r320u04234320 or so, and that is completely un-user-friendly.

Changing with the second method is still not changing the name in HA, is my experience just now.

The only thing that helps is search for the device after adding it in and rename it in  Z2M and the rename it the hard way in HA in the entity list. For a temperature sensor that means for all the attributes…

😐

adnansarajlic, 2024-05-05:
Abort page now
[image]

r0bb10, 2024-05-05:
> I also ran into this. The way it works now is just changed: If you change the name in Z2M device "About" page it will be default also change the HomeAssistant name.
>
> If you want a different name in [...]

No, it does not work like that, @bikeymouse and @Wolk9 are right, the option disappeared and renaming the device will not update the uid in homeassistant.

Please bring back the rename in homeassistant.

Koenkk, 2024-05-05:
Already fixed in the dev branch, a hotfix release will come soon.

closed 2024-05-05 by Koenkk