Koenkk/zigbee2mqtt #22466: Zigbee Dongle E and EZSP driver no longer work after upgrading to 1.37.0

jakwarrior, 2024-05-05:
### What happened?

When updating to version 1.37.0 with the Zigbee Dongle E adapter and EZSP driver, an error occurs at startup and no messages are sent or received anymore. I know that the EZSP driver is now deprecated, but I wanted to transition to version 1.37.0 before updating the dongle firmware and switching to the ember driver. Here are the logs during startup:

```
[2024-05-05 14:53:30] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-05 14:53:31] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-05 14:53:31] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-05 14:53:31] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-05-05 14:53:31] error: 	zh:ezsp:uart: <-- CRC error: 3d2eb1a96b2a164fa674eb04aaa76d6b2a134fa6944a25aa559249984e278d12ce67add57e|add5|bbbd
[2024-05-05 14:53:31] error: 	zh:ezsp:uart: Error while parsing to NpiFrame 'Error: <-- CRC error: 3d2eb1a96b2a164fa674eb04aaa76d6b2a134fa6944a25aa559249984e278d12ce67add57e|add5|bbbd
    at Frame.checkCRC (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/frame.ts:87:19)
    at SerialDriver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:169:19)
    at Parser.emit (node:events:517:28)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/parser.ts:49:26)
    at Parser.Transform._write (node:internal/streams/transform:175:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Parser.Writable.write (node:internal/streams/writable:337:10)
    at SerialPort.ondata (node:internal/streams/readable:809:22)
    at SerialPort.emit (node:events:517:28)'
[2024-05-05 14:53:35] info: 	z2m: zigbee-herdsman started (resumed)
[2024-05-05 14:53:35] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":"1 ","majorrel":"7","minorrel":"3","product":12,"revision":"7.3.1.0 build 176"},"type":"EZSP v12"}'
[2024-05-05 14:53:35] info: 	z2m: Currently 9 devices are joined:
[2024-05-05 14:53:35] info: 	z2m: Thermomètre salon (0x142d41fffe36f768): SNZB-02D - SONOFF Temperature and humidity sensor with screen (EndDevice)
[2024-05-05 14:53:35] info: 	z2m: prise1 (0xa4c13870f141b9ee): A1Z - Nous Smart plug (with power monitoring) (Router)
[2024-05-05 14:53:35] info: 	z2m: Hue Chambre (0x00178801029cbdf6): 915005106701 - Philips Hue white and color ambiance LightStrip plus (Router)
[2024-05-05 14:53:35] info: 	z2m: Styrbar1 (0x30fb10fffe01331e): E2001/E2002 - IKEA STYRBAR remote control (EndDevice)
[2024-05-05 14:53:35] info: 	z2m: Hue Salon (0x0017880103d19c4f): 8718696548738 - Philips Hue white ambiance E26/E27 (Router)
[2024-05-05 14:53:35] info: 	z2m: Hue Télécommande 1 (0x0017880102d7445e): 324131092621 - Philips Hue dimmer switch (EndDevice)
[2024-05-05 14:53:35] info: 	z2m: Capteur porte entrée (0x00124b002928c70a): SNZB-04 - SONOFF Contact sensor (EndDevice)
[2024-05-05 14:53:35] info: 	z2m: Thermomètre chambre bébé (0x142d41fffe36f747): SNZB-02D - SONOFF Temperature and humidity sensor with screen (EndDevice)
[2024-05-05 14:53:35] info: 	z2m: Ampoule champignon (0xb43522fffe6a97ca): LED2107C4 - IKEA TRADFRI bulb E14, white spectrum, candle, opal, 470 lm (Router)
[2024-05-05 14:53:35] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-05 14:53:36] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-05-05 14:53:36] info: 	z2m: Connected to MQTT server
[2024-05-05 14:53:36] info: 	z2m: Started frontend on port 8099
[2024-05-05 14:53:36] info: 	z2m: Zigbee2MQTT started!
```

When starting up in version 1.36.1, everything works perfectly:
```
info  2024-05-05 14:46:56: Logging to console and directory: '/config/zigbee2mqtt/log/2024-05-05.14-46-56' filename: log.txt
info  2024-05-05 14:46:56: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
info  2024-05-05 14:46:56: Starting zigbee-herdsman (0.40.3)
info  2024-05-05 14:47:01: zigbee-herdsman started (resumed)
info  2024-05-05 14:47:01: Coordinator firmware version: '{"meta":{"maintrel":"1 ","majorrel":"7","minorrel":"3","product":12,"revision":"7.3.1.0 build 176"},"type":"EZSP v12"}'
info  2024-05-05 14:47:01: Currently 9 devices are joined:
info  2024-05-05 14:47:01: Thermomètre salon (0x142d41fffe36f768): SNZB-02D - SONOFF Temperature and humidity sensor with screen (EndDevice)
info  2024-05-05 14:47:01: prise1 (0xa4c13870f141b9ee): A1Z - Nous Smart plug (with power monitoring) (Router)
info  2024-05-05 14:47:01: Hue Chambre (0x00178801029cbdf6): 915005106701 - Philips Hue white and color ambiance LightStrip plus (Router)
info  2024-05-05 14:47:01: Styrbar1 (0x30fb10fffe01331e): E2001/E2002 - IKEA STYRBAR remote control (EndDevice)
info  2024-05-05 14:47:01: Hue Salon (0x0017880103d19c4f): 8718696548738 - Philips Hue white ambiance E26/E27 (Router)
info  2024-05-05 14:47:01: Hue Télécommande 1 (0x0017880102d7445e): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2024-05-05 14:47:01: Capteur porte entrée (0x00124b002928c70a): SNZB-04 - SONOFF Contact sensor (EndDevice)
info  2024-05-05 14:47:01: Thermomètre chambre bébé (0x142d41fffe36f747): SNZB-02D - SONOFF Temperature and humidity sensor with screen (EndDevice)
info  2024-05-05 14:47:01: Ampoule champignon (0xb43522fffe6a97ca): LED2107C4 - IKEA TRADFRI bulb E14 WS candle 470lm, wireless dimmable white spectrum/chandelier opal white (Router)
info  2024-05-05 14:47:01: Zigbee: disabling joining new devices.
info  2024-05-05 14:47:01: Connecting to MQTT server at mqtt://core-mosquitto:1883
info  2024-05-05 14:47:02: Connected to MQTT server
info  2024-05-05 14:47:02: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2024-05-05 14:47:02: Started frontend on port 8099
info  2024-05-05 14:47:02: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0xe0798dfffef438f3/connection_state/config', payload '{"device":{"hw_version":"EZSP v12 7.3.1.0 build 176","identifiers":["zigbee2mqtt_bridge_0xe0798dfffef438f3"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.36.1"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0xe0798dfffef438f3_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
info  2024-05-05 14:47:02: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0xe0798dfffef438f3/restart_required/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"EZSP v12 7.3.1.0 build 176","identifiers":["zigbee2mqtt_bridge_0xe0798dfffef438f3"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.36.1"},"device_class":"problem","enabled_by_default":false,"entity_category":"diagnostic","name":"Restart required","object_id":"zigbee2mqtt_bridge_restart_required","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0xe0798dfffef438f3_restart_required_zigbee2mqtt","value_template":"{{ value_json.restart_required }}"}'
info  2024-05-05 14:47:02: MQTT publish: topic 'homeassistant/button/1221051039810110150109113116116_0xe0798dfffef438f3/restart/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/bridge/request/restart","device":{"hw_version":"EZSP v12 7.3.1.0 build 176","identifiers":["zigbee2mqtt_bridge_0xe0798dfffef438f3"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"1.36.1"},"device_class":"restart","name":"Restart","object_id":"zigbee2mqtt_bridge_restart","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"payload_press":"","unique_id":"bridge_0xe0798dfffef438f3_restart_zigbee2mqtt"}'
[... cut ...] join_timeout is defined, value_json.permit_join_timeout, None) }}"}'
[... 1 lines cut ...]
```

### What did you expect to happen?

The startup proceeds normally, and all devices work without any issues.

### How to reproduce it (minimal and precise)

Update from 1.36.1 to 1.37.0 through Home Assistant

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff Zigbee Dongle E

### Setup

Raspberry Pi 5, Home Assistant 2024.5.1, Home Assistant OS 12.2

LaurentChardin, 2024-05-05:
I am also using the 1.37.0 version with the same dongle (same firmware) and works fine on my side.

I could sometimes have an error on a rare occasion when launching z2m (usually when the dongle is not properly re-initiated) but restarting z2m usually does the trick.

jakwarrior, 2024-05-05:
I tried restarting z2m several times but the problem was still present.

LaurentChardin, 2024-05-05:
Ok i was checking on my side when z2m restarts (as i was testing some code) : i do have actually the same error message, but it does not stop z2m from working. Can you enable debug mode to see what's going on ?

jakwarrior, 2024-05-05:
In fact, I made a mistake, it's working. I thought it wasn't working anymore because the "Logs" tab in the frontend doesn't display anything. And when I open the log.log file, I see that the messages "z2m: MQTT publish: topic ..." are now at the debug level instead of info as before. I don't know if this logging issue is related to the startup error. So there is still an issue with the "Logs" tab which remains empty.

Koenkk, 2024-05-05:
This is because `MQTT publish` is not logged anymore to `info` but to `debug`, and debug is not logged in the frontend anymore (see the release notes)

closed 2024-05-05 by Koenkk