Koenkk/zigbee2mqtt #22467: Sunricher ZG2858A no action on color wheel

adescamps, 2024-05-05:
### What happened?

Hello,

I've included the [device](https://www.zigbee2mqtt.io/devices/ZG2858A.html), device is recognized and interviewed completed I can on/off, change color temperature, brightness up/down

When I tried to change the color with the color wheel, nothing happens and in the logs I can see:

```
debug: 	zh:deconz:frameparser: response payload: 1,5,6,51,254,6,0,0,0
debug: 	zh:controller: Received payload: clusterID=768, address=55992, groupID=47105, endpoint=1, destinationEndpoint=0, wasBroadcast=true, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":5,"commandIdentifier":6},"payload":{"hue":51,"saturation":254,"transtime":6},"command":{"ID":6,"parameters":[{"name":"hue","type":32},{"name":"saturation","type":32},{"name":"transtime","type":33}],"name":"moveToHueAndSaturation"}}
debug: 	zh:controller: Skipping command 'moveToHueAndSaturation' because it is missing from the lookup
```

When testing I never got the events:
`color_loop_set`, `enhanced_move_to_hue_and_saturation`, `hue_stop`

Seems that the error in the log is the same as in https://github.com/Koenkk/zigbee2mqtt/issues/21210#issuecomment-2021530645

Let me know if you need more logs/info or if it's enough.

Thanks !!

### What did you expect to happen?

That some events are triggered when using the color wheel so we can retrieve the desired color

### How to reproduce it (minimal and precise)

Include the remote with default converter and try to use the color wheel on any group.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Setup

RPi 4, Add-on on Home Assistant 2024.5.1, OS 12.2

Koenkk, 2024-05-05:
Fixed!

adescamps, 2024-05-06:
Awesome.
I confirm that I now get the action `move_to_hue_and_saturation` with 1.37.0-dev commit: [358e2ee](https://github.com/Koenkk/zigbee2mqtt/commit/358e2ee)

Thank you!

closed 2024-05-06 by adescamps