Koenkk/zigbee2mqtt #22468: iluminize LED driver no longer working  after upgrading to 1.37.0

nomike, 2024-05-05:
### What happened?

I just upgraded from version 1.36.1  to 1.37.0 and since then, three of my lights are no longer available. Zigbee2MQTT reports the Manufacturer in the Overview as "Unsupported". It shows the support status as "Null".

Zigbee Model: DIM Lighting
Zigbee Manufacturer: Sunricher

[image]

Device Manufacturer: c
Model:  511.210T

It worked perfectly fine prior to the update.

I will try to downgrade to the previous release to have my system working again.

### What did you expect to happen?

I would expect the light to still work after the update.

### How to reproduce it (minimal and precise)

1. Own the light
2. Add it to HA
3. Upgrade Zigbee2MQTT to version 1.37.0

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12 - Sonoff

### Setup

HA Addon, HA running in a kvm virtual machine on a baremettal desktop pc

### Debug log

[45df7312_zigbee2mqtt_2024-05-05T15-42-18.918Z.log](https://github.com/Koenkk/zigbee2mqtt/files/15213553/45df7312_zigbee2mqtt_2024-05-05T15-42-18.918Z.log)

nomike, 2024-05-05:
I can confirm that restoring the backup of 1.36.1 solved the issue.
If you tell me what you need, I'm happy to do some experiments over night and provide more logs or whatever you need.

Koenkk, 2024-05-05:
Could you provide the data/database.db entry of this device?

benjaminBoehme, 2024-05-07:
I lost support for iluminize 511.000 since upgrading to 1.37.0. I observe the same things as statet in the original post of this issue. Previous versions of Z2M work fine with it. Not sure if there's a general problem with devices of this manufacturer (see #22491) so I didn't want to open another issue (but if required/prefered I can do so of course).

<details>
<summary>db entry for this device</summary>
<code>{"id":11,"type":"Router","ieeeAddr":"0x000d6f0010682104","nwkAddr":65249,"manufId":4107,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"DIM Lighting","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25,4096],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{"scenes":{"1_20":{"name":"Hell-warm","state":{"state":"ON","brightness":254,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"2_20":{"name":"Hell-neutral","state":{"state":"ON","brightness":254,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"3_20":{"name":"Sonnenuntergang","state":{"state":"ON","brightness":173,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"4_20":{"name":"Aufwachen","state":{"state":"ON","brightness":254,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"5_20":{"name":"gedimmt","state":{"state":"ON","brightness":87,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}},"1_23":{"name":"Hell","state":{"state":"ON","brightness":254,"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106},"color_temp":370}}}}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"NULL","swBuildId":"2.4.1_r30","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1715091354240}</code>
</details>

Koenkk, 2024-05-07:
Thanks, fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-07 by Koenkk

frest-de, 2024-10-25:
I have exactly the same problem after updating, with another device of this vendor.

Device Manufacturer: iluminize
Model: 511.000

Worked for years with older versions. Seems to be a general Problem with devices of this vendor?