Koenkk/zigbee2mqtt #22469: The "ember" driver of Zigbee2MQTT v1.37.0 doesn't work with Sonoff Dongle-E firmware v7.4.2 (but is okay with fw v7.4.1)

E-DESVIGNE, 2024-05-05:
### What happened?

This morning, my Zigbee2MQTT was up to date (version 1.37.0), but I was still using the "ezsp" driver to control my Sonoff Dongle-E (SONOFF Zigbee 3.0 USB Dongle Plus V2), which had an old firmware version (ncp-uart-sw_v6.10.3_115200). Note: for your information, I'm working on a Raspberry Pi 3B+ running Debian GNU/Linux 11.

As I received the alert message "ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in the next release", I didn't want to wait. I wanted to update my dongle and switch to the new "ember" driver.

Easy steps:
1) I stopped zigbee2mqtt:
`service zigbee2mqtt stop`
2) I used "universal-silabs-flasher" to flash the latest firmware for my dongle (ncp v7.4.2.0) with the command: 
`universal-silabs-flasher --device /dev/ttyACM0 flash --firmware ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl`

Everything went smoothly except... I couldn't restart Zigbee2MQTT! I won't bore you with all the tests I did this afternoon, but I finally found the solution. I'm posting it here to help anyone who might encounter the same problem as me.

The "ember" driver of Zigbee2MQTT v1.37.0 doesn't work with the firmware "ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl"! I downgraded my dongle to the version just before (v7.4.1) with: 
`universal-silabs-flasher --device /dev/ttyACM0 flash --firmware ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl --force`

And since then... everything works perfectly fine :-)

Conclusion: Zigbee2MQTT version 1.37.0 doesn't work with a Sonoff Dongle-E having the firmware ncp v7.4.2.0, but works perfectly fine with the firmware ncp v7.4.1.0.

At this point, I'm not sure if the problem comes from the "ember" driver provided with Zigbee2MQTT version 1.37.0, or if it's the firmware "ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl" that has a defect that "ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl" doesn't have. I'll let the specialists investigate, but I'm sharing the information here in case it can save a few users from losing 2 hours like me ;-)

### How to reproduce it (minimal and precise)

A "SONOFF Zigbee 3.0 USB Dongle Plus V2" with "ncp-uart-hw-v7.4.2.0-zbdonglee" firmware, and Zigbee2MQTT version 1.37.0 with "ember" driver activated.
Ultimately, everything worked when I downgraded the firmware of the dongle to version "ncp-uart-hw-v7.4.1.0-zbdonglee".

### Zigbee2MQTT version

1.37.0 commit: ee5cf5a

### Adapter firmware version

ncp-uart-hw-v7.4.2.0-zbdonglee-115200.gbl

### Adapter

Sonoff Dongle-E (SONOFF Zigbee 3.0 USB Dongle Plus V2)

### Setup

Raspberry Pi 3B+ running Debian GNU/Linux 11 + Domoticz Version 2024.4

mamrai1, 2024-05-05:
You can try 7.4.3 firmware...
https://github.com/darkxst/silabs-firmware-builder/tree/4.4.3/firmware_builds

E-DESVIGNE, 2024-05-05:
> You can try 7.4.3 firmware... https://github.com/darkxst/silabs-firmware-builder/tree/4.4.3/firmware_builds

Well spotted... it works with v7.4.3! So, it's indeed the firmware 7.4.2 that has an issue. 
Thanks (I hadn't noticed there was a 7.4.3).

ChrisLSaar, 2024-05-05:
Hello everyone

I can't get the ZigbeeDongle E to work with the latest firmware either.

My setup...
- NiPoGI Intel N100, 16 GB RAM, Debian 12, Docker environment
- Zigbee2Mqtt 1.37.0
- SONOFF Zigbee 3.0 USB Dongle Plus V2, FW firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl

Debug log
[Zigbee2Mqtt_Log-2024.5.5_20.5.39-(HomeServer).tar.gz](https://github.com/Koenkk/zigbee2mqtt/files/15213984/Zigbee2Mqtt_Log-2024.5.5_20.5.39-.HomeServer.tar.gz)

Nerivec, 2024-05-05:
@ChrisLSaar `MQTT error: connect ECONNREFUSED 192.168.11.10:1883` Make sure your MQTT is okay and that the conf is correct.
_PS: The initial startup fail (your first log) is under investigation, looks to be related to some virtualization setups. It should start fine after that first error._

ChrisLSaar, 2024-05-05:
@Nerivec 

How embarrassing.
Thank you, of course you're right.
Sometimes you overlook the obvious when you concentrate on the little things.
Firewall release was missing.

Now it's running, I'll keep testing

yaskad, 2024-05-05:
Hey thanks for sharing! In my case it still showing me the error

22:41:57] INFO: Preparing to start...
[22:41:58] INFO: Socat not enabled
[22:41:59] INFO: Starting Zigbee2MQTT...
[2024-05-05 22:42:01] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-05 22:42:03] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-05 22:42:03] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-05 22:42:03] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-05-05 22:42:08] info: 	z2m: zigbee-herdsman started (resumed)
[2024-05-05 22:42:08] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"7","minorrel":"4","product":13,"revision":"7.4.3.0 build 0"},"type":"EZSP v13"}'

As you can see i am running the 7.4.3.0 build. And have changed the `configuration.yml` file to using ember
`serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230505134542-if00
  adapter: ember`
Ive also tried to change the serial adapter in the UI to ember but it reverts back to ezsp after saving the changes.

My setup
Sonoff Dongle-E (SONOFF Zigbee 3.0 USB Dongle Plus V2)
ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl
1.37.0
Home Assistant Blue as hardware

Not sure what to do tbh, thankful for any help! :)

Nerivec, 2024-05-05:
@yaskad Shut down Z2M, edit the setting directly inside the `configuration.yaml` file then start again. As long as you are editing the right file (in case you changed the default data directory, you have to change the file in that one), that should work.

```yaml
serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230505134542-if00
  adapter: ember
```

yaskad, 2024-05-06:
@Nerivec I was a bit confused about your comment. 

The configuration file that I have edited is in 
/config/zizbe2mqtt/configuration.yml

But somehow the addon did not pick up the changes even after stop and start. But, your comment get me thinking if there was another config that I missed. Then I found this out 

[image]

Once I changed there I could see ember was begin used now

[2024-05-06 08:07:44] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-06 08:07:44] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-06 08:07:44] info: 	zh:ember:ezsp: ======== EZSP starting ========

Anyway, I don't really know where that configuration was saving its value but good to know for the future:)! Thanks for your help!

johnholman, 2024-05-06:
I had the same problem as yaskad, and the solution was the same: if you are using the zigbee add-in you have to change "ezsp" to "ember" in the add-in configuration as this overrides the value in the config file. In fact it seems sufficient to change only the add-in config as that makes the change to the configuration.yaml file for you.

I think this requires a change to code or at least to documentation - it's easy to overlook this extra layer of configuration when using the zigbee2mqtt add-in.

rgforward4, 2024-05-06:
Hello, by now I feel like I've tried everything. Old and new firmwares, init_firmware, delete config file and recreate. Set driver to "ember" and back again. Long and short device path. Nothing helped. Maybe one of you will find a mistake on my part that I am simply overlooking.

**Zigbee2MQTT version:**
1.37.0 commit: ee5cf5a

**Adapter firmware version:**
ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl

**Adapter:**
Sonoff Dongle-E (SONOFF Zigbee 3.0 USB Dongle Plus V2)

**Setup:**
Lenovo ThinkCentre M93p
            
Lg Raphael

An e.g. Logfile

`[2024-05-06 00:18:39] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-06 00:18:40] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-06 00:18:40] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-06 00:18:40] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-06 00:18:40] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-06 00:18:40] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-06 00:18:40] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-05-06 00:18:40] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-06 00:18:40] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-06 00:18:42] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-06 00:18:42] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-06 00:18:45] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-06 00:18:45] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-06 00:18:47] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-06 00:18:47] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-06 00:18:50] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-06 00:18:50] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-06 00:18:52] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-06 00:18:52] error: 	z2m: Failed to start zigbee
[2024-05-06 00:18:52] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-06 00:18:52] error: 	z2m: Exiting...
[2024-05-06 00:18:53] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:798:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2660:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)`

The connfig:

`homeassistant: true
advanced:
  network_key:
    - 8
    - 44
    - 75
    - 185
    - 126
    - 223
    - 89
    - 158
    - 206
    - 107
    - 5
    - 135
    - 66
    - 25
    - 107
    - 67
  pan_id: 20816
  ext_pan_id:
    - 49
    - 144
    - 102
    - 147
    - 81
    - 91
    - 68
    - 168
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
serial:
  port: /dev/ttyACM0
frontend:
  port: 8099
device_options:
  legacy: false`

Nerivec, 2024-05-06:
@rgforward4 Looks like it's unable to init the adapter on start. Can you [set your log level](https://www.zigbee2mqtt.io/guide/usage/debug.html#debug) to `debug` and send back the log file so I can get more details?

Nerivec, 2024-05-06:
Looks like your adapter is not replying to Z2M at all. Use the `/dev/serial/by-id/...` [path for your adapter](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#configuration-of-the-zigbee-adapter) in case you have multiple devices plugged in, to ensure the proper one is used by Z2M.

rgforward4, 2024-05-06:
Yes, something is slowly dawning on me. I first tried it with a texas instrument. But I gave up because the system wasn't stable enough. Could it be that the first stick has written itself deep into the system somewhere? Do you mean the long device path? If so, I had already tried that. Or am I misunderstanding you?

rgforward4, 2024-05-06:
Okay, i testet it with `/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231214194237-if00` but it doesent work again.

Nerivec, 2024-05-06:
A few things to try:
- Flash this [firmware again](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl), make sure it loads properly again at the end (`Continue` button in the webflasher; it should load without error)
- Shutdown Z2M, physically unplug your adapter for a minute or so, plug it back in, then try starting Z2M.

mamrai1, 2024-05-06:
Weird problem. Mine works with both ember and ezsp driver and also all firmwares above 7.4.0
Even changing the driver and rebooting z2m had no problems. Can you try and flash the cleanup file:

universal-silabs-flasher --device /dev/ttyACM0 --bootloader-reset sonoff flash --firmware nvm3_initfile.gbl

felipejfc, 2024-05-06:
Updated to 7.4.2 today and changed config to use ember, no problem at all here

rgforward4, 2024-05-06:
> A few things to try:
>
> * Flash this [firmware again](https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl), make sure it [...]

Nice @Nerivec, that works!!! 7.4.1 works for me. All the other Firmware Version crasht, including 7.4.2

tomonetml, 2024-05-07:
I experienced all the problems raised in the last days found here and in the others Issues opened since i changed driver to ember and updated to 1.37.0. Tryed all firmware from 7.4.0 to 7.4.4, i had to roll back to 1.36.0 and flashed firmware 7.4.1.

FredC27, 2024-05-08:
Hi everyone,
I'm faced with a strange problem with Z2M 1.37.0 / Sonoff E v2 7.4.3. I tried many firmwares but those errors still raising with the 7.4 firmware.
I can't understand from where they come. I thought it was from GreenPower unicast,... It is not linked to a specific module neither. The messages comes like that :

[2024-05-08 04:35:26] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=91 Len=26].
[2024-05-08 04:35:26] error: 	zh:ember: !!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
[2024-05-08 04:35:26] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=91 Len=26].
[2024-05-08 04:35:26] error: 	zh:controller:device: Default response to 0xa4c138a17837c33e failed

Any ideas ? what the ID 52 stands for ? wrong direction ??

Thanks

Nerivec, 2024-05-08:
That wrong direction means Z2M received a "command" instead of a "response" from the adapter. It seems to happen with interferences sometimes (around the adapter), check that first.

Nerivec, 2024-05-08:
@ESANTOS74 English please.
`Failed to init port with error Error: Error Resource temporarily unavailable Cannot lock port`
Looks like something is currently using your port. Make sure to disable anything that might be before you start Z2M.

ESANTOS74, 2024-05-08:
Valeu Amigo, porem agora tenho esse erro:

[16:03:39] INFO: Preparing to start...
[16:03:39] INFO: Socat not enabled
[16:03:39] INFO: Starting Zigbee2MQTT...
[2024-05-08 16:03:39] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-08 16:03:40] info: 	z2m: Starting Zigbee2MQTT version 1.37.0 (commit #unknown)
[2024-05-08 16:03:40] info: 	z2m: Starting zigbee-herdsman (0.45.0)
[2024-05-08 16:03:40] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-08 16:03:40] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-08 16:03:40] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-08 16:03:40] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-05-08 16:03:40] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-08 16:03:40] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-08 16:03:42] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-08 16:03:42] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-08 16:03:45] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-08 16:03:45] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-08 16:03:47] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-08 16:03:47] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-08 16:03:50] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-08 16:03:50] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-08 16:03:53] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-08 16:03:53] error: 	z2m: Failed to start zigbee
[2024-05-08 16:03:53] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-08 16:03:53] error: 	z2m: Exiting...
[2024-05-08 16:03:53] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:798:19)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2660:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:108:27)
    at start (/app/index.js:107:5)

Nerivec, 2024-05-10:
Closing this as no longer relevant to the issue.

If you have trouble starting Z2M:
- make sure you are using the proper firmware, proper baudrate, and corresponding baudrate in Z2M configuration.
- unplug the adapter for 30+ seconds, then plug it back in and start Z2M again
- re-flash the same firmware

[FAQ](https://www.zigbee2mqtt.io/guide/faq/)
[Zigbee2MQTT fails to start](https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html)

Please open a new issue if you have a specific problem that these do not fix.

closed 2024-05-10 by Nerivec

Loic691, 2024-05-20:
Same for me.
With 7.4.2 firmware same error.
It would be ok with 7.4.1

RazorDarkAMG, 2024-05-21:
> > You can try 7.4.3 firmware... https://github.com/darkxst/silabs-firmware-builder/tree/4.4.3/firmware_builds
> 
> Well spotted... it works with v7.4.3! So, it's indeed the firmware 7.4.2 that has an issue. Thanks (I hadn't noticed there was a 7.4.3).

Hey,

I have the same hub and the same problem as you, how did you install the firmware on it?

thank you!

kwetnico, 2024-05-22:
> Closing this as no longer relevant to the issue.
>
> If you have trouble starting Z2M:
>
> * make sure you are using the proper firmware, proper baudrate, and corresponding baudrate in Z2M configuration. [...]

Hello,

The official firmware to sonoff zbdongle-e is 7.4.2 or 7.4.3 ? Thanks

de0xyrib0se, 2024-07-29:
Here is the link to the Sonoff Dongle-E firmwares:

https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware

artemkovalyov, 2024-12-19:
Worked for me with ember and 7.4.5 firmware, stable for over a month now.