Koenkk/zigbee2mqtt #22475: Error with ZNJLBL01LM Aqara Roller shade driver E1 since last z2mqtt Upate to Version 1.37.0

jochenruben, 2024-05-06:
### What happened?

After the Update of z2mqtt the shades don't update the state.
The position is always right. "state" and "running" are wrong.

### What did you expect to happen?

before the update everingthing worked fine for months

### How to reproduce it (minimal and precise)

restart, rebinding etc. doesn't help. There is always the same problem

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

Home Assistant on NUC with zigbee2mqtt an Moquitto Broker all versions up todate

### Debug log

Error 2024-05-06 08:08:37Failed to configure 'Rollo Treppenhaus mitte', attempt 4 (Error: Device 0x54ef4410006ba9fa has no input cluster manuSpecificLumi at getEndpointsWithCluster (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:30:15) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:67:55) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:112:23) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:160:27) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

mamrai1, 2024-05-06:
i can confirm theproblem with also ember and ezsp driver . the blinds work as expected and report position but i get in the logs:

[2024-05-06 19:02:47] debug: 	zh:ezsp: Device left network request received: 32696 54ef4410004a7065
[2024-05-06 19:02:47] debug: 	zh:controller: Device leave '0x54ef4410004a7065'
[2024-05-06 19:02:47] debug: 	zh:controller: Removing device from database '0x54ef4410004a7065'
[2024-05-06 19:02:47] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-05-06 19:02:47] warning: 	z2m: Device 'Left Kitchen Blind' left the network
[2024-05-06 19:02:47] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Left Kitchen Blind","ieee_address":"0x54ef4410004a7065"},"type":"device_leave"}'
[2024-05-06 19:02:47] debug: 	zh:ezsp:uart: <-- [753eb1a9702aadcd3ce40025ba7d317d5d1d9e4db6369c097e]
[2024-05-06 19:02:47] debug: 	zh:ezsp:uart: <-- DATA (7,5,0): 753eb1a9702aadcd3ce40025ba117d1d9e4db6369c097e
[2024-05-06 19:02:47] debug: 	zh:ezsp:uart: --> ACK  (0)
[2024-05-06 19:02:47] debug: 	zh:ezsp:uart: --> [8070787e]
[2024-05-06 19:02:47] debug: 	zh:ezsp:uart: <-- ACK (5): 753eb1a9702aadcd3ce40025ba117d1d9e4db6369c097e
[2024-05-06 19:02:47] debug: 	zh:ezsp:ezsp: <== Frame: 7c90012400b87f65704a001044ef540203919d
[2024-05-06 19:02:47] debug: 	zh:ezsp:ezsp: <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":32696,"newNodeEui64":{"type":"Buffer","data":[84,239,68,16,0,74,112,101]},"status":2,"policyDecision":3,"parentOfNewNodeId":40337}
[2024-05-06 19:02:47] debug: 	zh:ezsp: Device left network request received: 32696 54ef4410004a7065
[2024-05-06 19:02:47] debug: 	zh:controller: Device leave '0x54ef4410004a7065'
[2024-05-06 19:02:47] debug: 	zh:controller: Device leave is from unknown or already deleted device '0x54ef4410004a7065'
[2024-05-06 19:02:52] debug: 	zh:ezsp:uart: <-- [053eb1a9702aadcd3ce40025ba7d317d5d1d9d4e27ab27d37e]
[2024-05-06 19:02:52] debug: 	zh:ezsp:uart: <-- DATA (0,5,0): 053eb1a9702aadcd3ce40025ba117d1d9d4e27ab27d37e
[2024-05-06 19:02:52] debug: 	zh:ezsp:uart: --> ACK  (1)
[2024-05-06 19:02:52] debug: 	zh:ezsp:uart: --> [8160597e]
[2024-05-06 19:02:52] debug: 	zh:ezsp:uart: <-- ACK (5): 053eb1a9702aadcd3ce40025ba117d1d9d4e27ab27d37e
[2024-05-06 19:02:52] debug: 	zh:ezsp:ezsp: <== Frame: 7c90012400b87f65704a001044ef5401000000
[2024-05-06 19:02:52] debug: 	zh:ezsp:ezsp: <== 0x24: {"_cls_":"trustCenterJoinHandler","_id_":36,"_isRequest_":false,"newNodeId":32696,"newNodeEui64":{"type":"Buffer","data":[84,239,68,16,0,74,112,101]},"status":1,"policyDecision":0,"parentOfNewNodeId":0}
[2024-05-06 19:02:52] debug: 	zh:ezsp:driv: handleNodeJoined: change ManufacturerCode for ieee 54ef4410004a7065 to 4447
[2024-05-06 19:02:52] debug: 	zh:ezsp:ezsp: ==> setManufacturerCode: {"code":4447}
[2024-05-06 19:02:52] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"setManufacturerCode","_id_":21,"_isRequest_":true,"code":4447}
[2024-05-06 19:02:52] debug: 	zh:ezsp:uart: --> DATA (5,1,0): 7d000115005f11
[2024-05-06 19:02:52] debug: 	zh:ezsp:uart: --> [513f21a9412a4aa3a2fc7e]
[2024-05-06 19:02:52] debug: 	zh:ezsp:uart: -?- waiting (6)
[2024-05-06 19:02:52] debug: 	zh:ezsp: Device join request received: 32696 54ef4410004a7065
[2024-05-06 19:02:52] debug: 	zh:controller: Device '0x54ef4410004a7065' joined
[2024-05-06 19:02:52] debug: 	zh:controller: Device '0x54ef4410004a7065' accepted by handler
[2024-05-06 19:02:52] debug: 	zh:controller: Deleted device '0x54ef4410004a7065' joined, undeleting
[2024-05-06 19:02:52] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2024-05-06 19:02:52] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/Left Kitchen Blind', payload '{"battery":93,"charging_status":false,"device_temperature":23,"last_seen":"2024-05-06T19:02:52+03:00","linkquality":136,"motor_speed":"high","motor_state":"stopped","position":49,"power_outage_count":0,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":null}'
[2024-05-06 19:02:52] info: 	zh:controller: Interview for '0x54ef4410004a7065' started
[2024-05-06 19:02:52] debug: 	zh:controller:device: Interview - start device '0x54ef4410004a7065'
[2024-05-06 19:02:52] debug: 	zh:controller:device: Interview - skip node descriptor request for '0x54ef4410004a7065', already got it
[2024-05-06 19:02:52] debug: 	zh:ezsp: Requesting 'Active endpoints' for '32696'
[2024-05-06 19:02:52] debug: 	zh:ezsp:driv: ZDO Active_EP_req params: {"dstaddr":32696}
[2024-05-06 19:02:52] info: 	z2m: Device 'Left Kitchen Blind' joined
[2024-05-06 19:02:52] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Left Kitchen Blind","ieee_address":"0x54ef4410004a7065"},"type":"device_joined"}'
[2024-05-06 19:02:52] info: 	z2m: Starting interview of 'Left Kitchen Blind'
[2024-05-06 19:02:53] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Left Kitchen Blind","ieee_address":"0x54ef4410004a7065","status":"started"},"type":"device_interview"}'
[2024-05-06 19:02:53] debug: 	zh:ezsp:ezsp: ==> sendUnicast: {"type":0,"indexOrDestination":32696,"apsFrame":{"profileId":0,"sequence":54,"clusterId":5,"sourceEndpoint":0,"destinationEndpoint":0,"groupId":0,"options":256},"messageTag":55,"message":{"type":"Buffer","data":[54,184,127]}}
[... cut ...]
[2024-05-06 19:03:33] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/Left Kitchen Blind', payload '{"battery":93,"charging_status":false,"device_temperature":22,"last_seen":"2024-05-06T19:03:33+03:00","linkquality":132,"motor_speed":"high","motor_state":"closing","position":0,"power_outage_count":0,"running":true,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":null}'
[2024-05-06 19:03:33] debug: 	zh:ezsp:uart: <-- [60d5b1a97d312a15b65854b624ab5593499c55a31655b19874f1da3c98c774d1a7c8d3d16f8ffbf4dc7e]
[2024-05-06 19:03:33] debug: 	zh:ezsp:uart: <

mamrai1, 2024-05-06:
> ### What happened?
> After the Update of z2mqtt the shades don't update the state. The position is always right. "state" and "running" are wrong.
>
> ### What did you expect to happen?
> before the update [...]

The problem with position reporting is solved in latest dev build of zigbee2mqtt

Koenkk, 2024-05-06:
Should be fixed now, can you check?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

jochenruben, 2024-05-06:
Pardon, I don’t know how to do this at the Home Assistant GUI.

mamrai1, 2024-05-07:
I can confirm that with ezsp and ember driver is fixed

Drewskieza, 2024-05-07:
Thanks for the quick fix, when can we expect the fix in the production release?

jochenruben, 2024-05-07:
many thanks for the fix!

Koenkk, 2024-05-07:
I will make a hotfix release in the coming days.

closed 2024-05-07 by Koenkk