Koenkk/zigbee2mqtt #22476: String error Aqara Temperature and Humidity Sensor

Merril-r, 2024-05-06:
### What happened?

The specific settings for the sensor Temperature calibration, Temperature precision, Humidity calibration and humidity precision give an error that pops up in red in Z2MTT, the sensor is returning a string and not a number. This error continues to pop up for each of the settings. 

Example of error:

[image]

### What did you expect to happen?

Expected no error when leaving values default or changing them.

### How to reproduce it (minimal and precise)

Add any Aqara temperature and humidity sensor.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee Dongle

### Setup

Add on HA on intel nuc

### Debug log

[2024-05-06 08:11:13] error: 	zhc: Failed to apply calibration to 'temperature': 'temperature_precision' is not a number, got string ()
[2024-05-06 08:12:17] error: 	zhc: Failed to apply calibration to 'humidity': 'humidity_calibration' is not a number, got string ()
[2024-05-06 08:14:15] error: 	zhc: Failed to apply calibration to 'temperature': 'temperature_precision' is not a number, got string ()
[2024-05-06 08:17:16] error: 	zhc: Failed to apply calibration to 'humidity': 'humidity_calibration' is not a number, got string ()
[2024-05-06 08:17:26] error: 	zhc: Failed to apply calibration to 'temperature': 'temperature_precision' is not a number, got string ()
[2024-05-06 08:19:24] warning: 	z2m: Failed to ping 'Cabinet Light Two' (attempt 1/1, ZCL command 0xa4c138d2f870cfe8/1 genBasic.read(["zclVersion"],

lux73, 2024-05-07:
check your configuration.yaml or devices.yaml if used

delete all `xxx_calibration:` & `xxx_precision:` Lines

and re-add all your Calibration & Precision Values via WebIf again

i had similar issues few Versions before ^^

lux73, 2024-05-08:
try stop z2m & add the Values manually in the config:

`humidity_calibration: -9` this is what it must look - no qoutes or anything!

Merril-r, 2024-05-08:
> try stop z2m & add the Values manually in the config:
> 
> `humidity_calibration: -9` this is what it must look - no qoutes or anything!

I did what you said and so far no errors. Thank you @lux73

lux73, 2024-05-08:
nice!  pls add [solved] in the Thread Title and close the issue 😉

closed 2024-05-08 by Merril-r

desmond206x, 2024-06-17:
I have the same problem. Would it be helpful to log which device causes the problem?

m10x, 2024-10-16:
> I have the same problem. Would it be helpful to log which device causes the problem?

Yeah, it's pretty inconvinient that it's not logging which device config is causing this