Koenkk/zigbee2mqtt #22479: Aurora Lighting AU-A1ZBDSS double socket power reading stale

jwheare, 2024-05-06:
### What happened?

Running as an add-on on Home Assistant Yellow.

I updated to 1.37.0-1 and the power (W) value for all of my Aurora double sockets got stuck and stopped reporting changed values. The values were still present in the state but unchanging. No errors in logs. Rolling back to 1.36.1-1 fixed the problem.

Device: https://www.zigbee2mqtt.io/devices/AU-A1ZBDSS.html

### Zigbee2MQTT version

1.37.0-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Home Assistant Yellow - Silicon Labs MGM210P Mighty Gecko Module

### Setup

Add-on on Home Assistant Yellow

julianfs, 2024-05-06:
Fixed in #22464

closed 2024-05-06 by Koenkk

Busta999, 2026-10-20:
I’m seeing the same symptom on AU-A1ZBDSS and AU-A1ZBPIAB sockets on Zigbee2MQTT 2.14.2 (two instances). Power freezes at a stale value or stays at 0 W under load, while relays and last_seen keep working.

What I found on a stalled socket:

bridge/request/device/reporting/read for haElectricalMeasurement / activePower shows the stored config is intact (min 10 s, max 300 s, change 0, status 0), yet no reports arrive under a 1 kW load.
A raw dev-console Read of activePower returned 946 W, so the meter is working and only reporting has stopped.
The converter’s configure for these models binds haElectricalMeasurement and sets on/off reporting, but never configures activePower reporting (the AU-A1ZBPIA does).
Sending bridge/request/device/reporting/configure for activePower (min 5, max 300, change 5, per endpoint) returned ok, and the socket immediately resumed live reports. No delete or re-add was needed.

Remove, factory reset and re-add also fixed it, but Reconfigure, Interview and a plain re-pair did not. I don’t know what triggers the stall. This issue mentions it began after upgrading to 1.37.0-1, so it may be worth checking whether a configure or reporting change in that release is involved.