Koenkk/zigbee2mqtt #22480: Received undefined command from '0'

E99yBend, 2024-05-06:
### What happened?

When setting MQTT into pairing mode I get the following message: "Received undefined command from '0'" I have added lots of devices without a problem but with this error it does not do any interviews with new devices.

### What did you expect to happen?

I expected pairing mode to become active

### How to reproduce it (minimal and precise)

Put MQTT in pairing mode. I get the message every time i do that

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff zigbee 3.0

### Setup

HA on PI running from SSD

jschroeder624, 2024-05-06:
I am also getting this exact same error, although for me, it happens whether I am in pairing mode or not. When I click Disable join, or Permit Join, I immediately get the error. While updating Ikea firmware last night, I received the error every 3 minutes and 20 seconds in the logs. This morning, the error continues to be logged. It seems connected to the Ikea firmware updates that have showed up. So far, I have only been successful in updating 1 device out of 6 (3 Rodret, 2 Styrbar, 1 motion detector).

Aughmen, 2024-05-06:
Also me

Koenkk, 2024-05-06:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/22140

closed 2024-05-06 by Koenkk

BenDavidson90, 2024-05-10:
Same here, Z2M 1.37.0, Sonoff Zigbee Dongle E.
No problem with another Z2M container in 1.37.0 too but with a Conbee II.